Koenkk/zigbee2mqtt #3724: Hive SLR2 Support

peterguy04, 2020-06-12:
hello, i am new to zigbee2mqtt and i have tried to go through the instructions to add support for a device but its abit over my head at this point. So wondering if anyone can help me out

The device i would like supported is the Hive Receiver Model is SLR2 by Computime., there is already support for the single channel variant SLR1b (https://www.zigbee2mqtt.io/devices/SLR1b.html)

closed 2020-07-17 by peterguy04

robertalexa, 2021-08-02:
@peterguy04 Can you please give me a hand with something? What is the minimum and maximum temperature degrees that you can set on the SLR2 via the physical termostat? Is it 5 to 32? Thanks

peterguy04, 2021-08-02:
@robertalexa  yes thats correct 5-32 is the range from the thermostat itself

robertalexa, 2021-08-02:
@peterguy04 Thank you for the quick response. I will submit a PR to have z2m reflect that too instead of 7 to 30

robertalexa, 2021-08-02:
@peterguy04 Would you be kind to tell me if you have a thermostat too? If so, which version? I am looking for people to do a test on WPT1 and SLT2. I am tried to add battery reporting. Thanks