Koenkk/zigbee2mqtt #3727: eCozy support improvements

Heronimonimo, 2020-06-12:
# Bug Report
## What happened
Installed an eCozy thermostat unit (https://www.zigbee2mqtt.io/devices/1TST-EU.html). It seems that there are a few points that could need some refinement:
- it does except some commands like 'system_mode':'cool' given in the documentation, but Zigbee2mqtt doesn't show it in the results
- battery values need conversion. I found a mapping here: https://github.com/ckpt-martin/Hubitat/blob/master/eCozy/eCozy-ZigBee-Thermostat-Driver.groovy
- the Home Assistant discovery doesn't show the settings for cooling. Probably
   ` '1TST-EU': [thermostat(), cfg.sensor_battery],` in https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/homeassistant.js needs to be changed to thermostatHeatCool

I already added some of the above as comments in https://github.com/Koenkk/zigbee2mqtt/issues/129 but thought a separate issue might be more appropriate.

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version: 1.14.0
HA version : 0.111.1
Adapter hardware: CC2531,
Adapter firmware version: 20190608

Heronimonimo, 2020-06-12:
Some additional findings.

It seems that the whole battery range implementation might be the same as https://www.zigbee2mqtt.io/devices/WV704R0A0902.html:
"Possible values 0 to 100 (%). The level corresponds voltages in the range 2.2 V (0%) to 3.0 V (100%). The levels where chosen so a 0% indicates that batteries must be replaced, but the valve will still work."

Secondly I think there should be a way to expose the window open and close sensing.

Heronimonimo, 2020-06-12:
After enabling debug mode and looking at the logs it doesn't seem to report back the system mode. But when I send a command to set the mode to cool I can see that it reacts on it.

Is there a way to see a more detailed log?

closed 2020-07-26 by stale[bot]

SoulEater45, 2021-01-10:
Hi, I am using also using one of these eCozy and have some strange behavior.

- Sometime down the road I get an error for setting the temperature where I have no permit to write to the device. This leads me to disconnect and reconnect the device.
- If I completely reset the device (having the batteries longer out), I usually cannot retrieve any stats from the unit. Usually setting the temperature would also yield a response message from the thermostat, but nothing happens and I was forced to make myself buttons for that. So I have an inkonsistent behavior with that.
- And I also wanted to know if there is a **running_state** property in the json, since I do not have it in my responses.