Koenkk/zigbee2mqtt #3728: Develco Smoke Alarm sensor: Temperature & battery not read?

paddenstoeltje, 2020-06-12:
# Bug Report
## What happened
I've bought multiple Develco's Smoke Alarm sensors SMSZB-120. I'm able to configure them successfully and trigger the alarm manually, however the temperature and battery values are not read / updated? I have checked the log files (debug mode) when pairing the devices, however I cannot find the reason why the temperature is not showing up.

## What did you expect to happen
As the smoke alarm also features a temperature sensor, I would expect to have a temperature reading over MQTT. The same is true for the battery value.

## How to reproduce it (minimal and precise)
Each time the smoke alarm gives an update, everything is updated, except the temperatures and the battery values. Example of an update:

zigbee2mqtt:debug 2020-06-10 19:47:48: Received Zigbee message from '0x0015bc00310045ad', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":48,"extendedstatus":0}' from endpoint 35 with groupID 0
zigbee2mqtt:info  2020-06-10 19:47:48: MQTT publish: topic 'zigbee2mqtt/0x0015bc00310045ad', payload '{"voltage":3200,"battery_low":false,"linkquality":68,"enrolled":false,"smoke":false,"tamper":false,"supervision_reports":true,"restore_reports":true,"trouble":false,"ac_status":false}'

## Debug Info
Adapter hardware: CC2531
Version of Zigbee2Mqtt: 1.13.1-dev
Coordinator version: 20190608
I attached the develco manual, together with my pairing log.

Thanks for your time!
Wouter
[smoke-alarm-technical-manual.pdf](https://github.com/Koenkk/zigbee2mqtt/files/4771466/smoke-alarm-technical-manual.pdf)
[develco.txt](https://github.com/Koenkk/zigbee2mqtt/files/4771469/develco.txt)

Koenkk, 2020-06-13:
The reason is that the configure() fails. Please try changing the definition in `devices.js` to

```js
{
    zigbeeModel: ['SMSZB-120'],
    model: 'SMSZB-120',
    vendor: 'Develco',
    description: 'Smoke detector with siren',
    supports: 'smoke, warning, temperature',
    fromZigbee: [fz.temperature, fz.battery, fz.ias_smoke_alarm_1, fz.ignore_basic_report, fz.ignore_genOta],
    toZigbee: [tz.warning],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(35);
        await bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'ssIasZone', 'genBasic']);
        await configureReporting.batteryVoltage(endpoint);
        const endpoint2 = device.getEndpoint(38);
        await bind(endpoint2, coordinatorEndpoint, ['msTemperatureMeasurement']);
        await configureReporting.temperature(endpoint2);
    },
    endpoint: (device) => {
        return {default: 35};
    },
},
```

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

paddenstoeltje, 2020-06-15:
Hi Koenkk,

I modified the devices.js and now it works. Thank you! Can this be updated in the next release, or do I need to submit a pull request?

**The problem:**
configureReporting.batteryPercentageRemaining(endpoint);  
_should be modified to_
await configureReporting.batteryVoltage(endpoint);

Wouter

Koenkk, 2020-06-15:
Added the fix.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-06-15 by Koenkk