Koenkk/zigbee2mqtt #3729: Device reporting bad model Id

nnoally, 2020-06-12:
Hi

I try to connect this 2 gang switch to zigbee2mqtt: https://fr.aliexpress.com/item/4000807168877.html.
It seems to be a reband of https://fr.aliexpress.com/item/4000298926256.html that seems working (https://github.com/Koenkk/zigbee2mqtt/issues/2971)
BUT the model id is **'TS0601'**, so its defined as curtain motor in Home Assistant. When i try to "move curtains up" or "stop curtains" the first switch turn on. When i try to "move curtains down" the first switch turn off.

When i turn the first switch on then off.
`owvfni3: Unhandled DP #257: {"status":0,"transid":51,"dp":257,"fn":0,"data":{"type":"Buffer","data":[1]}}`
`owvfni3: Unhandled DP #257: {"status":0,"transid":52,"dp":257,"fn":0,"data":{"type":"Buffer","data":[0]}}`

When i turn the second switch on then off
`owvfni3: Unhandled DP #258: {"status":0,"transid":53,"dp":258,"fn":0,"data":{"type":"Buffer","data":[1]}}`
`owvfni3: Unhandled DP #258: {"status":0,"transid":54,"dp":258,"fn":0,"data":{"type":"Buffer","data":[0]}}`

I've try to map 'TS0601' device id with GDKES-03TZXD and GDKES-02TZXD by changing device.js. No success :-(
I think i just have the wrong tuya firmware in this crapy device.

Does anybody have an idea?

Zigbee2mqtt version: 1.14.0
Adapter hardware: CC2531
Adapter firmware version: 1.2 Source routing

Koenkk, 2020-06-13:
- Try to remove the existing entry from of `TS0601` from `devices.js`
- Change the entry of `GDKES-03TZXD` from `zigbeeModel: ['TS0013'],` -> `zigbeeModel: ['TS0013', 'TS0601'],`, does this work?

Machaeon, 2020-06-13:
Same problem. 
>     * Change the entry of `GDKES-03TZXD` from `zigbeeModel: ['TS0013'],` -> `zigbeeModel: ['TS0013', 'TS0601'],`, does this work?

It doesn't in my case. I also try TS0012 because I have 2 gang switch.

nnoally, 2020-06-13:
Not working too.

Koenkk, 2020-06-14:
Can you share more your debug logging after executing commands?

Machaeon, 2020-06-14:
For left button:
`zigbee2mqtt:info  2020-06-14 13:55:02: MQTT publish: topic 'zigbee/Bathroom switch', payload '{"state_left":"ON"}'
zigbee2mqtt:info  2020-06-14 13:55:03: MQTT publish: topic 'zigbee/Bathroom switch', payload '{"state_left":"OFF"}'`
Button state changed, but has no effect on switch.
For right button on:
`zigbee2mqtt:error 2020-06-14 14:03:55: Failed to call 'EntityPublish' 'onMQTTMessage' (TypeError: Cannot read property 'constructor' of undefined
    at EntityPublish.onMQTTMessage (/zigbee2mqtt-1.14.0/lib/extension/publish.js:169:52)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.14.0/lib/controller.js:352:44)
    at Controller.onMQTTMessage (/zigbee2mqtt-1.14.0/lib/controller.js:246:14)
    at MQTT.emit (events.js:223:5)
    at MQTT.onMessage (/zigbee2mqtt-1.14.0/lib/mqtt.js:95:14)
    at MqttClient.emit (events.js:223:5)
    at MqttClient._handlePublish (/zigbee2mqtt-1.14.0/node_modules/mqtt/lib/client.js:1269:12)
    at MqttClient._handlePacket (/zigbee2mqtt-1.14.0/node_modules/mqtt/lib/client.js:409:12)
    at work (/zigbee2mqtt-1.14.0/node_modules/mqtt/lib/client.js:320:12)
    at Writable.writable._write (/zigbee2mqtt-1.14.0/node_modules/mqtt/lib/client.js:334:5))`
Then trigger back to off state.

way2many, 2020-06-17:
Seems to be the same as the one I bought https://github.com/Koenkk/zigbee2mqtt/issues/3213

I don't have time currently so I didn't advance in the support of the device.

If you have time, please check the comments from @Koenkk and @dzegarra which may help.

tabletka-ao, 2020-06-17:
linked #2778

gerripeach, 2020-07-14:
This also happens for the Tuya Thermostat. My Thermostat is recocnized as curtain motor as well.
https://www.aliexpress.com/item/4001065140400.html?spm=a2g0o.productlist.0.0.70956b3cMRwtjE&algo_pvid=f117192c-bef2-4392-905f-59f7852f19b9&algo_expid=f117192c-bef2-4392-905f-59f7852f19b9-1&btsid=0b0a050115947494058005565e3f0c&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

So the modelId of `TS0601` is reported by several devices (always Tuya) which are clearly not the same.

if i grep the database...
`root@raspberrypi:/srv/zigbee2mqtt/data# cat database.db | grep TS0601
{"id":18,"type":"EndDevice","ieeeAddr":"0xbc33acfffe6d821d","nwkAddr":1196,"manufId":4098,"manufName":"_TZE200_ckud7u2l","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_ckud7u2l","powerSource":3,"zclVersion":3,"appVersion":83,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[]}},"appVersion":83,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1594748714808}
`

at least the "manufName":"_TZE200_ckud7u2l" corresponds to https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js#L1136

maybe this helps

gerripeach, 2020-07-14:
```
zigbee2mqtt:info  2020-07-14 19:48:32: Device '0xbc33acfffe6d821d' joined
zigbee2mqtt:info  2020-07-14 19:48:32: Starting interview of '0xbc33acfffe6d821d'
zigbee2mqtt:info  2020-07-14 19:48:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0xbc33acfffe6d821d"}}'
zigbee2mqtt:info  2020-07-14 19:48:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0xbc33acfffe6d821d"}}'
zigbee2mqtt:info  2020-07-14 19:48:33: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0xbc33acfffe6d821d"}}'
zigbee2mqtt:info  2020-07-14 19:48:35: Successfully interviewed '0xbc33acfffe6d821d', device has successfully been paired
zigbee2mqtt:info  2020-07-14 19:48:35: Device '0xbc33acfffe6d821d' is supported, identified as: TuYa Radiator valve with thermostat (TS0601_thermostat)
zigbee2mqtt:info  2020-07-14 19:48:35: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0xbc33acfffe6d821d","model":"TS0601_thermostat","vendor":"TuYa","description":"Radiator valve with thermostat","supported":true}}'
```

working in the dev branch for the thermostat

closed 2020-08-21 by github-actions[bot]