Koenkk/zigbee2mqtt #3731: Help for adding a new device

jd1900, 2020-06-12:
I've just added a new Legrand device.

However, before submitting I wanted to do it properly, and I have a few things left to add that I don't know how to do it. I was able to do the basics from here:
https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

The device is identified, paired, reports the value, and home assistant reads it.

It is a power measurement device. 

However, I have a few things left to do:

- I cannot read the "power" using MQTT. I can just see it in HA. How can I do it? I've tried this:
topic:
`zigbee2mqtt/device_name/get`
payload:
```
{
"power": ""
}
```
But I get this error:
`zigbee2mqtt:error 2020-06-12 22:56:16: No converter available for 'power' ()`

- The device reports the power every 5 minutes, but it is possible to ask the devices reading an attribute with this settings:
Cluster: 0x0b04, Command: Read Attributes (0x00), Attribute Active Power (0x050b)
However, I don't find the way to do it. I guess I should add it to "toZigbee.js". But..., how? There is no docs of how to do it. I find a lot of thing not defined anywhere. For instance, the "legrand_settingAlwaysEnableLed" uses "manuSpecificLegrandDevices". Where is that? How is that converted? What's the format of the entity? 
In summary, I don't see a way to write that "simple" command.
- And, after that, if I send the command, i would need to read the value replied. How can I read the value? Where is it stored? In theory the value is already being stored, so I wouldn't need to do it, right? But..., how can I read its value afterwards?
- Finally, I found a way to set an alert, but this uses an unknown attribute.  I just know the attribute (0xf002) and the data type, but... how do I construct the payload? How do I read the answer back?

Hopefully I'll be able to properly add the device with full features.

Koenkk, 2020-06-13:
- To read power, add the following converter below `toZigbee.js` and to the `toZigbee` list in `devices.js`. When now doing a get with `{"power": ""}` this converter will be triggered and will issue a read request, device will respond and the response is handled by the fromZigbee electrical_measurement_power

```js
electrical_measurement_power: {
        key: ['power'],
        convertGet: async (entity, key, meta) => {
            await entity.read('haElectricalMeasurement', ['activePower']);
        },
    },
```

jd1900, 2020-06-14:
Thank you. I'll try to send you the device today or tomorrow at most.
I've made quite a lot of progress.

closed 2020-06-14 by jd1900