Koenkk/zigbee2mqtt #3733: [Feature] Add last_changed attribute to all sensors

CodeFinder2, 2020-06-13:
Not a bug, just a feature request (or an info if it's already available but was not able to find something like this in the docs).

I was searching for a way to get the time when a sensor has changed lastly (in HA). Many forum posts suggest to use the `last_changed` attribute of an entity but, unfortunately, all my Z2M created devices /entities do not have that. (If this is possible with HA ingredients only, I am totally fine with that and would be grateful to get to know, how.)

So I would suggest to add such an attribute that reports the time when a sensor has *actually* changed. Please let me know what do you think about this! :-)
I would say to just reflect state changes, not changes of an attribute. In my setup, 
I would like to get the time my front door was left open (= close => open transition) so `last_changed` should not change when the battery level of that sensor drops, for example. 
(I know there is already `last_seen` but that's obviously something different.) And since there should be a dedicated 'battery entity' for that sensor anyway, `last_changed `of that entity can report the time of the last battery drop if someone needs it. (Hopefully, I am not mixing terms here.)

Thanks for considering this!

## Debug Info
Zigbee2mqtt version: 1.14.0 (latest)
Adapter hardware: CC1352P-2
Adapter firmware version: 2020-04-20 (latest)

Koenkk, 2020-06-13:
@emontnemery is the `last_changed` (or `last_updated`?) property supported for MQTT sensors?

emontnemery, 2020-06-13:
`last_changed` is built into the Home Assistant states, it's not something that needs to be explicitly supported.
There's an explanation here: https://community.home-assistant.io/t/last-updated-state-and-last-changed-in-lovelave/101701/2

Note that `last_changed` is not part of the `attributes` dict.

CodeFinder2, 2020-06-13:
Great, thank you! Exactly what I was looking for! For posterity, that's what I ended up using to print the time when my front door's state changed (Z2M v1.14.0, HA v0.111.1):
```
{{ as_timestamp(states.binary_sensor['0x00158d0002exxxxx_contact'].last_changed) | timestamp_custom('%X', True) }}
```

Will print something like (in my local time format): "16:00:51". Note that if a sensor starts with a digit (0 here which basically applies to all Z2M created sensors, AFAIK), you [cannot use](https://community.home-assistant.io/t/last-updated-state-and-last-changed-in-lovelave/101701/13) the form:
```
{{ states.domain.entity_id.last_changed }}
```
because this causes a syntax error. (`domain` is, e,.g., `binary_sensor` or `light`.)

```
Error rendering template: TemplateSyntaxError: expected token ',', got 'x00158d0002exxxxx_contact'
```

closed 2020-06-13 by CodeFinder2

davekr8, 2020-09-16:
> `last_changed` is built into the Home Assistant states

Is the usage of HomeAssistant a kind of "default use-case"? 

I think a "aquisition time" (the timestamp, when the zigbee message is received, before any time consuming processing is done) is a (the only) relevant info. I don't want the second or third service in chain to handle time-stamps, this will add delay and jitter as well.

So a +1 from me for the "last changed" field. I further would propose the wording "aquisition time" and refer those time-stamps as basis for all derived time-stamps, as "last_seen" etc. pp.

 What do you guys think ?