Koenkk/zigbee2mqtt #3734: zigbee2mqtt is looping with a cpu over 100%

tunerooster, 2020-06-13:
zigbee2mqtt is looping with a cpu over 100%

###### # top -b -n1
```
top - 06:20:54 up 10 min,  1 user,  load average: 1.21, 1.19, 0.72
Tasks: 199 total,   2 running, 197 sleeping,   0 stopped,   0 zombie
%Cpu(s): 25.4 us,  6.0 sy,  0.0 ni, 68.7 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st
MiB Mem :   7782.5 total,   6628.5 free,    779.8 used,    374.2 buff/cache
MiB Swap:   4096.0 total,   4096.0 free,      0.0 used.   6957.2 avail Mem 

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND
 5536 homeass+  20   0  912836 214124  41912 R 118.8   2.7  12:37.50 node
```
Running an strace on the node process reveals:
```
epoll_wait(13, [{EPOLLIN, {u32=16, u64=16}}], 1024, 483) = 1
read(16, "\1\0\0\0\0\0\0\0", 1024)      = 8
futex(0x557259b66fcc, FUTEX_WAKE_PRIVATE, 1) = 1
futex(0x557259b66f60, FUTEX_WAKE_PRIVATE, 1) = 1
epoll_wait(13, [{EPOLLIN, {u32=16, u64=16}}], 1024, 482) = 1
read(16, "\1\0\0\0\0\0\0\0", 1024)      = 8
futex(0x557259b66fcc, FUTEX_WAKE_PRIVATE, 1) = 1
futex(0x557259b66f60, FUTEX_WAKE_PRIVATE, 1) = 1
epoll_wait(13, [{EPOLLIN, {u32=16, u64=16}}], 1024, 482) = 1
read(16, "\1\0\0\0\0\0\0\0", 1024)      = 8
futex(0x557259b66fcc, FUTEX_WAKE_PRIVATE, 1) = 1
futex(0x557259b66f60, FUTEX_WAKE_PRIVATE, 1) = 1
epoll_wait(13, [{EPOLLIN, {u32=16, u64=16}}], 1024, 481) = 1
read(16, "\1\0\0\0\0\0\0\0", 1024)      = 8
futex(0x557259b66fc8, FUTEX_WAKE_PRIVATE, 1) = 1
```
which goes on forever.  This happens *only* when hass is running.

Environment:

* Beelink Gemini X55 8GB ram J5005 CPU
* Gentoo linux (Linux gemini 5.4.30-gentoo kernel - intel) No modemmanager
* vanilla homeassistant 0.108.2 (no docker etc...)
* node v14.0.0
* npm 6.14.5
* zigbee2mqtt (package.json:  "version": "1.13.1")
* USB Dongle: ID 0451:16a8 Texas Instruments, Inc. CC2531 ZigBee
* mosquitto version 1.6.9
* One (1) zigbee device (Xiaomi Aqara Temperature/Humidity/Pressure)
* hostname: homeassistant (local)
* user: homeassistant

CONFIG FILES:

###### # /etc/udev/rules.d/99-com.rules
ACTION=="add", ATTRS{idVendor}=="0451", ATTRS{idProduct}=="16a8", SYMLINK+="zigbee"
###### # ls -l /dev/zigbee
lrwxrwxrwx 1 root root 7 Jun 13 06:10 /dev/zigbee -> ttyACM0
###### # ls -l /dev/ttyACM0
crw-rw---- 1 root uucp 166, 0 Jun 13 06:15 /dev/ttyACM0
###### # ls -l /dev/serial/by-id/
lrwxrwxrwx 1 root root 13 Jun 13 06:39 usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0018ED303E-if00 -> ../../ttyACM0
###### # test -w /dev/ttyACM0 && echo success || echo failure
success
###### # test -w /dev/zigbee && echo success || echo failure
success

###### # grep homeassistant /etc/group
```
uucp::14:uucp,homeassistant
users::100:homeassistant
```
###### # /opt/zigbee2mqtt/data/configuration.yaml
```
homeassistant: true
[... 58 lines cut ...]
info  2020-06-13 06:10:31: Aqara1 (0x00158d000465b965): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
info  2020-06-13 06:10:31: Zigbee: disabling joining new devices.
info  2020-06-13 06:10:31: Connecting to MQTT server at mqtt://homeassistant
info  2020-06-13 06:10:33: Connected to MQTT server
info  2020-06-13 06:10:33: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
info  2020-06-13 06:10:33: MQTT publish: topic 'zigbee2mqtt/Aqara1', payload '{"battery":91,"voltage":2985,"temperature":21.79,"humidity":37.6,"pressure":855.5,"linkquality":18}'
info  2020-06-13 06:10:33: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.13.1","commit":"2cb234c","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190619}},"log_level":"info","permit_join":false}'
info  2020-06-13 06:10:33: MQTT publish: topic 'homeassistant/sensor/0x00158d000465b965/temperature/config', payload '{"unit_of_measurement":"°C","device_class":"temperature","value_template":"{{ value_json.temperature }}","state_topic":"zigbee2mqtt/Aqara1","json_attributes_topic":"zigbee2mqtt/Aqara1","name":"Aqara1_temperature","unique_id":"0x00158d000465b965_temperature_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d000465b965"],"name":"Aqara1","sw_version":"Zigbee2mqtt 1.13.1","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
info  2020-06-13 06:10:33: MQTT publish: topic 'homeassistant/sensor/0x00158d000465b965/humidity/config', payload '{"unit_of_measurement":"%","device_class":"humidity","value_template":"{{ value_json.humidity }}","state_topic":"zigbee2mqtt/Aqara1","json_attributes_topic":"zigbee2mqtt/Aqara1","name":"Aqara1_humidity","unique_id":"0x00158d000465b965_humidity_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d000465b965"],"name":"Aqara1","sw_version":"Zigbee2mqtt 1.13.1","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
info  2020-06-13 06:10:33: MQTT publish: topic 'homeassistant/sensor/0x00158d000465b965/pressure/config', payload '{"unit_of_measurement":"hPa","device_class":"pressure","value_template":"{{ value_json.pressure }}","state_topic":"zigbee2mqtt/Aqara1","json_attributes_topic":"zigbee2mqtt/Aqara1","name":"Aqara1_pressure","unique_id":"0x00158d000465b965_pressure_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d000465b965"],"name":"Aqara1","sw_version":"Zigbee2mqtt 1.13.1","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
info  2020-06-13 06:10:33: MQTT publish: topic 'homeassistant/sensor/0x00158d000465b965/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/Aqara1","json_attributes_topic":"zigbee2mqtt/Aqara1","name":"Aqara1_battery","unique_id":"0x00158d000465b965_battery_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d000465b965"],"name":"Aqara1","sw_version":"Zigbee2mqtt 1.13.1","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
info  2020-06-13 06:10:33: MQTT publish: topic 'homeassistant/sensor/0x00158d000465b965/linkquality/config', payload '{"icon":"mdi:signal","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/Aqara1","json_attributes_topic":"zigbee2mqtt/Aqara1","name":"Aqara1_linkquality","unique_id":"0x00158d000465b965_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d000465b965"],"name":"Aqara1","sw_version":"Zigbee2mqtt 1.13.1","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
info  2020-06-13 06:11:15: MQTT publish: topic 'zigbee2mqtt/Aqara1', payload '{"battery":91,"voltage":2985,"temperature":21.79,"humidity":37.6,"pressure":855.5,"linkquality":18}'
```

###### # /home/homeassistant/.npm/_logs/2020-06-11T14_36_50_641Z-debug.log

THINGS I HAVE TRIED:

* Press and hold reset button on zigbee USB dongle (closest to USB connector)
* Unplug and replug CC2531 USB dongle
* Successfully reflash firmware (CC2531_SOURCE_ROUTING_20190619.zip) twice
* Plugging the dongle using a USB extension cable

Please help!  I have been working on this for days...

Koenkk, 2020-06-13:
Can you try downgrading to node 12?

tunerooster, 2020-06-14:
> Can you try downgrading to node 12?

Done - no change CPU 121%  Same loop in strace...
node --version
v12.18.0

Thanks for the prompt reply!  Please advise...

Koenkk, 2020-06-14:
Do you see anything suspicious when starting zigbee2mqtt with `DEBUG=* npm start`?

tunerooster, 2020-06-16:
I FOUND THE PROBLEM (Solved)

It was not clear to me when I installed z2m, that it was an *alternative* to HA's ZHA integration (which I had enabled some time ago, even before receiving the zigbee dongle).  It only occured to me when I noticed that z2m would not start if hass was running (due to /dev/ttyACM0 being locked).  (My gentoo runcontrol always started z2m before starting hass.)  Then when hass was started it apparently did not check any locks but began to interfere with ttyACM0 making z2m go crazy.  Once I could see both hass *and* z2m were opening ttyACM0, it dawned on me that they must be independent integrations.  I deleted the HA ZHA integration (through the HA interface), restarted hass and, voild, my Aqara sensors all started working, and z2m no longer used excessive CPU.

If other suffer the same confusion, perhaps if would be useful to state the two are not both needed to use z2m (or maybe I'm just more clueless than most :)

At any rate, THANKS so much for a very useful module and for your attention and help!

closed 2020-06-16 by tunerooster