Koenkk/zigbee2mqtt #3735: Can't pair Ikea Tradfri...

thierry-rhone, 2020-06-13:
Hi,
I have Hassio 0.111.0 with zigbee2mqtt 1.14.0. Work fine with Hue dimmer switch and Hue white ambiance.
Today I bought  1 LED1836G9 and 2 E1743 Tradfri switch from Ikea and Zigbee 2 mqtt. I see nothing in the zigbee2mqtt log !
(I tried to pair switch with this : https://www.zigbee2mqtt.io/devices/E1743.html  without succes)

cichy63, 2020-06-13:
I just resovered my network after working few days with tradfri router. It paired fine and was handling trafic but after powering down just for test of network recovery i got big problem with getting it back to work. Somehow it stopped working and comunicating with coordinator. I was not able to even delete it from by hassio frontend. I needed to delete it manually from database.db file as after each restart it was comming back(router was not powered) . On the way i tried to pair one of the xiaomi door sensow withot luck. As you discribe it was not even giving any log in z2m when trying. I think somehow tradfri devices are not fully compatible and disturb network stability. I look forward for some improvement from more advanced user as this is quite cheap and highly avalible eqipment to use in home automation.

thierry-rhone, 2020-06-13:
Why do you speak of tradfri router it's a Z2m discussion !

cichy63, 2020-06-13:
To give you (and other users) a backroud of situation. It may be that if you paired other tradrfri device before no mather if it's a switch or router it could somehow block you're network and pairing to new devices. Do you have any other tradfi device?

thierry-rhone, 2020-06-13:
Indeed ! Since my network no longer works at all. I can't no longer pair my hues devices.
I deleted the share / zigbee folder (except configuration.yaml) without success. Do you know how I can find normal operation (without Tradfri)

cichy63, 2020-06-14:
What i did was to delete all Tradfri devices form hassio and database.db file (lines including Tradfri and my Xiaomi related line which somehow was still there). Restarted Home Assistant, unplug the zigbee coordinator (in my case CC2531) plug back in, restart and this way recovered the network so messages started to flow again and not used the route trough Tradfri router which as somehow blocking the way from enddivices to connect to any CC2530 router devices. After that i was able to pair again my Xiaomi door switch. Have a look at this page as well https://www.zigbee2mqtt.io/information/FAQ.html. There are point that may help in your case.

thierry-rhone, 2020-06-23:
Do not mix ikea products with other brands. Each time I tried to add an ikea, the recognition of new devices no longer works even deleting the share/zigbee2mqtt folder. I returned all of my ikea products. It's a shame because they were really cheap !

closed 2020-06-24 by thierry-rhone