Koenkk/zigbee2mqtt #3736: Brightness problem on/off after reboot

Hydci, 2020-06-13:
Hello I have a problem with the latest stable version I have two ikea tradfi (TRADFRI LED bulb E12/E14/E17 400 lumen, dimmable warm white, chandelier opal) bulbs when I turn on the state does not change (the light turns on) and therefore I cannot turn it off when I make a set% the light turns on and the state changes well it looks like mqtt retains the brightness at 0 this this product after a restart once done with a set% I can turn on and off normally

After reboot ON/OFF (no work)
juin 13 16:34:08 raspberrypi sudo[488]: zigbee2mqtt:info  2020-06-13 16:34:08: MQTT publish: topic 'zigbee2mqtt/0x000b57fffe8da195', payload '{"state":"ON","brightness":0,"update_available":false}'
juin 13 16:34:19 raspberrypi sudo[488]: zigbee2mqtt:info  2020-06-13 16:34:19: MQTT publish: topic 'zigbee2mqtt/0x000b57fffe8da195', payload '{"state":"OFF","brightness":0,"update_available":false}'
juin 13 16:34:30 raspberrypi sudo[488]: zigbee2mqtt:info  2020-06-13 16:34:30: MQTT publish: topic 'zigbee2mqtt/0x000b57fffe8da195', payload '{"state":"ON","brightness":0,"update_available":false}'
juin 13 16:35:13 raspberrypi sudo[488]: zigbee2mqtt:info  2020-06-13 16:35:13: MQTT publish: topic 'zigbee2mqtt/0x000b57fffe8da195', payload '{"state":"OFF","brightness":0,"update_available":false}'

After set% (the light turns on and the state changes I can therefore turn it off)
juin 13 16:43:27 raspberrypi sudo[488]: zigbee2mqtt:info  2020-06-13 16:43:27: MQTT publish: topic 'zigbee2mqtt/0x000b57fffe8da195', payload '{"state":"ON","brightness":28,"update_available":false}'

after reboot and set% (on/off work perfectly)
juin 13 16:43:57 raspberrypi sudo[488]: zigbee2mqtt:info  2020-06-13 16:43:57: MQTT publish: topic 'zigbee2mqtt/0x000b57fffe8da195', payload '{"state":"ON","brightness":28,"update_available":false}'
juin 13 16:43:58 raspberrypi sudo[488]: zigbee2mqtt:info  2020-06-13 16:43:58: MQTT publish: topic 'zigbee2mqtt/0x000b57fffe8da195', payload '{"state":"OFF","brightness":0,"update_available":false}'

## Debug Info
Zigbee2mqtt version:
Adapter hardware: CC2531
Adapter firmware version:CC2531_DEFAULT_20190608

Koenkk, 2020-06-14:
Can you provide a debug log where you reproduce the issue just after starting Zigbee2mqtt ?(provide the complete log from start)

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

Hydci, 2020-06-14:
Hello @Koenkk ,
Here is the log file as you asked me with the bug reproduced debug activate
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4776337/log.txt)

Koenkk, 2020-06-14:
Fixed in latest dev branch, can you confirm? (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Hydci, 2020-06-14:
Actually I confirm after restarting the on / off works perfectly by cons at the time of the brightness value ": 254 turns on the light fully off the light turns on with the old percentage before restarting I hope I have successfully explained my english and very bad

On after reboot
```juin 14 17:33:00 raspberrypi sudo[529]: zigbee2mqtt:info  2020-06-14 17:33:00: MQTT publish: topic 'zigbee2mqtt/0x000b57fffe910132', payload '{"state":"ON","brightness":254,"update_available":false}'
```
On before reboot
```juin 14 17:39:11 raspberrypi sudo[529]: zigbee2mqtt:info  2020-06-14 17:39:11: MQTT publish: topic 'zigbee2mqtt/0x000b57fffe8da195', payload '{"state":"ON","brightness":53,"update_available":false}'
```

it would be nice if the value that we put before the restart is retained after this one

Koenkk, 2020-06-14:
Also fixed that, assuming this can be closed now.

closed 2020-06-14 by Koenkk

Hydci, 2020-06-14:
Everything works perfectly as before thank you

Hydci, 2020-06-14:
I open the ticket again after some AQARA test and IKEA retains the percentage after the restart as indicated in my last message, however the HUE range loses the retention of the percentage can you fix that?