Koenkk/zigbee2mqtt #3738: IKEA LED1624G9 incorrect color temperature after color change

keteflips, 2020-06-14:
# Bug Report

I have HomeAssistant with CC2531.
Originaly the IKEA LED1624G9 has 2700k color temp with full 600 lumens, but when you change the color in HomeAssistant and get back to white, the temp color its like 4000k with less lumens.
To reset to the original state I have to off/on with the wall switch.

## Debug Info
Zigbee2mqtt version: 1.14.0
Adapter hardware: CC2531
Adapter firmware version: 1.14.0

Koenkk, 2020-06-14:
You should be able to figure out the default x/y color by sending a `{"color":""}` to `zigbee2mqtt/FRIENDLY_NAME_OF_BULB/get`.

keteflips, 2020-06-14:
> {"color":""}

I get this:

{"state":"ON","update_available":true,"brightness":254,"color":{"x":0.458,"y":0.41},"linkquality":68}

MattWestb, 2020-07-19:
True update the bulb firmware.
Its our old RGBWW and its only LL but its having getting one "refresh"  in the last upgrade pack.
"Release Version: 1.13.0
29th June 2020
WS2.0 OPAL (V-2.0.029).
CWS1.0 (1.3.013).
Fixed On Off State after OTA Upgrade."
The subversion its ticked more then one so probably more fixes then being written in the release notes.

ChankyTheConqueror, 2020-08-19:
Hi,
Thank you for supporting this wonderful bulb :)

Im facing further problems with the OTA upgrade to CWS1.0 (1.3.013). I don't know if it's an issue or intentional this way. If i set the color and the brightness, the settings are retain after a power off cycle (mechanically on off on), it's great but also annoying that its no longer possible to fallback to the default colour and brightness settings via toggling the mechanical switch. Can this be overriden? Thanks

chris1705, 2020-09-11:
I have experienced the same issue. The color looks like the bulb would ignore the white LED and set all RGB LEDs to full brightness instead. I have a Conbee II and CC2531. Is there any way I could help debugging the issue? When turning the bulb on it seems to report it's latest state before turning off.

closed 2020-10-19 by github-actions[bot]

michal-ambroziak, 2021-11-09:
Has anyone been lucky enough to fix this problem? I'm using latest version of Home Assistant 2021.11.2 and Zigbee2mqtt 1.22.0 and still not able to return to "opal white" after choosing color. My workaround was to physically turn off/on the bulb, but now for one of them, even that doesn't help.

ov1d1u, 2021-11-11:
@michal-ambroziak To possible workarounds for this:

1. Set the color to xy 0.458, 0.458.
2. Reset the bulb and, after it's back to the default color, [store a scene then recall it every time you need it](https://www.zigbee2mqtt.io/guide/usage/scenes.html).

michal-ambroziak, 2021-11-12:
thank you @ov1d1u

cooperaj, 2022-01-03:
I've just encountered this after re-adding my bulb. When pairing this is thrown in the logs 

```
Failed to read state of '0x90fd9ffffee780e1' after reconnect (Read 0x90fd9ffffee780e1/1 lightingColorCtrl(["colorMode","currentX","currentY","colorTemperature"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE'))```

Clicking the refresh icon on the colour temperature gives this error

```
Publish 'get' 'color_temp' to 'livingroom_lamp' failed: 'Error: Read 0x90fd9ffffee780e1/1 lightingColorCtrl(["colorMode","currentX","currentY","colorTemperature"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'```

Semmu, 2023-06-07:
I have the same issue.

Basically when the lightbulb is turned on for the first time, it seems all its LEDs are on (the white(s?) and the colored ones as well, so everything it has), thus giving full brightness, which is indeed really bright.

But when some custom color is set, there is no way to reset it to the original "full brightness" mode which has all the LEDs shining.

I did an OTA update on my device (for the first time since I bought it years ago), because I thought it would fix the issue, but now unfortunately I lost its "full brightness" mode, since it seems to remember the last set color when losing electricity and turns on glowing in that color.

I think this is a software design flaw by IKEA, but it's a shame we can't tell a smartbulb to simply give full power to all its LEDs...