Koenkk/zigbee2mqtt #3740: Devices unavailable on Home Assistant restart

toxuin, 2020-06-14:
# Bug Report
## What happened

Upon restarting Home Assistant, devices' availability is not updated.

Home Assistant has `will_message` and `birth_message` set and I can see those being sent and received in zigbee2mqtt logs too.

Here's a snippet from Home Assistant's config:
```yaml
mqtt:
  ...
  discovery: true
  birth_message:
    topic: 'hass/status'
    payload: 'online'
  will_message:
    topic: 'hass/status'
    payload: 'offline'
```

Here is my [zigbee2mqtt config](https://hasteb.in/ifanohod.yaml)

Here are [the logs](https://hasteb.in/ihexabuk.log) with debug enabled.

I've been able to trace this to the [related code](https://github.com/Koenkk/zigbee2mqtt/blob/9009de24aca1027283bfbfbe399f9071254184db/lib/extension/homeassistant.js#L2068), but it does not seem like there is anything there that would re-send availability information (this, or I have not enough knowledge on the zigbee2mqtt architecture).

Please tell me what am I doing wrong as something as simple as this would probably catch attention of wise humans before me. Any help would be appreciated.

## What did you expect to happen

When Home Assistant comes back online, zigbee2mqtt would re-publish the availability state of devices.

## How to reproduce it

- start homeassistant
- start zigbee2mqtt
- wait for devices to become available in HA
- In HA, click "restart Home Assistant"
- watch home assistant publish `online` to `hass/status` and count to 30
- watch zigbee2mqtt logs to see re-publish devices' state (light levels and on/off state), but not availability
- in HA, zigbee devices are unavailable

## Debug Info
Zigbee2mqtt version: docker image `latest-dev`, `imagePullPolicy: always`
Adapter hardware: CC1352P2 over tcp socket on esp8266 (most likely irrelevant)
Adapter firmware version: `20200417`

Koenkk, 2020-06-14:
Availability message are send with the retained flag and therefore don't have to be send again.

@emontnemery is there a way to debug received Home Assistant MQTT messages?

emontnemery, 2020-06-14:
Sure (example from a Tasmota device):
[image]
[image]

toxuin, 2020-06-14:
I can see HA receiving the messages with `retain: true`, and received instantly after I subscribe to it (= broker's retain is working fine):

[image]

But in HA, after a restart, my devices are still greyed out:
[image]
Weirdly enough, when I click "MQTT INFO" for any zigbee device in HA, I get an empty list. Tasmota devices, on the other hand (wifi-only, not tasmota-zigbee) are fine and have content in it (exactly as @emontnemery posted before).

Here's what zigbee2mqtt device's MQTT INFO card looks like:

[image]

It all goes back to normal when I restart the zigbee2mqtt pod.

Is it a bug with Home Assistant? Seems like it can't detect what device even is and what it's availability topics are.

And hey, thanks a lot for looking into this. I am kind of lost and exhausted all sane options (switched brokers, tried various zigbee2mqtt versions, reflashed coordinator and connected it directly). I really appreciate your support!

emontnemery, 2020-06-14:
Can you attach a home assistant log including all steps needed to reproduce (start + restart of HA) with MQTT debug enabled please? Have you configured client_id in HAs MQTT settings?

toxuin, 2020-06-14:
Yes, `client_id` is set to be `homeassistant-core` in Home Assistant configuration.

Here is the very verbose [log of home assistant](https://gist.github.com/toxuin/6ea942000ff12da55d061de91cfa4680) with mqtt in debug mode. I tried to annotate it to the best of my ability.

Thanks!

emontnemery, 2020-06-14:
The problem is not the availability, the problem is that the zigbee2mqtt devices aren't discovered until you restart zigbee2mqtt.

Tasmota retains the discovery messages, I don't think zigbee2mqtt does that, instead you need to setup zigbee2mqtt to send discovery messages when HA starts.

@Koenkk can you confirm this is the case?

toxuin, 2020-06-14:
yes, I can see zigbee2mqtt pick up the HA's birth message. You can see it in the [logs](https://hasteb.in/ihexabuk.log) I've posted before:

` 2020-06-14 07:58:06: Received MQTT message on 'hass/status' with data 'online'`

Then, it proceeds to send the state messages.

emontnemery, 2020-06-14:
It's not enough to send the state messages, it needs to send the discovery messages like this:
```
2020-06-14 19:09:50 DEBUG (MainThread) [homeassistant.components.mqtt] Received message on homeassistant/sensor/0xd0cf5efffeda1b65/battery/config: b'{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee/On_Off_Remote","json_attributes_topic":"zigbee/On_Off_Remote","name":"On_Off_Remote_battery","unique_id":"0xd0cf5efffeda1b65_battery_zigbee","device":{"identifiers":["zigbee2mqtt_0xd0cf5efffeda1b65"],"name":"On_Off_Remote","sw_version":"Zigbee2mqtt 1.14.0-dev","model":"TRADFRI ON/OFF switch (E1743)","manufacturer":"IKEA"},"availability_topic":"zigbee/On_Off_Remote/availability"}'
2020-06-14 19:09:50 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: sensor 0xd0cf5efffeda1b65 battery
```

emontnemery, 2020-06-14:
Just to be sure, does the behavior change if you don't set `client_id` in HA's config?

toxuin, 2020-06-14:
Same behaviour when I comment out `client_id` in HA's config.

Here's what I see in [zigbee2mqtt log](https://gist.github.com/toxuin/98188f06bafa0a347ef6f9ef38ae5134) after 30 seconds after HA sends its' birth message.

emontnemery, 2020-06-15:
Right, so there's no log of discovery messages sent as a result of receiving the birth message from HA.

toxuin, 2020-06-15:
Is it something in my configuration or is it a bug in zigbee2mqtt? Thanks for looking into this.

emontnemery, 2020-06-15:
Something may be wrong in your configuration.

Tasmota publishes it's discovery messages with retain flag set, and based on the code, so does zigbee2mqtt (I thought it doesn't, my mistake): https://github.com/Koenkk/zigbee2mqtt/blob/9009de24aca1027283bfbfbe399f9071254184db/lib/extension/homeassistant.js#L2052

However, HA receives the discovery messages *without* the retain flag set.

toxuin, 2020-06-15:
When I subscribe to this topic from desktop client MQTT.fx (`homeassistant/sensor/0xd0cf5efffeda1b65/battery/config`) I get an instant message delivery with all the contents that indicate the message was sent with retained flag. I can also see this in Home Assistant:

[image]

Seems like zigbee2mqtt is indeed sending the discovery message properly, but Home Assistant is failing to pick it up.

emontnemery, 2020-06-15:
OK, and what if you subscribe to `homeassistant/#` ?

toxuin, 2020-06-15:
Nothing! Is that it? I don't get the retained messages when I subscribe to `homeassistant/#`.

Can it be my broker? I am using RabbitMQ with MQTT plugin with `rabbit_mqtt_retained_msg_store_ets` store.

emontnemery, 2020-06-15:
Right, so this issue can be closed then.
I'd suggest to use mosquitto for broker btw, it for sure won't let you down by not supporting basic functionality.

toxuin, 2020-06-15:
Thank you very much for looking into this!

closed 2020-06-15 by toxuin