Koenkk/zigbee2mqtt #3741: configuration.yaml edit problem

salopette, 2020-06-14:
Hello, I'm using the last version of z2m and have a little problem.

If I expand the configuration.yaml, z2m no longer starts, if I write back, it works again.
What am I doing wrong?

thats how it works:

`homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
serial:
  port: /dev/ttyACM0
devices:
  '0x00178801086412ee':
    friendly_name: '0x00178801086412ee'
groups: {}`

and this is not how it works:

`homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
serial:
  port: /dev/ttyACM0
devices:
  '0x00178801086412ee':
    friendly_name: '0x00178801086412ee'
groups: {}
advanced:
pan_id: 0x1b63
ext_pan_id: [0xDA, 0xDB, 0xDC, 0xDD, 0xDE, 0xDF, 0xDG, 0xDH]
channel: 20
cache_state: true
log_level: info
log_directory: data/log/%TIMESTAMP%
log_file: log.txt
log_rotation: true
log_output:
    - console
    - file
rtscts: true
soft_reset_timeout: 0
network_key: [1, 3, 5, 7, 9, 11, 13, 15, 0, 2, 4, 6, 8, 10, 12, 13]
availability_blacklist:
    - DEVICE_FRIENDLY_NAME or DEVICE_IEEE_ADDRESS
	availability_whitelist:
    - DEVICE_FRIENDLY_NAME or DEVICE_IEEE_ADDRESS
report: true
timestamp_format: 'YYYY-MM-DD HH:mm:ss'
experimental:
output: 'json'
map_options:
graphviz:
colors:
fill:
        enddevice: '#fff8ce'
        coordinator: '#e04e5d'
        router: '#4ea3e0'
font:
        coordinator: '#ffffff'
        router: '#ffffff'
        enddevice: '#000000'
line:
        active: '#009900'
        inactive: '#994444'
device_options: {}`

MPM1107, 2020-06-14:
Could you post your config file in a way that is a bit more readable? Like this it's impossible to see whether your formatting is correct or not. Preferably you should use a service like pastebin or hastebin to paste the config and then just post the link to the paste in here. From what I can see you don't have the required indentations in the yaml.

salopette, 2020-06-14:
https://pastebin.com/aQ7b6w7q

MPM1107, 2020-06-14:
As I suspected, your indentations are pretty messed up. Check this page carefully: [configuration](https://www.zigbee2mqtt.io/information/configuration.html) and look at the indentation levels (number of spaces before each option)

Just one example, but you have many errors like this one in your config, so make sure the check the whole file:

Your config:
```yaml
experimental:
output: 'json'
```

Should be:
```yaml
experimental:
  output: 'json'
```

salopette, 2020-06-14:
I thought spaces didn't matter.

closed 2020-07-26 by stale[bot]