Koenkk/zigbee2mqtt #3744: Brightness problem hue

Hydci, 2020-06-14:
I open a new issue to report the rest of my message the fix for the last message works for the Aqara and Ikea bulb range but the HUE range has the same problem as before the percentage is not retained after reboot
https://github.com/Koenkk/zigbee2mqtt/issues/3736#issuecomment-643783670

Koenkk, 2020-06-15:
I've tested with a Hue white bulb but it works fine for me. Can you provide debug logging?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

Hydci, 2020-06-15:
Hello @Koenkk ,
Here is the log file as requesting the test light was 0x0017880104a1d8be
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4781859/log.txt)
handling for the test:
reboot
We
set%
off
we
off

Koenkk, 2020-06-16:
I never see you setting `0x0017880104a1d8be` on in this log

closed 2020-07-26 by stale[bot]