Koenkk/zigbee2mqtt #3745: Some devices unavailable after Coordinator power off/on

funtik039, 2020-06-14:
## Problem
I have next devices in my test zigbee network:
Coordinator CC2538+CC2592
Aqara water leak sensor SJCGQ11LM x2
Mi power plug ZigBee ZNCZ02LM
Aqara wireless relay controller LLKZMK11LM

Hassio  on RPI 4b.

When i power off and than power on Coordinator, 2 devices (ZNCZ02LM and LLKZMK11LM) go to unavailable state. Coordinator cant send commands to them and they cant send massages to coordinator. Need to repair they again.
2 other devices (SJCGQ11LM) works prefect. 
In logs after reconect coordinator i see errors:
`zigbee2mqtt:error 2020-06-14 20:48:12: Failed to execute LQI for '0x00158d000426fd08'
 zigbee2mqtt:error 2020-06-14 20:48:22: Failed to execute LQI for '0x00158d00035f2d3a'`
When i try to chage state of any Zigbee switch i got next error:
`zigbee2mqtt:error 2020-06-14 20:48:43: Publish 'set' 'state' to '0x00158d00035f2d3a' failed: 'Error: Command 0x00158d00035f2d3a/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Data request failed with error: 'MAC no ack' (233))'`

[Full dump of z2m folder](https://drive.google.com/drive/folders/1vuVdISP7SowiEgpNFQ4kLqweFZW0iuJn?usp=sharing)

## How to reproduce it (minimal and precise)
Power off and power on Coordinator. Try to change state of devices ZNCZ02LM and LLKZMK11LM.

## Debug Info
Zigbee2mqtt version: 1.14.0
Adapter hardware: CC2538+CC2592
Adapter firmware version: 20200211

### P.S.
Sorry for my bad English :P

elbueno222, 2020-06-17:
I don't have that adapter, but the latest firmware version for that adapter is 20200327, if you are using an old version (20200211), updating the firmware could fix the issue:

https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.0.x/bin/CC2538_CC2592_MODKAMRU_V3_20200327.zip

funtik039, 2020-07-08:
Have solution. Works fine on 23052019 firmware.

closed 2020-07-08 by funtik039