Koenkk/zigbee2mqtt #3746: GledOpto GL-D-004Z unsupported

markkNL, 2020-06-14:
Hello,

I have issues adding an unsupported device. There is support for GL-D-003Z, GL-D-005Z, and GL-D-004ZS, but not for GL-D-004Z. When I check the 'add unsupported device' page I get stuck at step 2.

Not sure if this is of any help but this is the small piece of the logging when trying to pair:
`zigbee2mqtt:warn  2020-06-14 16:49:32: Received message from unsupported device with Zigbee model 'GL-D-004Z'
zigbee2mqtt:warn  2020-06-14 16:49:32: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
zigbee2mqtt:info  2020-06-14 16:49:32: Successfully interviewed '0x00124b001bd89d07', device has successfully been paired
zigbee2mqtt:warn  2020-06-14 16:49:32: Device '0x00124b001bd89d07' with Zigbee model 'GL-D-004Z' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
zigbee2mqtt:info  2020-06-14 16:49:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x00124b001bd89d07","supported":false}}'
`

I'm not really familiar with all this stuff so if I should add other information please let me know. Thanks :)

MPM1107, 2020-06-14:
Where exactly did you get stuck on step 2? Are you able to find and edit the devices.js file as the instructions describe?

Heronimonimo, 2020-06-15:
Made a pull request for support: https://github.com/Koenkk/zigbee-herdsman-converters/pull/1316

Koenkk, 2020-06-15:
Merged, thanks @Heronimonimo . Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-06-15 by Koenkk

markkNL, 2020-06-15:
> Where exactly did you get stuck on step 2? Are you able to find and edit the devices.js file as the instructions describe?

Hi, I cannot find the file that is referenced. Somehow I cannot get to the correct folder using the configurator in Home Assistant. But I guess it is already fixed? I'll try to switch to the dev branch next time, thanks :)

Heronimonimo, 2020-06-15:
Please try the dev branch and see if it works properly.

markkNL, 2020-06-15:
> Please try the dev branch and see if it works properly.

Yes, great, I can now operate the light. Thank you all.
I did see that the 3Z and 5Z also have a ZS version (like this 4Z and ZS), could those maybe also be added? It seems like they are just a newer version.

Heronimonimo, 2020-06-16:
Good point.

@Koenkk: would you like them all as separate devices in devices.js? Or is there a way all these similar types with the same config can be put into one device descriptor?

Koenkk, 2020-06-16:
@Heronimonimo I think we shouldn't add devices based on a guessed `zigbeeModel` (because nobody tested if it actually worked).