Koenkk/zigbee2mqtt #3748: Timeout when trying to toggle state of zigbee devices

estebanz01, 2020-06-14:
# Bug Report
## What happened
Two specific devices (A806S-Q1R & PSM-29ZBSR) are having issues and I cannot keep their current state via HA or MQTT when trying to turn ON/OFF the device or changing the brightness, here's what I have in the logs:

Power Plug (PSM-29ZBSR):
```javascript
Jun 14 22:03:37 raspberrypi npm[4268]: zigbee2mqtt:error 2020-06-14 22:03:37: Publish 'set' 'state' to '0x00124b000554125f' failed: 'Error: Command 0x00124b000554125f/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Timeout - 35217 - 1 - 26 - 6 - 11 after 10000ms)'
Jun 14 22:03:37 raspberrypi npm[4268]: zigbee2mqtt:info  2020-06-14 22:03:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0x00124b000554125f' failed: 'Error: Command 0x00124b000554125f/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Error: Timeout - 35217 - 1 - 26 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"0x00124b000554125f"}}'
```

Dimmable Light (A806S-Q1R):
```javascript
Jun 14 22:14:16 raspberrypi npm[4268]: zigbee2mqtt:error 2020-06-14 22:14:16: Publish 'set' 'state' to '0x00158d0000cf7839' failed: 'Error: Command 0x00158d0000cf7839/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Timeout - 26333 - 1 - 30 - 6 - 11 after 10000ms)'
Jun 14 22:14:16 raspberrypi npm[4268]: zigbee2mqtt:info  2020-06-14 22:14:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0x00158d0000cf7839' failed: 'Error: Command 0x00158d0000cf7839/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Error: Timeout - 26333 - 1 - 30 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"0x00158d0000cf7839"}}'

# Brightness change
Jun 14 22:16:17 raspberrypi npm[4268]: zigbee2mqtt:error 2020-06-14 22:16:17: Publish 'set' 'brightness' to '0x00158d0000cf7839' failed: 'Error: Command 0x00158d0000cf7839/1 genLevelCtrl.moveToLevelWithOnOff({"level":167,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Timeout - 26333 - 1 - 31 - 8 - 11 after 10000ms)'
Jun 14 22:16:17 raspberrypi npm[4268]: zigbee2mqtt:info  2020-06-14 22:16:17: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'brightness' to '0x00158d0000cf7839' failed: 'Error: Command 0x00158d0000cf7839/1 genLevelCtrl.moveToLevelWithOnOff({\"level\":167,\"transtime\":0}, {\"timeout\":10000,\"disableResponse\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Error: Timeout - 26333 - 1 - 31 - 8 - 11 after 10000ms)'","meta":{"friendly_name":"0x00158d0000cf7839"}}'
```

let's say I have my light or power plug in a `OFF` state, if I try to turn them on, the light starts lighting and the power plug starts giving power, but HA or MQTT does not retain the expected state and I cannot toggle the device anymore.

## What did you expect to happen
I can toggle and see the current state of my devices.

## How to reproduce it (minimal and precise)
Connect both devices to the zigbee network, sync them into HA with MQTT integration and try to toggle the state.

## Debug Info
Zigbee2mqtt version: `HEAD detached at 1eed937`
Adapter hardware: CC2531
Adapter firmware version: Z-Stack HA 1.2 - 20190619

estebanz01, 2020-06-14:
I have HA linked to google home, so I tried to turn them on and off. Here's what I happens when I say turn on the light:

```javascript
Jun 14 22:23:53 raspberrypi npm[4268]: zigbee2mqtt:info  2020-06-14 22:23:53: MQTT publish: topic 'zigbee2mqtt/0x00124b000554125f', payload '{"state":"ON","linkquality":49,"power":0,"energy":0}'
Jun 14 22:24:13 raspberrypi npm[4268]: zigbee2mqtt:info  2020-06-14 22:24:13: MQTT publish: topic 'zigbee2mqtt/0x00124b000554125f', payload '{"state":"ON","linkquality":28,"power":0,"energy":0}'
```

Which turns ON as expected. And here's what happen when I try to turn it OFF
```javascript
Jun 14 22:24:30 raspberrypi npm[4268]: zigbee2mqtt:error 2020-06-14 22:24:30: Publish 'set' 'state' to '0x00158d0000cf7839' failed: 'Error: Command 0x00158d0000cf7839/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Timeout - 26333 - 1 - 41 - 6 - 11 after 10000ms)'
Jun 14 22:24:30 raspberrypi npm[4268]: zigbee2mqtt:info  2020-06-14 22:24:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0x00158d0000cf7839' failed: 'Error: Command 0x00158d0000cf7839/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Error: Timeout - 26333 - 1 - 41 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"0x00158d0000cf7839"}}'
Jun 14 22:24:53 raspberrypi npm[4268]: zigbee2mqtt:error 2020-06-14 22:24:53: Publish 'set' 'state' to '0x00158d0000cf7839' failed: 'Error: Command 0x00158d0000cf7839/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Timeout - 26333 - 1 - 42 - 6 - 11 after 10000ms)'
Jun 14 22:24:53 raspberrypi npm[4268]: zigbee2mqtt:info  2020-06-14 22:24:53: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0x00158d0000cf7839' failed: 'Error: Command 0x00158d0000cf7839/1 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Error: Timeout - 26333 - 1 - 42 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"0x00158d0000cf7839"}}'
```

which also turns OFF the light 🤔 . Weird, isn't ?

estebanz01, 2020-06-14:
for the power plug it fails, but it reports the later the state as expected:
```javascript
Jun 14 22:29:11 raspberrypi npm[4268]: zigbee2mqtt:error 2020-06-14 22:29:11: Publish 'set' 'state' to '0x00124b000554125f' failed: 'Error: Command 0x00124b000554125f/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Timeout - 35217 - 1 - 47 - 6 - 11 after 10000ms)'
Jun 14 22:29:11 raspberrypi npm[4268]: zigbee2mqtt:info  2020-06-14 22:29:11: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to '0x00124b000554125f' failed: 'Error: Command 0x00124b000554125f/1 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Error: Timeout - 35217 - 1 - 47 - 6 - 11 after 10000ms)'","meta":{"friendly_name":"0x00124b000554125f"}}'
Jun 14 22:29:25 raspberrypi npm[4268]: zigbee2mqtt:info  2020-06-14 22:29:25: MQTT publish: topic 'zigbee2mqtt/0x00124b000554125f', payload '{"state":"OFF","linkquality":49,"power":0,"energy":0}'
Jun 14 22:29:25 raspberrypi npm[4268]: zigbee2mqtt:info  2020-06-14 22:29:25: MQTT publish: topic 'zigbee2mqtt/0x00124b000554125f', payload '{"state":"OFF","linkquality":49,"power":0,"energy":0}'
```

Koenkk, 2020-06-15:
This error happens when the device does not respond to the coordinator. Could it be that these devices are out of range?

estebanz01, 2020-06-15:
I don't think so. If that would be the case, then I couldn't control my devices via Google home, no? Is it possible that the devices might be sending a response via a router and one of the routers is dropping the message?

estebanz01, 2020-06-15:
Also, I have two devices just 4 mts away from the ones with problems and they work as expected. Worth to mention that those two devices that work normally are shown as routers in ZigBee2MQTT

Koenkk, 2020-06-16:
> I don't think so. If that would be the case, then I couldn't control my devices via Google home, no? Is it possible that the devices might be sending a response via a router and one of the routers is dropping the message?

Only way to figure this out is by sniffing: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

estebanz01, 2020-06-16:
I'll give it a shot then.

estebanz01, 2020-06-17:
I couldn't understand what was happening, so I re-flashed my CC2531 coordinator and I did a complete re-pair of all devices. After a lot of trial and error I found out that the issues were caused by the power plug router. For some reason, it cannot be placed in relative close distance to another router. I moved the power plug to another room where no zigbee devices are present and that fixed the entire problem.

Also, I had some issues with the climax temperature sensor, but was because of this:
> An end device only has one parent, either the coordinator or a router, generally the closest device when it was paired.

The parent was a lightbulb in another room and it seems that the signal is not strong enough in the place where it's located, so it was timing out until it picked up the closest router.

Fun ride!

closed 2020-06-17 by estebanz01

darek-margas, 2023-11-02:
Just installed z2m, moved toys over from ZHA and got the same:

Zigbee2MQTT:error 2023-11-02 20:50:05: Publish 'set' 'state' to 'L1 Hall Socket' failed: 'Error: Command 0x08230211e00bfdb0/1 genOnOff.on({}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 36914 - 1 - 25 - 6 - 11 after 10000ms)'

These used to work with ZHA, now pretty much read-only as status is updated to whatever it is but z2m is unable to toggle.

VincentSC, 2024-11-09:
For me it was a bad USB-cable