Koenkk/zigbee2mqtt #3749: NEO Sirene / Temp / Humi sensors (0yu2xgi)

jvrrr, 2020-06-15:
Hi,
I have bought a Neo Sirene:
https://szneo.com/en/products/show.php?id=241

Features are:
-Sirene (multiple selectable tunes)
-Temp
-Humidity

Is there a way to make this work?
I like the selectable tunes...use it as door bell and alarm.

I have read this https://github.com/Koenkk/zigbee2mqtt/issues/2956
But I think it is never been finished, because my device is not reconized.

I like to help out, so what do I need to check?

MPM1107, 2020-06-15:
Yes nothing from that conversation you linked was ever actually implemented. You will have to edit your devices.js file with the code from that issue to start playing around with it. This can help you with that part: [supporting new devices](https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html)
Your zigbee2mqtt log should be saying something about an unusupported device with a specific "model id". What model id does it show you?

jvrrr, 2020-06-15:
I add this to devices.js:
{
    	zigbeeModel: ['0yu2xgi\u0000'],
    	model: 'neo_sirene',
    	vendor: 'neo',
    	description: 'Temperature & humidity sensor',
    	supports: 'temperature and humidity',
    	fromZigbee: [fz.neo_battery, fz.neo_temperature],
    	toZigbee: [tz.warning],
    },

Now I get this:

jun 16 00:15:49 zigbee npm[11310]:     at resolvedEntity.definition.fromZigbee.filter (/opt/zigbee2mqtt/lib/extension/receive.js:120:42)
jun 16 00:15:49 zigbee npm[11310]:     at Array.filter (<anonymous>)
jun 16 00:15:49 zigbee npm[11310]:     at Receive.onZigbeeEvent (/opt/zigbee2mqtt/lib/extension/receive.js:119:65)
jun 16 00:15:49 zigbee npm[11310]:     at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:352:44)
jun 16 00:15:49 zigbee npm[11310]:     at Controller.onZigbeeEvent (/opt/zigbee2mqtt/lib/controller.js:238:14)
jun 16 00:15:49 zigbee npm[11310]:     at Zigbee.emit (events.js:198:13)
jun 16 00:15:49 zigbee npm[11310]:     at Controller.herdsman.on (/opt/zigbee2mqtt/lib/zigbee.js:67:52)
jun 16 00:15:49 zigbee npm[11310]:     at Controller.emit (events.js:198:13)
jun 16 00:15:49 zigbee npm[11310]:     at Controller.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/controller.js:496:22)

MPM1107, 2020-06-15:
You cut off the important part of the error there but I'm guessing that you are missing the converter definitions in the toZigbee.js and fromZigbee.js files

jvrrr, 2020-06-16:
Thanks!!

I add:
neo_temperature: {
    cluster: 'manuSpecificTuyaDimmer',
    type: 'raw',
    convert: (model, msg, publish, options) => {
      const temperature = parseFloat(msg.data['measuredValue']) / 100.0;
      return {temperature: temperature};
    },
},
neo_battery: {
    cluster: 'manuSpecificTuyaDimmer',
    type: ['raw'],
    convert: (model, msg, publish, options, meta) => {
        if (msg.data.hasOwnProperty('batteryPercentageRemaining')) {
            return {battery: msg.data['batteryPercentageRemaining']};
        }
    },
},

jvrrr, 2020-06-16:
This is the only output I get:
{"ieeeAddr":"0xec1bbdfffe6c1edf","networkAddress":19143,"vendor":"-","description":"-","friendly_name":"0xec1bbdfffe6c1edf","lastSeen":1592335185459}]}'
jun 16 21:19:45 zigbee npm[19883]: zigbee2mqtt:info  2020-06-16 21:19:45: MQTT publish: topic 'domoticz/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0xec1bbdfffe6c1edf"}}'

jvrrr, 2020-06-16:
How can I check if everything is working fine?

closed 2020-07-26 by stale[bot]

kirovilya, 2020-08-04:
I am started https://github.com/kirovilya/zigbee-herdsman-converters/commit/ca19123a63dff1847a6c27f86b1b8e1f4c525a59

jvrrr, 2020-08-05:
Great news!! 
I like to test It out.
How can I test it?

kirovilya, 2020-08-06:
try dev branch of z2m

juliend20, 2020-08-11:
Great news!! 
I'll try your dev branch tomorrow. 
How can I set alarm on/off?
Thank you.

kirovilya, 2020-08-11:
you can to use dev version of z2m

Julienhug, 2020-08-12:
I tested today. It's work fine. I receive info from siren. 
`{"linkquality":86,"alarm":false,"temperature_alarm":false,"humidity_alarm":false,"melody":6,"volume":"high","temperature_min":18,"temperature_max":27,"humidity_min":45,"humidity_max":80,"humidity":59,"temperature":"24.1"}`

The temperature is weird because I have 26°C on another sensor.

Thank you so much for your work. How did you know the information to be send to the siren? did you have a tuya gateway and you have sniff the zigbee protocol?

kirovilya, 2020-08-12:
yes, I have a tuya gateway and I sniffed device communication with the gateway

Julienhug, 2020-08-12:
ok great thanks. Have you a wrong value of temperature too?

jvrrr, 2020-08-12:
I have the same issue  it is 29°C and is shows 27°C...so it is a issue.

Do you also have the issue is spontaneous sounds a melody?

jvrrr, 2020-08-12:
Thanks for great job!!

Julienhug, 2020-08-13:
@jvrrr I didn't find any problem with melody. But Can you be more precise? I'm not sure I understand meaning of "spontaneous sounds".

Julienhug, 2020-08-13:
It has been released https://www.zigbee2mqtt.io/devices/NAS-AB02B0.html We can use temperature calibration to add 2°C

Julienhug, 2020-08-13:
@kirovilya Hello I saw in documentation that we can add temperature_calibration. So I tested with temperature_calibration: 2 but It doesn't work and I didn't find anything about  temperature_calibration in fromZigbee.js

kirovilya, 2020-08-13:
@Julienhug Sorry, I didn’t write the documentation for the device and I don’t know why the calibration option is indicated there.

jvrrr, 2020-08-13:
> @jvrrr I didn't find any problem with melody. But Can you be more precise? I'm not sure I understand meaning of "spontaneous sounds".

It spontaneouse plays a melody...I'm not sure why.

Julienhug, 2020-08-14:
> 
> 
> @Julienhug Sorry, I didn’t write the documentation for the device and I don’t know why the calibration option is indicated there.

Ok, it'll be nice to have this option. Because I have an error on the temperature value. I'd like to add 2°C on it

Julienhug, 2020-08-14:
> 
> 
> > @jvrrr I didn't find any problem with melody. But Can you be more precise? I'm not sure I understand meaning of "spontaneous sounds".
> 
> It spontaneouse plays a melody...I'm not sure why.

Ok it's very weird. Maybe you can launch zigbee mqtt with logs. To ensure that it's not something that trigger the alarm by mistake

kirovilya, 2020-08-14:
The siren may be triggered by temperature or humidity. Check the relevant options

huyphamnhu, 2020-08-16:
@kirovilya Hi, could you tell me how to manually trigger the alarm?

kirovilya, 2020-08-17:
@huyphamnhu set alarm to true
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/toZigbee.js#L2601

pedroke, 2020-08-19:
Hi guys, do you know if it is possible to change the sound somehow? Btw, thanks Ilya for another great and affordable device supported!

kirovilya, 2020-08-19:
@pedroke set melody from 1 to 18

pedroke, 2020-08-19:
:) thanks... super!

bochelork, 2020-09-16:
Sorry for steeling a closed topic, but I have a small issue. 
When I post:
{
  "melody": 6,
  "alarm": true
}
only the alarm is activated. the melody does not change. 

If I only post a melody, the melody is set. so I need 2 separate commands. Can this be solved?

```
Zigbee2MQTT:debug 2020-09-16 19:14:41: Received MQTT message on 'zigbee2mqtt/ALARM1/set' with data
'{
  "melody": 6,
  "alarm": true
}'
Zigbee2MQTT:debug 2020-09-16 19:14:41: Publishing 'set' 'alarm' to 'ALARM1'
Zigbee2MQTT:debug 2020-09-16 19:14:43: Received Zigbee message from 'ALARM1', type 'commandSetDataResponse', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":[1],"type":"Buffer"},"dp":360,"fn":0,"status":0,"transid":4}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-09-16 19:14:43: MQTT publish: topic 'zigbee2mqtt/ALARM1', payload '{"alarm":true,"humidity":63,"last_seen":"2020-09-16T17:14:43.643Z","linkquality":48,"melody":4,"temperature":"21.1","volume":"high"}'
Zigbee2MQTT:debug 2020-09-16 19:14:53: Received Zigbee message from 'ALARM1', type 'commandSetDataResponse', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":[0,0,0,212],"type":"Buffer"},"dp":617,"fn":0,"status":0,"transid":2}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-09-16 19:14:53: MQTT publish: topic 'zigbee2mqtt/ALARM1', payload '{"alarm":true,"humidity":63,"last_seen":"2020-09-16T17:14:53.223Z","linkquality":36,"melody":4,"temperature":"21.2","volume":"high"}'
Zigbee2MQTT:debug 2020-09-16 19:14:54: Received Zigbee message from 'ALARM1', type 'commandSetDataResponse', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":[0],"type":"Buffer"},"dp":360,"fn":0,"status":0,"transid":4}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-09-16 19:14:54: MQTT publish: topic 'zigbee2mqtt/ALARM1', payload '{"alarm":false,"humidity":63,"last_seen":"2020-09-16T17:14:54.317Z","linkquality":36,"melody":4,"temperature":"21.2","volume":"high"}'

Zigbee2MQTT:debug 2020-09-16 19:14:41: Received MQTT message on 'zigbee2mqtt/ALARM1/set' with data
'{
  "melody": 5
}
Zigbee2MQTT:debug 2020-09-16 19:18:21: Received MQTT message on 'zigbee2mqtt/ALARM1/set' with data '{"melody":5}'
Zigbee2MQTT:debug 2020-09-16 19:18:21: Publishing 'set' 'melody' to 'ALARM1'
Zigbee2MQTT:debug 2020-09-16 19:18:24: Received Zigbee message from 'ALARM1', type 'commandSetDataResponse', cluster 'manuSpecificTuyaDimmer', data '{"data":{"data":[5],"type":"Buffer"},"dp":1126,"fn":0,"status":0,"transid":2}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-09-16 19:18:24: MQTT publish: topic 'zigbee2mqtt/ALARM1', payload '{"alarm":false,"humidity":63,"last_seen":"2020-09-16T17:18:24.174Z","linkquality":36,"melody":5,"temperature":"21.1","volume":"high"}'
```

misteriks, 2020-09-25:
> The siren may be triggered by temperature or humidity. Check the relevant options

The melody is indeed triggered by the humidity or temperature value min-max range

Zigbee2MQTT:info  2020-09-25 08:58:46: MQTT publish: topic 'zigbee2mqtt/Sirene', payload '{"alarm":false,"humidity":51,"**humidity_alarm":true,"**humidity_max":80,"humidity_min":45,"linkquality":60,"melody":5,"temperature":"18.2","**temperature_alarm":true**,"temperature_max":27,"temperature_min":18,"volume":"high"}'

Guess this are the default values for this device. 
Is there a way to change these values on the device so the get published to HA on next discovery?

misteriks, 2020-09-25:
Ok, to disable the melody on temperature and humidity publish the flowing via the mqtt.publish service

Service Data:
topic: 'zigbee2mqtt/Sirene/set'
retain: true
payload: '{"temperature_alarm":false}'

or '{"humidity_alarm":false}'

I also found a small bug with it. Sending a value for humidity_max in the payload actually changed the humidity_min and vise versa.

misteriks, 2020-09-26:
> Sorry for steeling a closed topic, but I have a small issue.
> When I post:
> {
> "melody": 6,
> "alarm": true
> }
> only the alarm is activated. the melody does not change.
>
> If I only post a melody, the melody [...]

Were you able to find a solution for this? I'm having the same issue

bochelork, 2020-09-28:
> Were you able to find a solution for this? I'm having the same issue

I just set the alarm to true after the change of melody is detected.

arduino43, 2020-10-07:
I recently purchased two of these devices but cannot get them to work.

Are their any specific pairing instructions? Holding the reset button for 5 seconds shows the unit blinking but no sign of connection attempts in logs.

1.) I've tried 1.15 and 15.0-dev
2.) Nothing appears in the logs, the device does not appear to attempt pairing.
3.) I've tried pairing with the unit powered by batteries only, USB only and batteries+USB.

Peet1986, 2022-01-30:
Hey guys, i am new in zigbee2mqtt and maybe you can help me how to add this step by step, i have bought 2 of the NEO Sirene's and they only shown as unsupported. Where i have to enter the config? Where to go in Menue? :(  Thanks!
[image]