Koenkk/zigbee2mqtt #3752: No effect on white Channel GL-C-007_GL-C-008

rafal83, 2020-06-16:
# Bug Report
## What happened
Can't change white channel, set white_value at 0, white say on... set value > 0 => no effect
set any value has no effect

## Log
zigbee2mqtt:debug 2020-06-16 06:22:49: Received MQTT message on 'zigbee2mqtt/gledopto Controller/set' with data '{"state": "ON", "white_value": 229}'                                                                                             
zigbee2mqtt:debug 2020-06-16 06:22:49: Publishing 'set' 'state' to 'gledopto Controller'                                 
zigbee2mqtt:info  2020-06-16 06:22:49: MQTT publish: topic 'zigbee2mqtt/gledopto Controller', payload '{"state":"ON","brightness":255,"color":{"x":0.508,"y":0.239},"color_temp":37,"white_value":-1}'                                            
zigbee2mqtt:debug 2020-06-16 06:22:49: Publishing 'set' 'white_value' to 'gledopto Controller'                           
zigbee2mqtt:info  2020-06-16 06:22:50: MQTT publish: topic 'zigbee2mqtt/gledopto Controller', payload '{"state":"ON","brightness":229,"color":{"x":0.323,"y":0.329},"color_temp":37,"white_value":229}' 

## Debug Info
Version of Zigbee2Mqtt: 1.13.1
Coordinator version: 20190608
Coordinator Zigbee Id: 0x00124b001949bca4
Adapter hardware: CC2531

Koenkk, 2020-06-16:
- Can it be controller with `color_temp`?
- Can you share your `devices.js`?
- Is this an color + color (RGB+CCT) temperature OR color + white (RGBW) controller?

rafal83, 2020-06-17:
> * Can it be controller with `color_temp`?
> * Can you share your `devices.js`?
> * Is this an color + color (RGB+CCT) temperature OR color + white (RGBW) controller?

* `color_temp` change RGB led
* where is this file ?
* color + white (RGBW) controller

Koenkk, 2020-06-17:
sorry I mean  `database.db` entry of the device (can be found under `data/database.db`)

rafal83, 2020-06-17:
[database.txt](https://github.com/Koenkk/zigbee2mqtt/files/4790971/database.txt)

rafal83, 2020-06-19:
now white is off, but when I send payload, I got this error

```
zigbee2mqtt:error 2020-06-19 12:24:29: Publish 'set' 'state' to 'Gledopto Controller' failed: 'TypeError: Cannot read property 'construc
tor' of undefined'                                                                                                                      
zigbee2mqtt:info  2020-06-19 12:24:29: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"
Publish 'set' 'state' to 'Gledopto Controller' failed: 'TypeError: Cannot read property 'constructor' of undefined'","meta":{"friendly_n
ame":"Gledopto Controller"}}'  
```

rafal83, 2020-06-19:
it works with this change

from
```
        endpoint: (device) => {
            if (device.getEndpoint(10) && device.getEndpoint(11) && device.getEndpoint(13)) {
                return {rgb: 11, white: 15};
            } else {
                return {rgb: 11, white: 15};
            }
        },
```
to
```
        endpoint: (device) => {
            if (device.getEndpoint(10) && device.getEndpoint(11) && device.getEndpoint(13)) {
                return {rgb: 11, white: 10};
            } else {
                return {rgb: 11, white: 15};
            }
        },
```

Koenkk, 2020-06-20:
Added the fix to the latest dev branch.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-06-20 by Koenkk