Koenkk/zigbee2mqtt #3753: Failed to configure device

SargonofAssyria, 2020-06-16:
Just updated to zigbee2mqtt version 1.14.0-dev (commit #f8d082e)
Raspbian 4.19.118-v7+
Coordinator CC1352P_2 firmware version: '{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20200417}}'
I get the following error with device Xiaomi GZCGQ01LM light sensor.
`Failed to configure 'Buiten Lux', attempt 1 (Error: Bind 0x04cf8cdf3c7755e8/1 genPowerCfg fro
m '0x00124b001ca1b0bc/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
    at Endpoint.<anonymous> (/home/pi/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:255:23)

    at Generator.throw (<anonymous>)
    at rejected (/home/pi/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
`
In the configuration I put
`  '0x04cf8cdf3c7755e8':
    friendly_name: 'Buiten Lux'
    illuminance_lux_precision: {1000: 0, 100: 1}
`
The device s seems to operate normally.

Koenkk, 2020-06-16:
The configure probably fails because the device is sleeping. You can force a configure via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure , but make sure to wake the device up right before doing this (by clicking a button on it). Once the configure is successful it won't be done again unless the configure method is changed for this device (which was done in this case to fix the spamming of the sensor)

SargonofAssyria, 2020-06-17:
Tnx, the device is configured now, and the errors are gone. So is the update frequency of every 5-15 seconds (which might be a little to much)
Is there a parameter for the update frequency of lux now? It looks now that the update frequency is more then 9 minutes. In Domoticz I have a script running every minute and average those values for  10 minutes, to get a stable average value, on which I can trigger roller shutters etc...

SargonofAssyria, 2020-06-17:
I waited a little long but updates can come with an interval of more then 20 minutes.
My scripts have become totally useless.
Please put some parameters in the config file.

Koenkk, 2020-06-17:
The minimum reporting frequency is now set to 15 seconds, if no change is reported after that it means that the illuminance is stable (there is no significant change, this is done to save battery).

Currently it's not possible to specify a different reportable change. In case you want to do this it can be achieved by editing devices.js and changing `await configureReporting.illuminance(endpoint, 15, repInterval.HOUR, 500);` to `await configureReporting.illuminance(endpoint, 15, repInterval.HOUR, 0);` for the `GZCGQ01LM` entry.

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

SargonofAssyria, 2020-06-18:
OK, Tnx, I will do that and see what will be the best setting in my case.

closed 2020-06-18 by SargonofAssyria

gtankit96, 2022-03-23:
Failed to configure develco energy meter (SMMZB-310) but was able to ping it , and this is the log message below:-

Zigbee2MQTT:error 2022-03-23 07:48:47: Failed to configure '0x0015bc001902abd3', attempt 1 (Error: ConfigureReporting 0x0015bc001902abd3/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 6306 - 1 - 3 - 6 - 7 after 10000ms)
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7))

Can anyone please help me here? @Koenkk  @SargonofAssyria

Loic691, 2024-02-11:
Hi @Koenkk 
I have the following error message every minutes or 2 minutes witch Orvibo ST30 temperature and humidity sensor.
The device works well.
How can we improve this or resolve this issue ?
`2024-02-11 14:23:39Failed to configure 'Sonde sun', attempt 1 (Error: ConfigureReporting 0xb4e3f9fffe25ad37/1 msTemperatureMeasurement([{"attribute":"measuredValue","minimumReportInterval":10,"maximumReportInterval":3600,"reportableChange":100}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 49560 - 1 - 9 - 1026 - 7 after 10000ms) at Object.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/waitress.ts:63:23) at EZSPAdapter.sendZclFrameToEndpointInternal (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ezsp/adapter/ezspAdapter.ts:486:47) at Queue.execute (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:35:20) at Request.send (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:79:20) at Endpoint.configureReporting (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:669:28) at Object.temperature (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:98:5) at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/devices/orvibo.ts:150:13) at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:120:13) at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:174:17))`