Koenkk/zigbee2mqtt #3754: Keen Home Smart vents broken since version 1.6

alexanv1, 2020-06-16:
# Bug Report
## What happened

Keen Home Smart Vents have been broken since version 1.6 (unfortunately I just got around to troubleshooting it now, been running on 1.6 seamlessly for the past many months).

Everything looks ok in the logs, the vent reports the state changing but the vent doesn't actually open or close.
`debug 2020-06-16 00:22:36: Received MQTT message on 'zigbee2mqtt/Office Smart Vent/set' with data 'OPEN'
debug 2020-06-16 00:22:36: Publishing 'set' 'state' to 'Office Smart Vent'
info  2020-06-16 00:22:37: MQTT publish: topic 'zigbee2mqtt/Office Smart Vent', payload '{"battery":90,"linkquality":99,"position":100,"state":"CLOSE","temperature":20.23,"pressure":1010.3}'
debug 2020-06-16 00:22:37: Received Zigbee message from 'Office Smart Vent', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":255}' from endpoint 1 with groupID 0
info  2020-06-16 00:22:37: MQTT publish: topic 'zigbee2mqtt/Office Smart Vent', payload '{"battery":90,"linkquality":84,"position":"100","state":"OPEN","temperature":20.23,"pressure":1010.3}'
debug 2020-06-16 00:22:51: Received MQTT message on 'zigbee2mqtt/Office Smart Vent/set' with data 'CLOSE'
debug 2020-06-16 00:22:51: Publishing 'set' 'state' to 'Office Smart Vent'
debug 2020-06-16 00:22:54: Received Zigbee message from 'Office Smart Vent', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"manufacturerCode":4443,"imageType":0,"fileVersion":18}' from endpoint 1 with groupID 0
info  2020-06-16 00:22:55: MQTT publish: topic 'zigbee2mqtt/Office Smart Vent', payload '{"battery":90,"linkquality":84,"position":0,"state":"OPEN","temperature":20.23,"pressure":1010.3}'
debug 2020-06-16 00:22:55: Received Zigbee message from 'Office Smart Vent', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":0}' from endpoint 1 with groupID 0
info  2020-06-16 00:22:55: MQTT publish: topic 'zigbee2mqtt/Office Smart Vent', payload '{"battery":90,"linkquality":92,"position":"0","state":"CLOSE","temperature":20.23,"pressure":1010.3}'`

Another hint that something is wrong is that all of my smart vents fail to configure:
`error 2020-06-16 00:10:39: Failed to configure 'Office Smart Vent', attempt 1 (Error: Bind 0x0022a30000131d5f/1 genPowerCfg from '0x00124b0019369472/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
    at Endpoint.<anonymous> (C:\zigbee2mqtt\node_modules\zigbee-herdsman\dist\controller\model\endpoint.js:255:23)
    at Generator.throw (<anonymous>)
    at rejected (C:\zigbee2mqtt\node_modules\zigbee-herdsman\dist\controller\model\endpoint.js:6:65))`

## What did you expect to happen

On version 1.6 and below, the logging is identical but the vent does actually move so it's a SW problem, not a problem with the vents.

## Debug Info
Zigbee2mqtt version: 1.14.0 (commit #9009de2)
Adapter hardware: CC2531
Adapter firmware version: 20190619

Koenkk, 2020-06-16:
I've checked the code but I cannot find any differences in it. 

Can you try to remove and repair the device? (https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove)

alexanv1, 2020-06-20:
I've downgraded and upgraded again and the Smart Vents are temporarily working (sometimes it takes a day or two for it to start misbehaving). When this happens again, I will try removing and re-pairing (I've tried resetting the Smart Vent before without removing it first and that had no effect).

I am seeing other errors in the log though. Not sure if these are related or whether they can be ignored.

`error 2020-06-20 12:45:59: Failed to configure 'Downstairs Hallway Smart Vent', attempt 1 (Error: ConfigureReporting 0x0022a30000156191/1 msTemperatureMeasurement([{"attribute":"measuredValue","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":25}], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Status 'INVALID_VALUE')
    at Endpoint.<anonymous> (/zigbee2mqtt-1.14.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:341:23)
    at Generator.next (<anonymous>)
    at fulfilled (/zigbee2mqtt-1.14.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:5:58))
error 2020-06-20 12:45:59: Failed to configure 'Bonus Room Smart Vent - Left', attempt 1 (Error: Bind 0x0022a30000150d0f/1 genLevelCtrl from '0x00124b0019369472/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
    at Endpoint.<anonymous> (/zigbee2mqtt-1.14.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:255:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.14.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))`

alexanv1, 2020-06-21:
Spoke too soon (above)... Two of my Smart Vents didn't work after upgrade (the others do).
Removed and re-added the device, no difference in behavior. In fact, it's even slightly worse after re-adding... The Smart Vent is not reporting the current state at all (no "Zigbee message from...." in the log. Opening and closing it appears to succeed but it doesn't actually move.

Is there additional logging I can enable?

`9265 info  2020-06-20 20:36:35: Clearing Home Assistant discovery topic for '0x0022a3000012b3d4'
9266 info  2020-06-20 20:36:35: MQTT publish: topic 'homeassistant/cover/0x0022a3000012b3d4/cover/config', payload 'null'
9267 info  2020-06-20 20:36:35: MQTT publish: topic 'homeassistant/sensor/0x0022a3000012b3d4/temperature/config', payload 'null'
9268 info  2020-06-20 20:36:35: MQTT publish: topic 'homeassistant/sensor/0x0022a3000012b3d4/pressure/config', payload 'null'
9269 info  2020-06-20 20:36:35: MQTT publish: topic 'homeassistant/sensor/0x0022a3000012b3d4/battery/config', payload 'null'
9270 info  2020-06-20 20:36:35: MQTT publish: topic 'homeassistant/sensor/0x0022a3000012b3d4/linkquality/config', payload 'null'
9288 info  2020-06-20 20:37:47: Device '0x0022a3000012b3d4' joined
9289 info  2020-06-20 20:37:47: Starting interview of '0x0022a3000012b3d4'
9290 info  2020-06-20 20:37:47: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x0022a3000012b3d4"}}'
9291 info  2020-06-20 20:37:47: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x0022a3000012b3d4"}}'
9292 debug 2020-06-20 20:37:47: Device '0x0022a3000012b3d4' announced itself
9293 info  2020-06-20 20:37:47: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x0022a3000012b3d4"}}'
9296 debug 2020-06-20 20:38:06: Received Zigbee message from '0x0022a3000012b3d4', type 'readResponse', cluster 'genBasic', data '{"modelId":"SV01-410-MP-1.1"}' from endpoint 1 with groupID 0
9298 info  2020-06-20 20:38:06: MQTT publish: topic 'homeassistant/cover/0x0022a3000012b3d4/cover/config', payload '{"command_topic":"zigbee2mqtt/0x0022a3000012b3d4/set","position_topic":"zigbee2mqtt/0x0022a3000012b3d4","set_position_topic":"zigbee2mqtt/0x0022a3000012b3d4/set","set_position_template":"{ \"position\": {{ position }} }","value_template":"{{ value_json.position }}","json_attributes_topic":"zigbee2mqtt/0x0022a3000012b3d4","name":"0x0022a3000012b3d4_cover","unique_id":"0x0022a3000012b3d4_cover_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x0022a3000012b3d4"],"name":"0x0022a3000012b3d4","sw_version":"Zigbee2mqtt 1.14.0","model":"Smart vent (SV01)","manufacturer":"Keen Home"},"availability_topic":"zigbee2mqtt/bridge/state"}'
9299 info  2020-06-20 20:38:06: MQTT publish: topic 'homeassistant/sensor/0x0022a3000012b3d4/temperature/config', payload '{"unit_of_measurement":"Â°C","device_class":"temperature","value_template":"{{ value_json.temperature }}","state_topic":"zigbee2mqtt/0x0022a3000012b3d4","json_attributes_topic":"zigbee2mqtt/0x0022a3000012b3d4","name":"0x0022a3000012b3d4_temperature","unique_id":"0x0022a3000012b3d4_temperature_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x0022a3000012b3d4"],"name":"0x0022a3000012b3d4","sw_version":"Zigbee2mqtt 1.14.0","model":"Smart vent (SV01)","manufacturer":"Keen Home"},"availability_topic":"zigbee2mqtt/bridge/state"}'
9300 info  2020-06-20 20:38:06: MQTT publish: topic 'homeassistant/sensor/0x0022a3000012b3d4/pressure/config', payload '{"unit_of_measurement":"hPa","device_class":"pressure","value_template":"{{ value_json.pressure }}","state_topic":"zigbee2mqtt/0x0022a3000012b3d4","json_attributes_topic":"zigbee2mqtt/0x0022a3000012b3d4","name":"0x0022a3000012b3d4_pressure","unique_id":"0x0022a3000012b3d4_pressure_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x0022a3000012b3d4"],"name":"0x0022a3000012b3d4","sw_version":"Zigbee2mqtt 1.14.0","model":"Smart vent (SV01)","manufacturer":"Keen Home"},"availability_topic":"zigbee2mqtt/bridge/state"}'
9301 info  2020-06-20 20:38:06: MQTT publish: topic 'homeassistant/sensor/0x0022a3000012b3d4/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/0x0022a3000012b3d4","json_attributes_topic":"zigbee2mqtt/0x0022a3000012b3d4","name":"0x0022a3000012b3d4_battery","unique_id":"0x0022a3000012b3d4_battery_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x0022a3000012b3d4"],"name":"0x0022a3000012b3d4","sw_version":"Zigbee2mqtt 1.14.0","model":"Smart vent (SV01)","manufacturer":"Keen Home"},"availability_topic":"zigbee2mqtt/bridge/state"}'
9302 info  2020-06-20 20:38:06: MQTT publish: topic 'homeassistant/sensor/0x0022a3000012b3d4/linkquality/config', payload '{"icon":"mdi:signal","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x0022a3000012b3d4","json_attributes_topic":"zigbee2mqtt/0x0022a3000012b3d4","name":"0x0022a3000012b3d4_linkquality","unique_id":"0x0022a3000012b3d4_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x0022a3000012b3d4"],"name":"0x0022a3000012b3d4","sw_version":"Zigbee2mqtt 1.14.0","model":"Smart vent (SV01)","manufacturer":"Keen Home"},"availability_topic":"zigbee2mqtt/bridge/state"}'
[... cut ...]
9521 debug 2020-06-20 20:46:41: Received MQTT message on 'zigbee2mqtt/0x0022a3000012b3d4/set' with data 'CLOSE'
9522 debug 2020-06-20 20:46:41: Publishing 'set' 'state' to '0x0022a3000012b3d4'
9523 info  2020-06-20 20:46:46: MQTT publish: topic 'zigbee2mqtt/0x0022a3000012b3d4', payload '{"position":0}'
9535 debug 2020-06-20 20:48:02: Received MQTT message on 'zigbee2mqtt/0x0022a3000012b3d4/set' with data 'OPEN'
9536 debug 2020-06-20 20:48:02: Publishing 'set' 'state' to '0x0022a3000012b3d4'
9538 info  2020-06-20 20:48:04: MQTT publish: topic 'zigbee2mqtt/0x0022a3000012b3d4', payload '{"position":100}'
`

Koenkk, 2020-06-24:
Can you provide the herdsman debug logging when opening/closing the vent?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

alexanv1, 2020-06-25:
Log snipped for closing the vent (vent didn't move):

`zigbee2mqtt:debug 2020-06-24 22:29:13: Received MQTT message on 'zigbee2mqtt/Office Smart Vent/set' with data 'CLOSE'
zigbee2mqtt:debug 2020-06-24 22:29:13: Publishing 'set' 'state' to 'Office Smart Vent'
2020-06-25T05:29:13.857Z zigbee-herdsman:controller:endpoint Command 0x0022a30000131d5f/1 genLevelCtrl.moveToLevel({"level":"0","transtime":0}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null})
2020-06-25T05:29:13.858Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":14093,"destendpoint":1,"srcendpoint":1,"clusterid":8,"transid":132,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[1,86,0,0,0,0]}}
2020-06-25T05:29:13.858Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,16,36,1,13,55,1,1,8,0,132,0,30,6,1,86,0,0,0,0,204]
2020-06-25T05:29:13.871Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2020-06-25T05:29:13.871Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2020-06-25T05:29:13.871Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2020-06-25T05:29:13.872Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2020-06-25T05:29:13.872Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-06-25T05:29:13.882Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,132,66]
2020-06-25T05:29:13.882Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,132,66]
2020-06-25T05:29:13.882Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,132] - 66
2020-06-25T05:29:13.882Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":132}
2020-06-25T05:29:13.883Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-06-25T05:29:16.946Z zigbee-herdsman:controller:log Permit joining
2020-06-25T05:29:16.947Z zigbee-herdsman:adapter:zStack:znp:SREQ --> ZDO - mgmtPermitJoinReq - {"addrmode":15,"dstaddr":65532,"duration":254,"tcsignificance":0}
2020-06-25T05:29:16.947Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,5,37,54,15,252,255,254,0,228]
2020-06-25T05:29:16.963Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,101,54,0,82]
2020-06-25T05:29:16.963Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,101,54,0,82]
2020-06-25T05:29:16.963Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 54 - [0] - 82
2020-06-25T05:29:16.963Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - mgmtPermitJoinReq - {"status":0}
2020-06-25T05:29:16.963Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-06-25T05:29:16.967Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":133,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,87,2,11,254,0]}}
2020-06-25T05:29:16.967Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,26,36,2,2,253,255,0,0,0,0,0,0,242,0,0,242,33,0,133,0,30,6,0,25,87,2,11,254,0,57]
2020-06-25T05:29:16.979Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,2,0,103]
2020-06-25T05:29:16.979Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,2,0,103]
2020-06-25T05:29:16.980Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 2 - [0] - 103
2020-06-25T05:29:16.980Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequestExt - {"status":0}
2020-06-25T05:29:16.980Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-06-25T05:29:16.990Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,182,0,0,0,240]
2020-06-25T05:29:16.990Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,182,0,0,0,240]
2020-06-25T05:29:16.990Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 182 - [0,0,0] - 240
2020-06-25T05:29:16.991Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - mgmtPermitJoinRsp - {"srcaddr":0,"status":0}
2020-06-25T05:29:16.991Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-06-25T05:29:16.994Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,242,133,176]
2020-06-25T05:29:16.994Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,242,133,176]
2020-06-25T05:29:16.994Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,242,133] - 176
2020-06-25T05:29:16.994Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":242,"transid":133}
2020-06-25T05:29:16.994Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-06-25T05:29:17.985Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,25,68,129,0,0,8,0,13,55,1,1,0,63,0,232,188,6,0,0,5,8,86,11,0,0,206,170,28,171]
2020-06-25T05:29:17.985Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,25,68,129,0,0,8,0,13,55,1,1,0,63,0,232,188,6,0,0,5,8,86,11,0,0,206,170,28,171]
2020-06-25T05:29:17.986Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 25 - 2 - 4 - 129 - [0,0,8,0,13,55,1,1,0,63,0,232,188,6,0,0,5,8,86,11,0,0,206,170,28] - 171
[... cut ...]
zigbee2mqtt:info  2020-06-24 22:32:43: MQTT publish: topic 'zigbee2mqtt/Office Smart Vent', payload '{"battery":90,"linkquality":65,"position":"100","state":"OPEN","temperature":14.95,"pressure":1011.4}'
2020-06-25T05:32:43.049Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []'`

Incidentally, is there a way with HASSIO Add-On to get the herdsman logging output to go to the same log as the rest of zigbee2mqtt logging?
I can only see herdsman logs in the logs in the Logs tab for the add-on in Hassio but that's not very convenient as it only shows the last few hundred lines.

Koenkk, 2020-06-28:
> I can only see herdsman logs in the logs in the Logs tab for the add-on in Hassio but that's not very convenient as it only shows the last few hundred lines

AFAIK this is not possible atm.

I cannot find anything strange in the log. Are you able to sniff the traffic on both 1.6 (where open/close command works) and on 1.14.0 (where close/open command fails)? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

alexanv1, 2020-06-29:
I will need to obtain another CC2531 for sniffing... Will provide the results in a few days.
Thanks!

alexanv1, 2020-07-02:
Zigbee traffic captures:
[Captures.zip](https://github.com/Koenkk/zigbee2mqtt/files/4862220/Captures.zip)

The difference that I see is that 1.6 is sending _Command: Move to Level with OnOff (0x04)_ and 1.14.1 is sending _Command: Move to Level (0x00)_
Looks like the vent doesn't move when Move To Level is sent (though it reports the current setting as having changed).

alexanv1, 2020-07-02:
Btw, I think the reason it appears sometimes works is if the vent is already open, it will respond to "move to level". However, if it's closed, it will only respond to "move to level with OnOff". That might explain why I sometimes saw it working with 1.14. Might also explain why the fix for the original issue [after the converters refactor for 1.7] with Keen Smart Vents (#2323) appeared to be complete.

Seems like a one-line change in toZigbee.js but I don't know if it can break other cover devices and would require a custom converter for Keen Home... Also, I have no easy way to test the fix on hassio (don't have root ssh setup).

Koenkk, 2020-07-03:
Well found, `moveToLevelWithOnOff` should be fine as this was also used in 1.6, it seems that I just simply fixed it wrong in https://github.com/Koenkk/zigbee-herdsman-converters/commit/3b5345c7c3eefdebe099acc7e6ad3261656bcb62. 

Pushed changes to the latest dev branch, fix will be available in a few hours from now in the edge version of the hassio addon.

alexanv1, 2020-07-04:
Works great, thanks!

There is another minor issue with the smart vents (also seems to have regressed since 1.6) - the pressure is not always reported since the vent appears to report pressure in two ways and only one is being converted. I will open a separate bug on that later.

closed 2020-07-04 by alexanv1