Koenkk/zigbee2mqtt #3755: Xiaomi GZCGQ01LM (Light sensor) appears as disconnected in the network map

McGiverGim, 2020-06-16:
# Bug Report
## What happened
The device GZCGQ01LM has started to appear as disconnected in the network map. It connects and disconnects continuously but it seems it is working because I receive the device data. 
It is connected usually through a router, one Xiaomi LLKZMK11LM relay.
The signal quality is good, the device is right next to the router and at the same room (at 2 or 3m ) of the coordinator.

Here I attach the complete map returned, the device with the problem is `luminosidad_comedor` with ieeeAddr `0x04cf8cdf3c78c831`.

```json

{
	"nodes": [
		{
			"ieeeAddr": "0x00124b0018e86d21",
			"friendlyName": "Coordinator",
			"type": "Coordinator",
			"networkAddress": 0,
			"failed": [],
			"lastSeen": null
		},
		{
			"ieeeAddr": "0x00158d0004147c40",
			"friendlyName": "presencia_pasillo",
			"type": "EndDevice",
			"networkAddress": 30780,
			"manufacturerName": "LUMI",
			"modelID": "lumi.sensor_motion.aq2",
			"lastSeen": 1584096893082
		},
		{
			"ieeeAddr": "0x00158d0003f431e6",
			"friendlyName": "presencia_cocina",
			"type": "EndDevice",
			"networkAddress": 43747,
			"manufacturerName": "LUMI",
			"modelID": "lumi.sensor_motion.aq2",
			"lastSeen": 1584096928708
		},
		{
			"ieeeAddr": "0x00158d00045ee734",
			"friendlyName": "puerta_trastero",
			"type": "EndDevice",
			"networkAddress": 53354,
			"manufacturerName": "LUMI",
			"modelID": "lumi.sensor_magnet",
			"lastSeen": 1584097817803
		},
		{
			"ieeeAddr": "0x00158d0003e7cf82",
			"friendlyName": "pulsador_alarma",
			"type": "EndDevice",
			"networkAddress": 29730,
			"manufacturerName": "LUMI",
			"modelID": "lumi.sensor_switch",
			"lastSeen": 1584100985957
		},
		{
			"ieeeAddr": "0x00158d0003612ca2",
			"friendlyName": "detector_humo",
			"type": "EndDevice",
			"networkAddress": 1439,
			"manufacturerName": "LUMI",
			"modelID": "lumi.sensor_smoke",
			"lastSeen": 1584617280269
		},
		{
			"ieeeAddr": "0x00158d00036b8ca4",
			"friendlyName": "temperatura_banyo_pasillo",
			"type": "EndDevice",
			"networkAddress": 10359,
[... 375 lines cut ...]
				"networkAddress": 54002
			},
			"linkquality": 196,
			"depth": 1,
			"routes": [],
			"sourceIeeeAddr": "0x00158d00044adbcc",
			"targetIeeeAddr": "0x00158d0003f453e3",
			"sourceNwkAddr": 48768,
			"lqi": 196,
			"relationship": 2
		}
	]
}
```

## What did you expect to happen
The device must appear as connected to the network map always.

## How to reproduce it (minimal and precise)
I doubt how to do this. It started to happen yesterday, I don't know exactly why.

## Debug Info
Zigbee2mqtt version: 1.14.0
Adapter hardware: CC2531
Adapter firmware version: zStack12.20190608

Koenkk, 2020-06-16:
This is normal for battery powered devices (end devices) because they are sleeping while the network scan is executed. When they are sleeping for some longer time, the router will mark the route as inactive (which is perfectly fine).

McGiverGim, 2020-06-16:
Ok, if it is expected, then I close the issue. The strange thing is only happens with this, that is precisely the device that sends more messages in the network 😋

closed 2020-06-16 by McGiverGim