Koenkk/zigbee2mqtt #3756: Not connected to MQTT server

noohi, 2020-06-16:
# Bug Report
## What happened
Getting "Not connected to MQTT server!" in the log after upgrading, need to reset multiple times for it to finally connect.
HASS and NodeRed on the same docker setup and MQTT entries, do not show any problems reconnecting.
MQTT Broker Mosquitto.
Let me know if there is anything more than this one line that might help debug this, already have debug logging enabled.

## What did you expect to happen
Silently reconnect, as it usually does.

## How to reproduce it (minimal and precise)
Potentially it's just on upgrades, might be on machine reboots as well, not sure as it's seemingly random.

## Debug Info
Zigbee2mqtt version: latest dev docker
Adapter hardware: CC1352P-2
Adapter firmware version: latest

Koenkk, 2020-06-16:
Please provide debug logging.

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

noohi, 2020-06-17:
Attached, thanks.

[log.zip](https://github.com/Koenkk/zigbee2mqtt/files/4794076/log.zip)

Koenkk, 2020-06-17:
Does it ever connect to the MQTT server? (check log for `Connected to MQTT server`)

noohi, 2020-06-17:
Sure, on reboot, just does not have to be the first one, sometimes it needs 10 of them.
Attached, hopefully you can see the diff and conclude what is different.
Sorry about the extra new lines, had to pull it from the P
[z2mConnected.zip](https://github.com/Koenkk/zigbee2mqtt/files/4794742/z2mConnected.zip)
ortainer, was not in the files for some reason.

Koenkk, 2020-06-18:
I've tested but in my case Zigbee2mqtt reconnects fine. I've tested:
- MQTT off, Zigbee2mqtt start, wait 10 seconds, start MQTT -> connects
- MQTT on, zigbee2mqtt start, wait 10 seconds, stop mqtt, wait 10 seconds, start mqtt -> connects

I need some extra logging in your case, can you try replacing the following file in your docker container: `/app/lib/mqtt.js` with https://raw.githubusercontent.com/Koenkk/zigbee2mqtt/issue/3756/lib/mqtt.js and provide the lgoging?

noohi, 2020-06-19:
OK, I think this is something about upgrading, you need to first upgrade via compose from one dev version to another, then on first turning on it does not work.
And it's not that the connection mechanism doesn't work, it does, it's just that it doesn't fire in the first place

New logs:
[log (restarted).txt](https://github.com/Koenkk/zigbee2mqtt/files/4802955/log.restarted.txt)
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4802956/log.txt)

One note, I don't know for sure if the new JS files actually took place, I add them in the compose file under volumes:
- /home/nuhi/mqtt.js:/app/lib/mqtt.js

Since it "compiled", I presume it worked.

Let me know if you need anything else.

Koenkk, 2020-06-20:
I've updated https://github.com/Koenkk/zigbee2mqtt/blob/issue/3756/lib/mqtt.js, can you try again? (now logs are also written to the file instead of STDOUT only).

noohi, 2020-06-20:
Sure:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4807941/log.txt)
[log (connected).txt](https://github.com/Koenkk/zigbee2mqtt/files/4807942/log.connected.txt)

The key takeaway from this is I'm seeing "reconnecting false connected true" in the version which works, after some short time - would still prefer if it would check sooner, but it connected so fine.

However, the non-working one never displays anything similar, it does not run that loop.
Just full of:
error: Cannot send message: topic: ...
error: Not connected to MQTT server!

I can now switch between working and non-working on almost each reboot, and the working one sometimes loses connection and becomes non-working - if that happens I'll store the log for the reconnect line.
Which is weird, it's all on the same server, should have no connection issues nor NodeRed or HASS report anything similar.
I can see the Mosquitto logs and only Z2M reconnects - so I know it's not the connection problem.

Thanks.

Koenkk, 2020-06-20:
Updated https://github.com/Koenkk/zigbee2mqtt/blob/issue/3756/lib/mqtt.js with more debug logging.

As the timer only triggers every 10 seconds, make sure to let it run for at least 20 seconds (to be on the safe side)

noohi, 2020-06-20:
OK, here it is:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4808103/log.txt)

Koenkk, 2020-06-20:
Looks that it doesn't get into the mqtt.connect() for some reason. Can you try to also replace `lib/controller.js` with https://github.com/Koenkk/zigbee2mqtt/blob/issue/3756/lib/controller.js

noohi, 2020-06-20:
Thanks, cannot make it break at the moment, maybe lucky, maybe you changed something.
Will post when it happens, I see the new messages "IN TIMER reconnecting false connected true"

Caught this on one of the reboots:

error 2020-06-20 18:42:12: Not connected to MQTT server! (MESSAGE)
error 2020-06-20 18:42:12: Cannot send message: topic: 'mqtt/prizemlje_wc_motion', payload: '{"battery":100,"voltage":3015,"linkquality":78,"occupancy":true}
info  2020-06-20 18:42:13: CONTROLLER 5
info  2020-06-20 18:42:13: CONTROLLER 6
info  2020-06-20 18:42:13: Connecting to MQTT server at mqtt://mosquitto:1883
error 2020-06-20 18:42:13: GOT HERE 1
error 2020-06-20 18:42:13: GOT HERE 2
debug 2020-06-20 18:42:13: Using MQTT client ID: 'z2m'
error 2020-06-20 18:42:13: GOT HERE 3
error 2020-06-20 18:42:13: GOT HERE 4
error 2020-06-20 18:42:13: GOT HERE 5
error 2020-06-20 18:42:13: GOT HERE 6
error 2020-06-20 18:42:13: GOT HERE 8
error 2020-06-20 18:42:13: GOT HERE 7
info  2020-06-20 18:42:13: Connected to MQTT server
info  2020-06-20 18:42:13: MQTT publish: topic 'mqtt/bridge/state', payload 'online'
info  2020-06-20 18:42:13: CONTROLLER 7

Does that help, or you need more?

And thanks, of course.

Koenkk, 2020-06-21:
Issue should be fixed in the latest dev branch.

I noticed that there are also a lot of other errors on startup:

```
error 2020-06-20 17:09:32: Failed to setup reporting for '0x001788010302bfca' - Error: Bind 0x001788010302bfca/11 genOnOff from '0x00124b001ca1bc4c/1' failed (Error: SREQ '--> ZDO - bindReq - {"dstaddr":33813,"srcaddr":"0x001788010302bfca","srcendpoint":11,"clusterid":6,"dstaddrmode":3,"dstaddress":"0x00124b001ca1bc4c","dstendpoint":1}' failed with status '(0x10: MEM_ERROR)' (expected '(0x00: SUCCESS)'))
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:255:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)

```

These should also be fixed now.

noohi, 2020-06-21:
Thanks a lot!

closed 2020-06-21 by noohi