Koenkk/zigbee2mqtt #3758: OSRAM Smart+ plug behaving weirdly

DBa2016, 2020-06-16:
# Bug Report
To be honest, I am not sure this is a z2m bug. More likely this has something to do with OTA.
## What happened
Today at 12:14pm, one of my OSRAM SMART+ plugs started to "flash" slowly. It turns on for roughly half a second, maybe a whole second, then it turns off for a minute, then again. This went on for about 15 minutes, then stopped. However, the plug also stopped responding to z2m. After pressing the button on the plug, issue went away (for now).

Following entries are written into the log: https://hastebin.com/coyuzosadu.apache

## What did you expect to happen
Nothing should have happened, or, more precisely, no light status changes.
## How to reproduce it (minimal and precise)
Not sure. 
## Debug Info
Zigbee2mqtt version: dev-latest
Adapter hardware: conbee2
Adapter firmware version: ConBee2 does not report FW version.

Koenkk, 2020-06-16:
`deviceAnnounce` is (most of the times) send when a devices is started (after having no power). Probably the software on the plug crashes.

DBa2016, 2020-06-18:
Interestingly enough, a second Ledvance plug (well, that one is actually an OSRAM one, if that matters) started behaving in similar way, and it is not even connected to the ZB network (it was in the past, bzt I forgot to re-add it during a re-setup of ZB net).
I contacted LEDVANCE, they are going to replace the plugs - we will see if it fixes the issue.

closed 2020-07-26 by stale[bot]