Koenkk/zigbee2mqtt #3760: Sinope TH1123ZB - keypadLockout UNREPORTABLE_ATTRIBUTE

Maelstrom96, 2020-06-16:
# Bug Report
## What happened
I've added a Sinope TH1123ZB and it isn't reporting the "KeypadLock" status properly. However, sending a Unlock or Lock command seems to work fine. 

I've checked logs while having zigbee-herdsman log in debug mode. It doesn't seem to result in more relevent information than this : 

```2020-06-15T20:50:45.651Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0,0,4,2,77,32,1,1,0,131,0,31,221,184,0,0,7,24,7,7,140,0,1,0,77,32,9,186]
2020-06-15T20:50:45.652Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,4,2,77,32,1,1,0,131,0,31,221,184,0,0,7,24,7,7,140,0,1,0,77,32,9,186]
2020-06-15T20:50:45.652Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,4,2,77,32,1,1,0,131,0,31,221,184,0,0,7,24,7,7,140,0,1,0,77,32,9] - 186
2020-06-15T20:50:45.653Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":516,"srcaddr":8269,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":131,"securityuse":0,"timestamp":12115231,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[24,7,7,140,0,1,0]}}
2020-06-15T20:50:45.656Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":7,"manufacturerCode":null,"commandIdentifier":7},"Payload":[{"status":140,"direction":0,"attrId":1}]},"address":8269,"endpoint":1,"linkquality":131,"groupID":0}'
2020-06-15T20:50:45.657Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-06-15T20:50:45.658Z zigbee-herdsman:controller:endpoint ConfigureReporting 0x500b91400002466d/1 hvacUserInterfaceCfg([{"attribute":"keypadLockout","minimumReportInterval":1,"maximumReportInterval":0}], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Status 'UNREPORTABLE_ATTRIBUTE')
^[[31mzigbee2mqtt:error^[[39m 2020-06-15 16:50:45: Failed to configure '0x500b91400002466d', attempt 1 (Error: ConfigureReporting 0x500b91400002466d/1 hvacUserInterfaceCfg([{"attribute":"keypadLockout","minimumReportInterval":1,"maximumReportInterval":0}], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Status 'UNREPORTABLE_ATTRIBUTE')
    at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:341:23)
    at Generator.next (<anonymous>)
    at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:5:58))
```

I've tried to see if I couldn't find something wrong in `zigbee-herdsman-converters` but I'm really not sure I understand the flow to properly debug it effectively.

Let me know if you need any more informations from me.

## Debug Info
Zigbee2mqtt version: 1.14.0 (commit #9009de2)
Adapter hardware: CC2531
Adapter firmware version: v3.0.X-20190425

Koenkk, 2020-06-16:
Changing the entry to 

```js
{
    zigbeeModel: ['TH1123ZB'],
    model: 'TH1123ZB',
    vendor: 'Sinope',
    description: 'Zigbee line volt thermostat',
    supports: 'local temp, units, keypad lockout, mode, state, backlight, outdoor temp, time',
    fromZigbee: [
        fz.thermostat_att_report,
        fz.hvac_user_interface,
        fz.metering_power,
        fz.ignore_temperature_report,
        fz.sinope_thermostat_state,
    ],
    toZigbee: [
        tz.thermostat_local_temperature,
        tz.thermostat_occupied_heating_setpoint,
        tz.thermostat_unoccupied_heating_setpoint,
        tz.thermostat_temperature_display_mode,
        tz.thermostat_keypad_lockout,
        tz.thermostat_system_mode,
        tz.thermostat_running_state,
        tz.sinope_thermostat_occupancy,
        tz.sinope_thermostat_backlight_autodim_param,
        tz.sinope_thermostat_time,
        tz.sinope_thermostat_enable_outdoor_temperature,
        tz.sinope_thermostat_outdoor_temperature,
    ],
    meta: {configureKey: 2},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        const binds = [
            'genBasic', 'genIdentify', 'genGroups', 'hvacThermostat',
            'hvacUserInterfaceCfg', 'msTemperatureMeasurement', 'seMetering',
        ];

        await bind(endpoint, coordinatorEndpoint, binds);
        await configureReporting.thermostatTemperature(endpoint, 10, 60, 50);
        await configureReporting.thermostatOccupiedHeatingSetpoint(endpoint, 1, 0, 50);
        await configureReporting.thermostatSystemMode(endpoint, 1, 0);
        await configureReporting.thermostatPIHeatingDemand(endpoint, 1, 900, 5);
        await readMeteringPowerConverterAttributes(endpoint);
        await configureReporting.instantaneousDemand(endpoint);
    },
},
```

will probably fix it

Maelstrom96, 2020-06-16:
I've tried removing the line `await configureReporting.thermostatKeypadLockMode(endpoint, 1, 0);` from the device definition, which didn't quite work.

I had that error : 
```zigbee2mqtt:error 2020-06-16 13:53:34: Failed to configure '0x500b91400002466d', attempt 1 (Error: ConfigureReporting 0x500b91400002466d/1 seMetering([{"attribute":"instantaneousDemand","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":1}], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:341:23)
    at Generator.next (<anonymous>)
    at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:5:58))
```

Removing the line `await configureReporting.instantaneousDemand(endpoint);` fixed all errors but both attributes (`keypadLock` & `instantaneousDemand`) aren't being reported.

Koenkk, 2020-06-17:
Good, I've added the fix to the latest dev branch. It seems that yours doesn't support these attributes. Not much we can do from the zigbee2mqtt side.

Maelstrom96, 2020-06-17:
It's weird because they should support those features without a problem. Here is the SmartThings device configuration for the Sinope TH1123ZB and TH1124ZB (they're the same other than the watt rating) and it defines the reporting for the lock - https://www.sinopetech.com/wp-content/uploads/2019/03/Sinope-Technologies-TH1123ZB_TH1124ZB-V.1.0.5-SVN-503.txt

I'll try to test and see if I can find why it's not being ingested properly.

hamsando, 2020-07-10:
On a very related note to this, as @Maelstrom96 noted TH1123ZB and TH1124ZB  are the same except for the Watt rating for the heater they can manage. Is there some way in the devices.js file to combine those 2 devices into 1 so that corrections only need to be made once?

closed 2020-08-17 by github-actions[bot]