Koenkk/zigbee2mqtt #3763: Errors in LOG

keteflips, 2020-06-17:
zigbee2mqtt running in a Docker, I see this errors in log:

```
2020/06/16 16:48:58 http error: No administrator account found inside the database (err=Object not found inside the database) (code=404)

2020/06/16 16:48:58 http error: No administrator account found inside the database (err=Object not found inside the database) (code=404)

2020/06/17 07:39:47 http error: Invalid JWT token (err=Invalid JWT token) (code=401)
```

Koenkk, 2020-06-17:
I don't recognize this as being zigbee2mqtt logging, are you sure its from zigbee2mqtt?

MPM1107, 2020-06-17:
That's a portainer error. Maybe try flushing your browser cache, but this for sure isn't the right place to ask. I'm 99% sure this has nothing to do with zigbee2mqtt and is just an error in your portainer installation somewhere.

closed 2020-06-17 by Koenkk