Koenkk/zigbee2mqtt #3765: Lock Danalock V3 not reporting status after CC2531 router has been reboot

elbueno222, 2020-06-17:
# Bug Report
## What happened

My lock Danalock V3 is connected to a CC2531 with router firmware and it was working as expected. I turned this CC2531 router off and on again and now I can still send the lock/unlock commands and the lock will lock/unlock the door but it won't send any response with the status.

Something strange is that if I send the command `zigbee2mqtt/[device_id]/get`, the lock does send the current lock/unlock status.

I had a similar issue with an Ikea blind, where sometimes after the pairing process the blind will receive commands and it will open/close, but it won't report the current position of the blind.

The only solution I have found so far is to unpair and pair the device again and again until it reports the status, but I would like to find the origin of the issue and fix it.

## What did you expect to happen

Danalock V3 should send status (lock/unlock) after every lock/unlock action.

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version: 1.14.0
Adapter hardware: CC2531
Adapter firmware version: Source_Routing_20190619

Koenkk, 2020-06-17:
Could you make a sniff of this? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html 

Carefully indicate at what `No.` the message is send

elbueno222, 2020-06-17:
I just made a sniff, and using the source as a filter (zbee_nwk.src == 0x6b1c) to see just the messages sent from the lock, I cannot see any message coming from the lock after the lock/unlock operations are performed, but that doesn't make any sense to me, if that is true why the lock is not sending the status message now if it was sending it before?

Koenkk, 2020-06-18:
@elbueno222 you can try to reconfigure it via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure , make sure to wake the device up right before executing it.

elbueno222, 2020-06-18:
I have unpaired and paired the Danalock to see the behaviour when it's working properly, and now it is sending the status after the lock/unlock operations and I can see the status messages in the sniffer. My question is why is the device sometimes pairing properly and sending the status after the lock/unlock operations and other times it doesn't send the status unless I send the command asking for the status? How can I know if this a defect of the device itself or it has something to do with HA and zigbee2mqtt?

Koenkk, 2020-06-18:
In the non working case, did the configure execute successfully? This can be checked by either:
- After pairing it seeing a message in the log like `Succesfully configured .... `
- In case you want to check later, open `data/database.db`, and check if the line of the device contains `meta":{"configured": NUMBER}` (the `NUMBER` value doesn't matter)

elbueno222, 2020-06-19:
Thanks @Koenkk, I will check it next time the failure happens. How can I access data/database.db in Home Assistant? (I tried accessing HA using both ports 22 and also 22222 to access Zigbee2mqtt container with no luck)

elbueno222, 2020-06-22:
> @elbueno222 you can try to reconfigure it via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure , make sure to wake the device up right before executing it.

@Koenkk The issue happened again, and sending ´zigbee2mqtt/bridge/configure´ fixed it, thank you very much.

It looks like the lock stops sending the status after some days working properly, I will keep an eye to see if the issue happens following a pattern (i.e. locks stops sending the status x hours/days after sending the configure command?) and in that case I will add an automation to send this configure command when needed as a workaround.

closed 2020-08-01 by stale[bot]