Koenkk/zigbee2mqtt #3766: [info request] how to represent a Zigbee group as single light entity in HA (using MQTT light platform?)

netweaver1970, 2020-06-17:
Hi,
I have a working Z2M landscape, 40+ Zigbee devices and also a few Zigbee groups bundling bulbs and bound to  few remotes. That's all working, thanks to the good documentation.

Now, 1 step further, I would like to represent each Zigbee group as a single device in HA (as well as the individual ones).

I expect the MQTT light platform might be the solution but I see [here](https://www.home-assistant.io/integrations/light.mqtt/) that there are 3 schema variants,: DEFAULT, JSON and TEMPLATE.

How would one find out what the exact entries are that the MQTT device (aka Zigbee group) supports? And what MQTT light schema is supported by Z2M? It looks like the JSON schema 'can do most'. But can you chose, being an enduser? I guess not :)

In my test case there are 3 IKEA bulbs E27 CWS opal 600lm, model: 'LED1624G9' in a Zigbee group called 'WhiteCubeCorner'.

What would be the syntax to use in this case? I'm trying to piece together all nuggets of info I can find but I'm a bit lost on this for now. Maybe this can help other Zigbee group novice in the HA world too.

Thanks for this great piece of software.

Koenkk, 2020-06-17:
An example can be found here: https://www.zigbee2mqtt.io/integration/home_assistant.html#groups

netweaver1970, 2020-06-17:
aaaaargh, I do remember seeing that page long time ago, way before I was getting into Zigbee groups. It actually makes sense to put it there :)
Dank je wel ! We can never read too much documentation :)

closed 2020-06-17 by netweaver1970