Koenkk/zigbee2mqtt #3767: IKEA TRADFRI control outlet softwareBuildID not updated after ota_update

jesseruiter, 2020-06-17:
# Bug Report
## What happened
I updated my IKEA TRADFRI control outlet (E1603/E1702) and everything seems to work allright. However the log says it is updated to `null`; see line below.
```
zigbee2mqtt:info  2020-06-17 17:50:42: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"ota_update","message":"Finished update of 'standing'","meta":{"status":"update_succeeded","device":"standing","from":{"softwareBuildID":"1.4.020","dateCode":"20180528"},"to":null}}'
```
After running `zigbee2mqtt/bridge/config/devices/get` the softwareBuildID of the device turns out to be still the same. However `"update_available":false`.
## What did you expect to happen
I expected the softwareBuildID to be incremented.
## How to reproduce it (minimal and precise)
Follow the ota_update process, as described [here](https://www.zigbee2mqtt.io/information/ota_updates.html).
## Debug Info
Zigbee2mqtt version: 1.14.0
Adapter hardware: CC2531
Adapter firmware version: 20180815

Koenkk, 2020-06-18:
This is indeed an open issue. We try to read it after the update but because the device is already sleeping it will never answer it. A workaround is to repair the device.

Morphy99, 2020-07-14:
@Koenkk if we try to wake the device with a button press at the end of the update would this help reading the new value? I could set up some notification in home assistant to advise the user to do this in this case.

Koenkk, 2020-07-15:
@Morphy99 yes that should solve it (but note that you need to keep pressing until the value appears). Preferably this is solved in zigbee-herdsman. The correct way to handle this would be to wait until the device shows a sign of life and right after that execute the command.

Morphy99, 2020-07-15:
So I tried tapping the button every so often at the end but it still didn't take the new softwareBuildID. Maybe I need to press it more often? How quickly do these things sleep?

```
info  2020-07-15 21:01:28: Update of 'Remote Bathroom' at 99.59%, +- 1 minutes remaining
info  2020-07-15 21:01:28: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"ota_update","message":"Update of 'Remote Bathroom' at 99.59%, +- 1 minutes remaining","meta":{"status":"update_progress","device":"Remote Bathroom","progress":99.59}}'
info  2020-07-15 21:01:30: MQTT publish: topic 'zigbee2mqtt/Remote Bathroom', payload '{"linkquality":47,"update_available":true,"action":"brightness_down_click","last_seen":"2020-07-15T21:01:30.433Z"}'
info  2020-07-15 21:01:30: MQTT publish: topic 'zigbee2mqtt/Remote Bathroom', payload '{"linkquality":47,"update_available":true,"action":"","last_seen":"2020-07-15T21:01:30.433Z"}'
info  2020-07-15 21:01:30: MQTT publish: topic 'zigbee2mqtt/Remote Bathroom/action', payload 'brightness_down_click'
info  2020-07-15 21:01:30: MQTT publish: topic 'zigbee2mqtt/Light Bathroom', payload '{"state":"ON","color_temp":335,"color":{"x":0.525,"y":0.413},"brightness":1,"update_available":false,"linkquality":39,"last_seen":"2020-07-15T21:01:30.755Z"}'
info  2020-07-15 21:01:58: Update of 'Remote Bathroom' at 99.98%, +- 0 minutes remaining
info  2020-07-15 21:01:58: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"ota_update","message":"Update of 'Remote Bathroom' at 99.98%, +- 0 minutes remaining","meta":{"status":"update_progress","device":"Remote Bathroom","progress":99.98}}'
info  2020-07-15 21:02:01: MQTT publish: topic 'zigbee2mqtt/Remote Bathroom', payload '{"linkquality":47,"update_available":true,"action":"brightness_down_click","last_seen":"2020-07-15T21:02:01.231Z"}'
info  2020-07-15 21:02:01: MQTT publish: topic 'zigbee2mqtt/Remote Bathroom', payload '{"linkquality":47,"update_available":true,"action":"","last_seen":"2020-07-15T21:02:01.231Z"}'
info  2020-07-15 21:02:01: MQTT publish: topic 'zigbee2mqtt/Remote Bathroom/action', payload 'brightness_down_click'
info  2020-07-15 21:02:01: MQTT publish: topic 'zigbee2mqtt/Light Bathroom', payload '{"state":"ON","color_temp":335,"color":{"x":0.525,"y":0.413},"brightness":1,"update_available":false,"linkquality":39,"last_seen":"2020-07-15T21:02:01.539Z"}'
info  2020-07-15 21:02:15: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"Remote Bathroom"}}'
info  2020-07-15 21:02:25: MQTT publish: topic 'zigbee2mqtt/Remote Bathroom', payload '{"linkquality":47,"update_available":true,"action":"brightness_down_click","last_seen":"2020-07-15T21:02:25.705Z"}'
info  2020-07-15 21:02:25: MQTT publish: topic 'zigbee2mqtt/Remote Bathroom', payload '{"linkquality":47,"update_available":true,"action":"","last_seen":"2020-07-15T21:02:25.705Z"}'
info  2020-07-15 21:02:25: MQTT publish: topic 'zigbee2mqtt/Remote Bathroom/action', payload 'brightness_down_click'
info  2020-07-15 21:02:26: MQTT publish: topic 'zigbee2mqtt/Light Bathroom', payload '{"state":"ON","color_temp":335,"color":{"x":0.525,"y":0.413},"brightness":1,"update_available":false,"linkquality":39,"last_seen":"2020-07-15T21:02:26.070Z"}'
info  2020-07-15 21:02:27: MQTT publish: topic 'zigbee2mqtt/Leak Bath', payload '{"battery":100,"voltage":3045,"linkquality":47,"water_leak":false,"last_seen":"2020-07-15T21:02:27.881Z"}'
error 2020-07-15 21:02:54: Failed to call 'OTAUpdate' 'onZigbeeEvent' (AssertionError [ERR_ASSERTION]: No image available for manufacturerCode '4174' imageType '48'
    at getImageMeta (/app/node_modules/zigbee-herdsman-converters/ota/ledvance.js:15:5)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:97:5)
    at async isNewImageAvailable (/app/node_modules/zigbee-herdsman-converters/ota/ledvance.js:48:18)
    at async Object.isUpdateAvailable (/app/node_modules/zigbee-herdsman-converters/ota/common.js:168:23)
    at async OTAUpdate.onZigbeeEvent (/app/lib/extension/otaUpdate.js:37:31)
    at async Controller.callExtensionMethod (/app/lib/controller.js:360:21))
info  2020-07-15 21:03:00: MQTT publish: topic 'zigbee2mqtt/Remote Bathroom', payload '{"linkquality":47,"update_available":true,"action":"brightness_down_click","last_seen":"2020-07-15T21:03:00.794Z"}'
info  2020-07-15 21:03:00: MQTT publish: topic 'zigbee2mqtt/Remote Bathroom', payload '{"linkquality":47,"update_available":true,"action":"","last_seen":"2020-07-15T21:03:00.794Z"}'
info  2020-07-15 21:03:00: MQTT publish: topic 'zigbee2mqtt/Remote Bathroom/action', payload 'brightness_down_click'
info  2020-07-15 21:03:01: MQTT publish: topic 'zigbee2mqtt/Light Bathroom', payload '{"state":"ON","color_temp":335,"color":{"x":0.525,"y":0.413},"brightness":1,"update_available":false,"linkquality":39,"last_seen":"2020-07-15T21:03:01.127Z"}'
info  2020-07-15 21:03:32: Update of 'Remote Bathroom' at 100.00%
info  2020-07-15 21:03:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"ota_update","message":"Update of 'Remote Bathroom' at 100.00%","meta":{"status":"update_progress","device":"Remote Bathroom","progress":100}}'
info  2020-07-15 21:03:42: MQTT publish: topic 'zigbee2mqtt/Remote Bathroom', payload '{"linkquality":47,"update_available":true,"action":"brightness_down_click","last_seen":"2020-07-15T21:03:42.289Z"}'
[... cut ...]
info  2020-07-15 21:03:55: Finished update of 'Remote Bathroom'
info  2020-07-15 21:03:55: MQTT publish: topic 'zigbee2mqtt/Remote Bathroom', payload '{"linkquality":47,"update_available":false,"last_seen":"2020-07-15T21:03:42.289Z"}'
info  2020-07-15 21:03:55: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"ota_update","message":"Finished update of 'Remote Bathroom'","meta":{"status":"update_succeeded","device":"Remote Bathroom","from":{"softwareBuildID":"1.2.214","dateCode":"20170302"},"to":null}}'
```

Koenkk, 2020-07-15:
Until https://github.com/Koenkk/zigbee2mqtt/issues/3767#issuecomment-658594726 is solved, it might be better to just rejoin the device (first [remove](https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove) and then pair again). Note that this is just a cosmetic thing.