Koenkk/zigbee2mqtt #3768: Zigbee2MQTT fails to start

peterselieman, 2020-06-17:
Hi koenK,

I am fighting the whole evening on getting zigbee2mqtt working.

when I enter **npm start** in /opt/zigbee2mqtt it gives me the following error message.

`> zigbee2mqtt@1.14.0 start /opt/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-06-18 00:38:33: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-06-18.00-38-33' filename: log.txt
zigbee2mqtt:info  2020-06-18 00:38:33: Starting zigbee2mqtt version 1.14.0 (commit #9009de2)
zigbee2mqtt:info  2020-06-18 00:38:33: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-06-18 00:38:41: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-06-18 00:38:41: Failed to start zigbee
zigbee2mqtt:error 2020-06-18 00:38:41: Exiting...
zigbee2mqtt:error 2020-06-18 00:38:41: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:72:23)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:6:65)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.14.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.14.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-06-17T22_38_41_477Z-debug.log
`

I tried the suggestions on your fails to start page. 
The port that the cc2531 stick is on should be **/ttyACM1**
I have changed the yaml file to this port.
And I also checked if the port was usable and i got the success message.

Do you have the slightest clue what is going on?

MPM1107, 2020-06-17:
If you really are 100% sure that the serial port is correct, the only other thing I could think of is corrupt firmware/an error on the stick side of things. What firmware did you flash onto the stick?

peterselieman, 2020-06-18:
I flashed the CC2531_DEFAULT_20190608 firmware.
Do i have to flash the source_routing firmware?

Is there also a way to check if the flashed firmware is correct?

MPM1107, 2020-06-18:
Source routing does have its uses but there's nothing wrong with the one you flashed, it should be working. Depending on the way you flashed this varies but usually the flashing software checks the flashed image for errors directly after flashing and errors out if any errors occurred.

Just to be sure about the serial port, can you try running `ls /dev/serial/by-id/` while the stick is plugged in and then using `/dev/serial/by-id/id_of_stick_from_ls_command` as the serial port in zigbee2mqtt?

peterselieman, 2020-06-19:
Hi MPM1107,

I found the id of the stick

usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B00014D63F1-if00

so i edited the configuration.yaml with it.

`
# Serial settings
serial:
  # Location of CC2531 USB sniffer
  port: /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B00014D63F1-if00
`

but it still gives the error message.

MPM1107, 2020-06-19:
Could you share your whole configuration file?

peterselieman, 2020-06-20:
Yes, Ofcourse.

[configuration.zip](https://github.com/Koenkk/zigbee2mqtt/files/4807760/configuration.zip)

t-e-d-d-y, 2020-06-22:
Same error, fixed with : https://www.zigbee2mqtt.io/information/FAQ.html#verify-that-the-user-you-run-zigbee2mqtt-as-has-write-access-to-the-port

oopiicaa, 2020-07-18:
@peterselieman did you manage to fix it?

rubenvg95, 2020-08-01:
Reflash works for me

peterselieman, 2020-09-07:
Well, i've just bought a conbee2 and that works.

korro1975, 2020-09-08:
Hi,. thanks a lot for sharing the information.
I have never used this stick, what about the distance to the sensor, right?. I have seen that cc2531 do not reach a long distance :-(.
Thanks in advance.
Regards,

peterselieman, 2020-09-09:
The range to my furthest lamp is 7m withe 3 walls between them, and that works just fine with the conbee2.
The good thing is that i use tradfri lamps and they are repeaters of the zigbee network. So it doesnt matter what the range is. The other lamps are also passing the commands sent over the air.

korro1975, 2020-09-10:
Interesting. Thanks a lot for sharing the information.Regards.

closed 2020-09-10 by peterselieman