Koenkk/zigbee2mqtt #3769: Issues with Fan State

boojew, 2020-06-18:
# Bug Report
## What happened
I am using a Hampton Bay fan (HDC52EastwindFan) and it does not appear to publish/execute the on/off states received from Homeassistant. When HomeAssistnt calls "off", it appears to ignore that and return on UNLESS also sent with a speed. See the video. I am not quite clear if this is an MQTT issue, with the converter, or the homeassistant extension...

## How to reproduce it (minimal and precise)
See gif
[image]

## Debug Info
Zigbee2mqtt version: 1.4.0
Adapter hardware: CC2531Adapter firmware version:CC2531_DEFAULT_20190608.zip.

Koenkk, 2020-06-18:
I think we need to adapter the integration a bit here.

- Can you provide the debug logging when toggling the switch in Home Assistant?
- Besides a fan, there is also a light discovered for this device, changing the brightness should change speed, does this work?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

boojew, 2020-06-18:
Yea, sorry I forgot the debug logs

```
zigbee2mqtt:info  2020-06-18 13:14:14: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.13.1","commit":"3046680","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190425}},"log_level":"debug","permit_join":false}'
zigbee2mqtt:debug 2020-06-18 13:14:45: Received MQTT message on 'zigbee2mqtt/Office Fan/set' with data '{"state": "OFF"}'
zigbee2mqtt:debug 2020-06-18 13:14:45: Publishing 'set' 'state' to 'Office Fan'
zigbee2mqtt:info  2020-06-18 13:14:45: MQTT publish: topic 'zigbee2mqtt/Office Fan', payload '{"fan_mode":"OFF","fan_state":"ON","last_seen":1592500418972,"linkquality":70,"state":"OFF","brightness":254}'
zigbee2mqtt:debug 2020-06-18 13:14:46: Received MQTT message on 'zigbee2mqtt/Office Fan/set' with data '{"state": "ON"}'
zigbee2mqtt:debug 2020-06-18 13:14:46: Publishing 'set' 'state' to 'Office Fan'
zigbee2mqtt:info  2020-06-18 13:14:46: MQTT publish: topic 'zigbee2mqtt/Office Fan', payload '{"fan_mode":"OFF","fan_state":"ON","last_seen":1592500418972,"linkquality":70,"state":"ON","brightness":254}'
zigbee2mqtt:debug 2020-06-18 13:14:50: Received MQTT message on 'zigbee2mqtt/Office Fan/set/fan_state' with data 'OFF'
zigbee2mqtt:debug 2020-06-18 13:14:50: Publishing 'set' 'fan_state' to 'Office Fan'
zigbee2mqtt:info  2020-06-18 13:14:50: MQTT publish: topic 'zigbee2mqtt/Office Fan', payload '{"fan_mode":"OFF","fan_state":"ON","last_seen":1592500490764,"linkquality":70,"state":"ON","brightness":254}'
zigbee2mqtt:debug 2020-06-18 13:14:52: Received MQTT message on 'zigbee2mqtt/Office Fan/set/fan_state' with data 'OFF'
zigbee2mqtt:debug 2020-06-18 13:14:52: Publishing 'set' 'fan_state' to 'Office Fan'
zigbee2mqtt:info  2020-06-18 13:14:52: MQTT publish: topic 'zigbee2mqtt/Office Fan', payload '{"fan_mode":"OFF","fan_state":"ON","last_seen":1592500492905,"linkquality":70,"state":"ON","brightness":254}'
```

The light itself works perfectly. Thanks!

Koenkk, 2020-06-18:
Ah, if I remember correctly, this fan also has a light right? When manually publishing `zigbee2mqtt/Office Fan/set` with payload `{"state": "OFF"}` or `{"state": "ON"}`, this controls the light right? (and not the fan!)

boojew, 2020-06-18:
Just tested, and you are correct ```zigbee2mqtt/Office Fan/set``` is the light.

Koenkk, 2020-06-20:
Are you using the Home Assistant discovery? If yes, when toggling the switch in HA  I expect the command to be send to `zigbee2mqtt/Office Fan/set/fan_state`. Can you post the Zigbee2mqtt startup log so we can see what zigbee2mqtt discovers?

boojew, 2020-06-23:
Sorry for the late response - I am using discovery... and yea, "zigbee2mqtt/Office Fan/set/fan_state" topic with on/off does indeed control the fan state..
Here are the logs

```
> zigbee2mqtt@1.13.1 start /app
> node index.js

zigbee2mqtt:info  2020-06-23 11:52:22: Logging to console and directory: '/app/data/log/2020-06-23.11-52-22' filename: log.txt
zigbee2mqtt:debug 2020-06-23 11:52:22: Removing old log directory '/app/data/log/2020-05-30.19-08-36'
zigbee2mqtt:debug 2020-06-23 11:52:23: Loaded state from file /app/data/state.json
zigbee2mqtt:info  2020-06-23 11:52:23: Starting zigbee2mqtt version 1.13.1 (commit #3046680)
zigbee2mqtt:info  2020-06-23 11:52:23: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-06-23 11:52:23: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"},"adapter":{"concurrent":null}}'
zigbee2mqtt:info  2020-06-23 11:52:25: zigbee-herdsman started
zigbee2mqtt:info  2020-06-23 11:52:25: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190425}}'
zigbee2mqtt:debug 2020-06-23 11:52:25: Zigbee network parameters: {"panID":6754,"extendedPanID":"0xdddddddddddddddd","channel":11}
zigbee2mqtt:info  2020-06-23 11:52:25: Currently 23 devices are joined:
zigbee2mqtt:info  2020-06-23 11:52:25: Living Room - Temperature Sensor (0x00158d00023f5163): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
zigbee2mqtt:info  2020-06-23 11:52:25: Backdoor Contact Sensor (0x00158d0001d39f43): MCCGQ11LM - Xiaomi Aqara door & window contact sensor (EndDevice)
zigbee2mqtt:info  2020-06-23 11:52:25: Mudroom Lightbulb (0xdd16abfff1a195ed): D1532 - EcoSmart A19 soft white bulb (Router)
zigbee2mqtt:info  2020-06-23 11:52:25: Office Fan (0x0022a3000020c945): 99432 - Hampton Bay Universal wink enabled white ceiling fan premier remote control (EndDevice)
zigbee2mqtt:info  2020-06-23 11:52:25: Office Motion Sensor (0x00158d00010c2723): RTCGQ01LM - Xiaomi MiJia human body movement sensor (EndDevice)
zigbee2mqtt:info  2020-06-23 11:52:25: Tradfri Button (0x000b57fffea8ebee): E1524/E1810 - IKEA TRADFRI remote control (EndDevice)
zigbee2mqtt:info  2020-06-23 11:52:25: Tradfri Repeater - Master Bedroom (0x000d6ffffe2add89): E1746 - IKEA TRADFRI signal repeater (Router)
zigbee2mqtt:info  2020-06-23 11:52:25: Tradfri Repeater - Basement (0x000d6ffffe9cd34e): E1746 - IKEA TRADFRI signal repeater (Router)
zigbee2mqtt:info  2020-06-23 11:52:25: Ben Lamp (0xdd16abffff070a94): D1821 - EcoSmart A19 RGB bulb (Router)
zigbee2mqtt:info  2020-06-23 11:52:25: Tradfri Dimmer (0x000b57fffe9605a4): ICTC-G-1 - IKEA TRADFRI wireless dimmer (EndDevice)
zigbee2mqtt:info  2020-06-23 11:52:25: Nate's Wake Up Light (0xdd16abffff07178e): D1821 - EcoSmart A19 RGB bulb (Router)
zigbee2mqtt:info  2020-06-23 11:52:25: Utility Motion Sensor (0x00158d000132e9c7): RTCGQ01LM - Xiaomi MiJia human body movement sensor (EndDevice)
zigbee2mqtt:info  2020-06-23 11:52:25: Nate Room - Temperature Sensor (0x00158d0001217472): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor (EndDevice)
zigbee2mqtt:info  2020-06-23 11:52:25: Jared Room - Temperature Sensor (0x00158d0001216c95): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor (EndDevice)
zigbee2mqtt:info  2020-06-23 11:52:25: Office - Temperature Sensor (0x00158d0002325c75): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
zigbee2mqtt:info  2020-06-23 11:52:25: Kitchen - Temperature Sensor (0x00158d0002437c0a): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
zigbee2mqtt:info  2020-06-23 11:52:25: Utility Room Lamp (0xdd16abfff1a17a17): D1531 - EcoSmart A19 bright white bulb (Router)
zigbee2mqtt:info  2020-06-23 11:52:25: Nates Room - Motion Sensor (0x00158d000165fe3b): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
zigbee2mqtt:info  2020-06-23 11:52:25: Ground Floor Hallway Light (0xdd16abffff07042a): D1821 - EcoSmart A19 RGB bulb (Router)
zigbee2mqtt:info  2020-06-23 11:52:25: 0xdd16abffff070772 (0xdd16abffff070772): D1821 - EcoSmart A19 RGB bulb (Router)
zigbee2mqtt:info  2020-06-23 11:52:25: Basement water sensor (0x00158d0001d5948f): SJCGQ11LM - Xiaomi Aqara water leak sensor (EndDevice)
zigbee2mqtt:info  2020-06-23 11:52:25: Garage Fridge Temp Sensor (0x00158d000182dd03): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor (EndDevice)
zigbee2mqtt:info  2020-06-23 11:52:25: Kitchen - Motion Sensor (0x00158d0001a24bba): RTCGQ11LM - Xiaomi Aqara human body movement and illuminance sensor (EndDevice)
zigbee2mqtt:info  2020-06-23 11:52:25: Zigbee: disabling joining new devices.
zigbee2mqtt:info  2020-06-23 11:52:25: Adding 'Tradfri Button' to group 'group_1'
zigbee2mqtt:info  2020-06-23 11:52:26: Connecting to MQTT server at mqtt://192.168.0.229
zigbee2mqtt:info  2020-06-23 11:52:26: Connected to MQTT server
zigbee2mqtt:info  2020-06-23 11:52:26: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
[... cut ...] lity }}","state_topic":"zigbee2mqtt/Kitchen - Motion Sensor","json_attributes_topic":"zigbee2mqtt/Kitchen - Motion Sensor","name":"Kitchen - Motion Sensor_linkquality","unique_id":"0x00158d0001a24bba_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d0001a24bba"],"name":"Kitchen - Motion Sensor","sw_version":"Zigbee2mqtt 1.13.1","model":"Aqara human body movement and illuminance sensor (RTCGQ11LM)","manufacturer":"Xiaomi"},"availability_topic":"zigbee2mqtt/bridge/state"}'
```

Koenkk, 2020-06-24:
I'm very confused by this. So when controlling the fan entity in Home Assistant the commands are send to `zigbee2mqtt/Office Fan/set` while in the discovery we clearly see `"command_topic":"zigbee2mqtt/Office Fan/set/fan_state"`

alexanv1, 2020-07-02:
I saw something similar happen as well. My fans would occasionally report fan_state=ON when the actual fan was OFF (and fan_mode was OFF). Physically the fan would turn on and off but the state reported in HA was wrong.

I ended up "fixing" by modifying devices.yaml to add the following for my fans which got the state to be reported correctly in HA.

```
  homeassistant:
    fan:
      state_value_template: '{{ ''ON'' if value_json.fan_mode|lower != ''off'' else ''OFF'' }}'
```

Didn't pursue it further since I had to go back to 1.6 due to other issues and it was working correctly in 1.6.

boojew, 2020-07-06:
Anything further I can provide to be of assistance?

Koenkk, 2020-07-07:
Can you check if it works with the following manual configuration?

```yaml
fan:
  - platform: mqtt
    name: "My Fan"
    state_topic: "zigbee2mqtt/Office Fan"
    state_value_template":"{{ value_json.fan_state }}"
    command_topic: "zigbee2mqtt/Office Fan/set/fan_state"
    speed_state_topic: "zigbee2mqtt/Office Fan"
    speed_command_topic: "zigbee2mqtt/Office Fan/set/fan_mode"
    speed_value_template:"{{ value_json.fan_mode }}"
    json_attributes_topic:"zigbee2mqtt/Office Fan"
    availability_topic:"zigbee2mqtt/bridge/state"
    speeds:
      - "off"
      - low
      - medium
      - high
      - on
      - auto
      - smart
```

boojew, 2020-07-14:
Hmmm.. I get this error with that config stanza. 

`Invalid config for [fan.mqtt]: [state_value_template"] is an invalid option for [fan.mqtt]. Check: fan.mqtt->state_value_template". (See ?, line ?).
`

Note, the yaml was missing some spaces between the colons and quotes
```
fan:
  - platform: mqtt
    name: "My Fan"
    state_topic: "zigbee2mqtt/Office Fan"
    state_value_template": "{{ value_json.fan_state }}"
    command_topic: "zigbee2mqtt/Office Fan/set/fan_state"
    speed_state_topic: "zigbee2mqtt/Office Fan"
    speed_command_topic: "zigbee2mqtt/Office Fan/set/fan_mode"
    speed_value_template: "{{ value_json.fan_mode }}"
    json_attributes_topic: "zigbee2mqtt/Office Fan"
    availability_topic: "zigbee2mqtt/bridge/state"
    speeds:
      - "off"
      - low
      - medium
      - high
      - on
      - auto
      - smart

```

boojew, 2020-07-14:
There was an extra " in the yaml.  If anyone wants it, the correct yaml is below... 

Anyways, I get the same behaviour 

```
fan:
  - platform: mqtt
    name: "My Fan"
    state_topic: "zigbee2mqtt/Office Fan"
    state_value_template: "{{ value_json.fan_state }}"
    command_topic: "zigbee2mqtt/Office Fan/set/fan_state"
    speed_state_topic: "zigbee2mqtt/Office Fan"
    speed_command_topic: "zigbee2mqtt/Office Fan/set/fan_mode"
    speed_value_template: "{{ value_json.fan_mode }}"
    json_attributes_topic: "zigbee2mqtt/Office Fan"
    availability_topic: "zigbee2mqtt/bridge/state"
    speeds:
      - "off"
      - low
      - medium
      - high
      - on
      - auto
      - smart
```

Koenkk, 2020-07-14:
Probably a stupid question (but better to check): are you sure you are using the manual entry instead of the auto discovered one? (does the one you control have "My Fan" as name)?

If you are using the correct one, what do you see in the zigbee2mqtt log when turning on/off?

boojew, 2020-07-14:
Always ask the "stupid questions", i say :) But yea, I am. Here. is a gif of it all.
[image]

boojew, 2020-07-14:
and here are the logs
```

zigbee2mqtt:debug 2020-07-14 15:58:23: Received MQTT message on 'zigbee2mqtt/Office Fan/set/fan_state' with data 'ON'
zigbee2mqtt:debug 2020-07-14 15:58:23: Publishing 'set' 'fan_state' to 'Office Fan'
zigbee2mqtt:debug 2020-07-14 15:58:23: Received MQTT message on 'zigbee2mqtt/Office Fan/set/fan_mode' with data 'medium'
zigbee2mqtt:debug 2020-07-14 15:58:23: Publishing 'set' 'fan_mode' to 'Office Fan'
zigbee2mqtt:info  2020-07-14 15:58:23: MQTT publish: topic 'zigbee2mqtt/Office Fan', payload '{"fan_mode":"ON","fan_state":"ON","last_seen":1594756703942,"linkquality":70,"state":"ON","brightness":0}'
zigbee2mqtt:info  2020-07-14 15:58:24: MQTT publish: topic 'zigbee2mqtt/Office Fan', payload '{"fan_mode":"medium","fan_state":"ON","last_seen":1594756704062,"linkquality":70,"state":"ON","brightness":0}'
zigbee2mqtt:debug 2020-07-14 15:58:27: Received MQTT message on 'zigbee2mqtt/Office Fan/set/fan_state' with data 'ON'
zigbee2mqtt:debug 2020-07-14 15:58:27: Publishing 'set' 'fan_state' to 'Office Fan'
zigbee2mqtt:debug 2020-07-14 15:58:27: Received MQTT message on 'zigbee2mqtt/Office Fan/set/fan_mode' with data 'low'
zigbee2mqtt:debug 2020-07-14 15:58:27: Publishing 'set' 'fan_mode' to 'Office Fan'
zigbee2mqtt:info  2020-07-14 15:58:27: MQTT publish: topic 'zigbee2mqtt/Office Fan', payload '{"fan_mode":"ON","fan_state":"ON","last_seen":1594756707468,"linkquality":70,"state":"ON","brightness":0}'
zigbee2mqtt:info  2020-07-14 15:58:27: MQTT publish: topic 'zigbee2mqtt/Office Fan', payload '{"fan_mode":"low","fan_state":"ON","last_seen":1594756707587,"linkquality":70,"state":"ON","brightness":0}'
zigbee2mqtt:debug 2020-07-14 15:58:31: Received MQTT message on 'zigbee2mqtt/Office Fan/set/fan_state' with data 'ON'
zigbee2mqtt:debug 2020-07-14 15:58:31: Publishing 'set' 'fan_state' to 'Office Fan'
zigbee2mqtt:debug 2020-07-14 15:58:31: Received MQTT message on 'zigbee2mqtt/Office Fan/set/fan_mode' with data 'off'
zigbee2mqtt:debug 2020-07-14 15:58:31: Publishing 'set' 'fan_mode' to 'Office Fan'
zigbee2mqtt:info  2020-07-14 15:58:31: MQTT publish: topic 'zigbee2mqtt/Office Fan', payload '{"fan_mode":"ON","fan_state":"ON","last_seen":1594756711843,"linkquality":70,"state":"ON","brightness":0}'
zigbee2mqtt:info  2020-07-14 15:58:31: MQTT publish: topic 'zigbee2mqtt/Office Fan', payload '{"fan_mode":"off","fan_state":"OFF","last_seen":1594756711960,"linkquality":70,"state":"ON","brightness":0}'
zigbee2mqtt:debug 2020-07-14 15:58:35: Received MQTT message on 'zigbee2mqtt/Office Fan/set/fan_state' with data 'ON'
zigbee2mqtt:debug 2020-07-14 15:58:35: Publishing 'set' 'fan_state' to 'Office Fan'
zigbee2mqtt:info  2020-07-14 15:58:35: MQTT publish: topic 'zigbee2mqtt/Office Fan', payload '{"fan_mode":"ON","fan_state":"ON","last_seen":1594756715364,"linkquality":70,"state":"ON","brightness":0}'
zigbee2mqtt:debug 2020-07-14 15:58:44: Received MQTT message on 'zigbee2mqtt/Office Fan/set/fan_state' with data 'OFF'
zigbee2mqtt:debug 2020-07-14 15:58:44: Publishing 'set' 'fan_state' to 'Office Fan'
zigbee2mqtt:info  2020-07-14 15:58:44: MQTT publish: topic 'zigbee2mqtt/Office Fan', payload '{"fan_mode":"OFF","fan_state":"ON","last_seen":1594756724327,"linkquality":70,"state":"ON","brightness":0}'
zigbee2mqtt:debug 2020-07-14 15:59:53: Saving state to file /app/data/state.json
zigbee2mqtt:debug 2020-07-14 16:00:00: Received MQTT message on 'zigbee2mqtt/bridge/config/devices/get' with data ''
zigbee2mqtt:info  2020-07-14 16:00:00: MQTT publish: topic 'zigbee2mqtt/bridge/config/devices', payload '[{"ieeeAddr":"0x00124b0014d9e40f","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack12","dateCode":"20190425","lastSeen":1594756800006},{"ieeeAddr":"0x00158d00023f5163","type":"EndDevice","networkAddress":31174,"model":"WSDCGQ11LM","vendor":"Xiaomi","description":"Aqara temperature, humidity and pressure sensor","friendly_name":"Living Room - Temperature Sensor","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","modelID":"lumi.weather","hardwareVersion":30,"softwareBuildID":"3000-0001","dateCode":"20161129","lastSeen":1592065887630},{"ieeeAddr":"0x00158d0001d39f43","type":"EndDevice","networkAddress":42864,"model":"MCCGQ11LM","vendor":"Xiaomi","description":"Aqara door & window contact sensor","friendly_name":"Backdoor Contact Sensor","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"Battery","modelID":"lumi.sensor_magnet.aq2","lastSeen":1594756388606},{"ieeeAddr":"0xdd16abfff1a195ed","type":"Router","networkAddress":25701,"model":"D1532","vendor":"EcoSmart","description":"A19 soft white bulb","friendly_name":"Mudroom Lightbulb","manufacturerID":4445,"manufacturerName":"2.1.0_16d_YF05","powerSource":"Unknown","modelID":"\u0000\u0002\u0000\u0004\u0012 �P\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u000e","hardwareVersion":1,"dateCode":"","lastSeen":1594599504744},{"ieeeAddr":"0x0022a3000020c945","type":"EndDevice","networkAddress":23263,"model":"99432","vendor":"Hampton Bay","description":"Universal wink enabled white ceiling fan premier remote control","friendly_name":"Office Fan","manufacturerID":4098,"manufacturerName":"King Of Fans,  Inc.","powerSource":"Unknown","modelID":"HDC52EastwindFan","lastSeen":1594756724327},{"ieeeAddr":"0x00158d00010c2723","type":"EndDevice","networkAddress":51165,"model":"RTCGQ01LM","vendor":"Xiaomi","d [... cut ...]
  "0x00158d000182dd03" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{Garage Fridge Temp Sensor|0x00158d000182dd03 (47753)|Xiaomi MiJia temperature & humidity sensor (WSDCGQ01LM)|18 minutes ago}"];
  "0x00158d000182dd03" -> "0x00124b0014d9e40f" [penwidth=1, weight=0, color="#994444", label="2"]
  "0x00158d0001a24bba" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{Kitchen - Motion Sensor|0x00158d0001a24bba (10025)|Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)|16 minutes, 4 seconds ago}"];
  "0x00158d0001a24bba" -> "0x00124b0014d9e40f" [penwidth=1, weight=0, color="#994444", label="170"]
```

Koenkk, 2020-07-15:
Thanks (and keep up the gifs because it allows me to easily understand what is wrong). Can you try changing https://github.com/Koenkk/zigbee-herdsman-converters/blob/5ffa2fae358b3fd11f825ba34f690736c5d71fa0/converters/toZigbee.js#L1044 to `return {state: {fan_mode: value, fan_state: value.toLowerCase() === 'off' ? 'OFF' : 'ON'}};`. Does it improve things?

boojew, 2020-07-15:
I changed it and now the behavior is that I can turn on the fan, but not turn it off or change the speed. 

```

zigbee2mqtt:info  2020-07-15 09:22:16: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.14.1","commit":"7e46435","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190425}},"log_level":"debug","permit_join":false}'
zigbee2mqtt:debug 2020-07-15 09:22:31: Received MQTT message on 'zigbee2mqtt/Office Fan/set/fan_state' with data 'OFF'
zigbee2mqtt:debug 2020-07-15 09:22:31: Publishing 'set' 'fan_state' to 'Office Fan'
zigbee2mqtt:debug 2020-07-15 09:22:34: Received MQTT message on 'zigbee2mqtt/Office Fan/set/fan_state' with data 'OFF'
zigbee2mqtt:debug 2020-07-15 09:22:34: Publishing 'set' 'fan_state' to 'Office Fan'
zigbee2mqtt:error 2020-07-15 09:22:35: Publish 'set' 'fan_state' to 'Office Fan' failed: 'Error: Write 0x0022a3000020c945/1 hvacFanCtrl({"fanMode":0}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Data request failed with error: 'MAC no ack' (233))'
zigbee2mqtt:debug 2020-07-15 09:22:35: Error: Write 0x0022a3000020c945/1 hvacFanCtrl({"fanMode":0}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Data request failed with error: 'MAC no ack' (233))
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:162:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
zigbee2mqtt:info  2020-07-15 09:22:35: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'fan_state' to 'Office Fan' failed: 'Error: Write 0x0022a3000020c945/1 hvacFanCtrl({\"fanMode\":0}, {\"timeout\":10000,\"disableResponse\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Error: Data request failed with error: 'MAC no ack' (233))'","meta":{"friendly_name":"Office Fan"}}'
zigbee2mqtt:error 2020-07-15 09:22:40: Publish 'set' 'fan_state' to 'Office Fan' failed: 'Error: Write 0x0022a3000020c945/1 hvacFanCtrl({"fanMode":0}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Data request failed with error: 'MAC no ack' (233))'
zigbee2mqtt:debug 2020-07-15 09:22:40: Error: Write 0x0022a3000020c945/1 hvacFanCtrl({"fanMode":0}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Data request failed with error: 'MAC no ack' (233))
    at Endpoint.<anonymous> (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:162:23)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65)
zigbee2mqtt:info  2020-07-15 09:22:40: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'fan_state' to 'Office Fan' failed: 'Error: Write 0x0022a3000020c945/1 hvacFanCtrl({\"fanMode\":0}, {\"timeout\":10000,\"disableResponse\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Error: Data request failed with error: 'MAC no ack' (233))'","meta":{"friendly_name":"Office Fan"}}'
```
[image]

Koenkk, 2020-07-15:
I don't see how this change could lead to the above error, does reverting it fix it?

boojew, 2020-07-15:
It does indeed!

Koenkk, 2020-07-15:
Can you change the body to

```js
const fanMode = common.fanMode[value.toLowerCase()];
const fanModeWrite = fanMode === 0 ? 1 : fanMode;
await entity.write('hvacFanCtrl', {fanMode: fanModeWrite});
return {state: {fan_mode: value, fan_state: value === 'off' ? 'OFF' : 'ON'}};
```

If setting the `low` speed previously worked, I expect this to work, except that when turning off now the low speed will be set.

boojew, 2020-07-16:
Setting the speeds now works as expected.. but the on/off toggle is still an issue.

```
zigbee2mqtt:debug 2020-07-16 11:35:35: Received MQTT message on 'zigbee2mqtt/Office Fan/set/fan_state' with data 'ON'
zigbee2mqtt:debug 2020-07-16 11:35:35: Publishing 'set' 'fan_state' to 'Office Fan'
zigbee2mqtt:debug 2020-07-16 11:35:35: Received MQTT message on 'zigbee2mqtt/Office Fan/set/fan_mode' with data 'off'
zigbee2mqtt:debug 2020-07-16 11:35:35: Publishing 'set' 'fan_mode' to 'Office Fan'
zigbee2mqtt:info  2020-07-16 11:35:35: MQTT publish: topic 'zigbee2mqtt/Office Fan', payload '{"fan_mode":"ON","fan_state":"ON","last_seen":1594913735393,"linkquality":70,"state":"ON","brightness":254}'
zigbee2mqtt:info  2020-07-16 11:35:35: MQTT publish: topic 'zigbee2mqtt/Office Fan', payload '{"fan_mode":"off","fan_state":"OFF","last_seen":1594913735493,"linkquality":70,"state":"ON","brightness":254}'
zigbee2mqtt:debug 2020-07-16 11:35:37: Received MQTT message on 'zigbee2mqtt/Office Fan/set/fan_state' with data 'ON'
zigbee2mqtt:debug 2020-07-16 11:35:37: Publishing 'set' 'fan_state' to 'Office Fan'
zigbee2mqtt:debug 2020-07-16 11:35:37: Received MQTT message on 'zigbee2mqtt/Office Fan/set/fan_mode' with data 'low'
zigbee2mqtt:debug 2020-07-16 11:35:37: Publishing 'set' 'fan_mode' to 'Office Fan'
zigbee2mqtt:info  2020-07-16 11:35:38: MQTT publish: topic 'zigbee2mqtt/Office Fan', payload '{"fan_mode":"ON","fan_state":"ON","last_seen":1594913738068,"linkquality":70,"state":"ON","brightness":254}'
zigbee2mqtt:info  2020-07-16 11:35:38: MQTT publish: topic 'zigbee2mqtt/Office Fan', payload '{"fan_mode":"low","fan_state":"ON","last_seen":1594913738249,"linkquality":70,"state":"ON","brightness":254}'
zigbee2mqtt:debug 2020-07-16 11:35:40: Received MQTT message on 'zigbee2mqtt/Office Fan/set/fan_state' with data 'OFF'
zigbee2mqtt:debug 2020-07-16 11:35:40: Publishing 'set' 'fan_state' to 'Office Fan'
zigbee2mqtt:info  2020-07-16 11:35:40: MQTT publish: topic 'zigbee2mqtt/Office Fan', payload '{"fan_mode":"OFF","fan_state":"ON","last_seen":1594913740308,"linkquality":70,"state":"ON","brightness":254}'
zigbee2mqtt:debug 2020-07-16 11:35:41: Received MQTT message on 'zigbee2mqtt/Office Fan/set/fan_state' with data 'OFF'
zigbee2mqtt:debug 2020-07-16 11:35:41: Publishing 'set' 'fan_state' to 'Office Fan'
zigbee2mqtt:info  2020-07-16 11:35:41: MQTT publish: topic 'zigbee2mqtt/Office Fan', payload '{"fan_mode":"OFF","fan_state":"ON","last_seen":1594913741630,"linkquality":70,"state":"ON","brightness":254}'
```

Koenkk, 2020-07-16:
And with:

```js
const fanMode = common.fanMode[value.toLowerCase()];
const fanModeWrite = fanMode === 0 ? 1 : fanMode;
await entity.write('hvacFanCtrl', {fanMode: fanModeWrite});
return {state: {fan_mode: value, fan_state: value.toLowerCase() === 'off' ? 'OFF' : 'ON'}};
```

boojew, 2020-07-16:
Sooo.. I get some very interesting behaviour with this code...  

So assuming the fan is initially OFF, if I turn it, it goes to high; but does not set the speed in hass... When I turn it off, it sets the fan to low... Speed indications are not reliably sent to HASS... happens in both "My Fan" and "Office Fan"
[image]

```
zigbee2mqtt:debug 2020-07-16 16:40:55: Received MQTT message on 'zigbee2mqtt/Office Fan/set/fan_state' with data 'OFF'
zigbee2mqtt:debug 2020-07-16 16:40:55: Publishing 'set' 'fan_state' to 'Office Fan'
zigbee2mqtt:info  2020-07-16 16:40:55: MQTT publish: topic 'zigbee2mqtt/Office Fan', payload '{"fan_mode":"OFF","fan_state":"OFF","last_seen":1594932055856,"linkquality":70,"state":"ON","brightness":254}'
zigbee2mqtt:debug 2020-07-16 16:41:07: Received Zigbee message from 'Office - Temperature Sensor', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2808}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-07-16 16:41:07: MQTT publish: topic 'zigbee2mqtt/Office - Temperature Sensor', payload '{"temperature":28.08,"linkquality":113,"humidity":33.42,"pressure":1012.4,"battery":86,"voltage":2975,"last_seen":1594932067355}'
zigbee2mqtt:debug 2020-07-16 16:41:32: Received MQTT message on 'zigbee2mqtt/Office Fan/set/fan_state' with data 'ON'
zigbee2mqtt:debug 2020-07-16 16:41:32: Publishing 'set' 'fan_state' to 'Office Fan'
zigbee2mqtt:info  2020-07-16 16:41:32: MQTT publish: topic 'zigbee2mqtt/Office Fan', payload '{"fan_mode":"ON","fan_state":"ON","last_seen":1594932092829,"linkquality":70,"state":"ON","brightness":254}'
zigbee2mqtt:debug 2020-07-16 16:41:40: Received MQTT message on 'zigbee2mqtt/Office Fan/set/fan_state' with data 'OFF'
zigbee2mqtt:debug 2020-07-16 16:41:40: Publishing 'set' 'fan_state' to 'Office Fan'
zigbee2mqtt:info  2020-07-16 16:41:40: MQTT publish: topic 'zigbee2mqtt/Office Fan', payload '{"fan_mode":"OFF","fan_state":"OFF","last_seen":1594932100568,"linkquality":70,"state":"ON","brightness":254}'
zigbee2mqtt:debug 2020-07-16 16:42:12: Received Zigbee message from 'Office Fan', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"manufacturerCode":4098,"imageType":0,"fileVersion":15}' from endpoint 1 with groupID 0

```

Koenkk, 2020-07-16:
And with:

```js
const fanMode = common.fanMode[value.toLowerCase()];
const fanModeWrite = fanMode === 0 ? 1 : fanMode;
await entity.write('hvacFanCtrl', {fanMode: fanModeWrite});
return {state: {fan_mode: value.toLowerCase(), fan_state: value.toLowerCase() === 'off' ? 'OFF' : 'ON'}};
```

boojew, 2020-07-17:
MUCH closer now. 

Power On = Fan high
low,med,high do exactly what they should... 
Power off = does nothing

State and speed are actually represented. EXPECT off which shows this, but the fan is still on.
[image]

```

zigbee2mqtt:debug 2020-07-17 11:17:15: Received MQTT message on 'zigbee2mqtt/Office Fan/set/fan_state' with data 'ON'
zigbee2mqtt:debug 2020-07-17 11:17:15: Publishing 'set' 'fan_state' to 'Office Fan'
zigbee2mqtt:debug 2020-07-17 11:17:15: Received MQTT message on 'zigbee2mqtt/Office Fan/set/fan_mode' with data 'medium'
zigbee2mqtt:debug 2020-07-17 11:17:15: Publishing 'set' 'fan_mode' to 'Office Fan'
zigbee2mqtt:info  2020-07-17 11:17:16: MQTT publish: topic 'zigbee2mqtt/Office Fan', payload '{"fan_mode":"on","fan_state":"ON","last_seen":1594999036111,"linkquality":70,"state":"ON","brightness":254}'
zigbee2mqtt:info  2020-07-17 11:17:16: MQTT publish: topic 'zigbee2mqtt/Office Fan', payload '{"fan_mode":"medium","fan_state":"ON","last_seen":1594999036268,"linkquality":70,"state":"ON","brightness":254}'
zigbee2mqtt:debug 2020-07-17 11:17:18: Received Zigbee message from 'Kitchen - Motion Sensor', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3025,"3":28,"4":17320,"5":51,"6":[1,17694738],"10":25701,"11":32,"100":0}}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-07-17 11:17:18: MQTT publish: topic 'zigbee2mqtt/Kitchen - Motion Sensor', payload '{"illuminance":32,"linkquality":23,"occupancy":false,"battery":100,"voltage":3025,"last_seen":1594999038565,"illuminance_lux":32}'
zigbee2mqtt:debug 2020-07-17 11:17:18: Received Zigbee message from 'Kitchen - Motion Sensor', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3025,"3":28,"4":17320,"5":51,"6":[1,17694738],"10":25701,"11":32,"100":0}}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-07-17 11:17:18: MQTT publish: topic 'zigbee2mqtt/Kitchen - Motion Sensor', payload '{"illuminance":32,"linkquality":23,"occupancy":false,"battery":100,"voltage":3025,"last_seen":1594999038670,"illuminance_lux":32}'
zigbee2mqtt:debug 2020-07-17 11:17:18: Received Zigbee message from 'Kitchen - Motion Sensor', type 'attributeReport', cluster 'genBasic', data '{"65281":{"1":3025,"3":28,"4":17320,"5":51,"6":[1,17694738],"10":25701,"11":32,"100":0}}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-07-17 11:17:18: MQTT publish: topic 'zigbee2mqtt/Kitchen - Motion Sensor', payload '{"illuminance":32,"linkquality":23,"occupancy":false,"battery":100,"voltage":3025,"last_seen":1594999038934,"illuminance_lux":32}'
zigbee2mqtt:debug 2020-07-17 11:17:19: Received MQTT message on 'zigbee2mqtt/Office Fan/set/fan_state' with data 'OFF'
zigbee2mqtt:debug 2020-07-17 11:17:19: Publishing 'set' 'fan_state' to 'Office Fan'
zigbee2mqtt:info  2020-07-17 11:17:19: MQTT publish: topic 'zigbee2mqtt/Office Fan', payload '{"fan_mode":"off","fan_state":"OFF","last_seen":1594999039991,"linkquality":70,"state":"ON","brightness":254}'
zigbee2mqtt:debug 2020-07-17 11:17:23: Saving state to file /app/data/state.json

```

Koenkk, 2020-07-18:
And with:

```js
const fanMode = common.fanMode[value.toLowerCase()];
await entity.write('hvacFanCtrl', {fanMode});
return {state: {fan_mode: value.toLowerCase(), fan_state: value.toLowerCase() === 'off' ? 'OFF' : 'ON'}};
```

boojew, 2020-07-20:
Success! I think we're perfect now.

Koenkk, 2020-07-20:
@boojew to confirm; with this change everything is working as expected, even the auto discovered home asisstant entry?

boojew, 2020-07-23:
Sorry. I missed this. Yes, it's 100% working properly w/ the auto discovered entity

Koenkk, 2020-07-23:
Thanks, changed the code, assuming this can be closed now.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-07-23 by Koenkk