Koenkk/zigbee2mqtt #3770: Device paired not successfully while support a new contact sensor

zijings, 2020-06-18:
# Bug Report
## What happened
I have tried to support three new device, the other two(a temperature & humidity sensor and a wireless switch) was paired successfully and everything was fine. But the contact sensor could not be paired and interview successfully. 
Though the device could report message(the contact is ture or false) and the z2m could receive, translate and publish to the MQTT rightly. So it seemed the device has joined the network. I have tried to repaired for many times but it didn't work.
## What did you expect to happen
Successfully interview the new device and it is paired successfully
## How to reproduce it (minimal and precise)
add the device to the network
## Debug Info
zigbee2mqtt:info  2020-06-18 10:53:05: Device '0x00124b0018c1c914' joined
zigbee2mqtt:info  2020-06-18 10:53:05: Starting interview of '0x00124b0018c1c914'
zigbee2mqtt:info  2020-06-18 10:53:05: MQTT publish: topic 'z2m_1000_1/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00124b0018c1c914"}}'
zigbee2mqtt:info  2020-06-18 10:53:05: MQTT publish: topic 'z2m_1000_1/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00124b0018c1c914"}}'
zigbee2mqtt:debug 2020-06-18 10:53:07: Device '0x00124b0018c1c914' announced itself
zigbee2mqtt:info  2020-06-18 10:53:07: MQTT publish: topic 'z2m_1000_1/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00124b0018c1c914"}}'
zigbee2mqtt:debug 2020-06-18 10:53:07: Received Zigbee message from '0x00124b0018c1c914', type 'readResponse', cluster 'genBasic', data '{"modelId":"MULTI-MECI--EA01","manufacturerName":"Shuncom         "}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-18 10:53:07: No converter available for 'MQMCCGQ001' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"MULTI-MECI--EA01","manufacturerName":"Shuncom         "}'
zigbee2mqtt:debug 2020-06-18 10:53:07: Received Zigbee message from '0x00124b0018c1c914', type 'readResponse', cluster 'genBasic', data '{"powerSource":3,"zclVersion":2}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-18 10:53:07: No converter available for 'MQMCCGQ001' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":3,"zclVersion":2}'
zigbee2mqtt:debug 2020-06-18 10:53:07: Received Zigbee message from '0x00124b0018c1c914', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-18 10:53:07: No converter available for 'MQMCCGQ001' with cluster 'genBasic' and type 'readResponse' and data '{}'
zigbee2mqtt:debug 2020-06-18 10:53:07: Received Zigbee message from '0x00124b0018c1c914', type 'readResponse', cluster 'genBasic', data '{"hwVersion":2,"dateCode":"20060831        "}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-18 10:53:07: No converter available for 'MQMCCGQ001' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":2,"dateCode":"20060831        "}'
zigbee2mqtt:debug 2020-06-18 10:53:08: Received Zigbee message from '0x00124b0018c1c914', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"MULTI-MECI-V2.0-PB"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-18 10:53:08: No converter available for 'MQMCCGQ001' with cluster 'genBasic' and type 'readResponse' and data '{"swBuildId":"MULTI-MECI-V2.0-PB"}'
zigbee2mqtt:debug 2020-06-18 10:53:08: Received Zigbee message from '0x00124b0018c1c914', type 'readResponse', cluster 'ssIasZone', data '{"zoneState":0}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-18 10:53:08: No converter available for 'MQMCCGQ001' with cluster 'ssIasZone' and type 'readResponse' and data '{"zoneState":0}'
zigbee2mqtt:debug 2020-06-18 10:53:10: Received Zigbee message from '0x00124b0018c1c914', type 'readResponse', cluster 'ssIasZone', data '{"zoneState":0}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-18 10:53:10: No converter available for 'MQMCCGQ001' with cluster 'ssIasZone' and type 'readResponse' and data '{"zoneState":0}'
zigbee2mqtt:error 2020-06-18 10:53:10: Failed to interview '0x00124b0018c1c914', device has not successfully been paired
zigbee2mqtt:info  2020-06-18 10:53:10: MQTT publish: topic 'z2m_1000_1/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0x00124b0018c1c914"}}'
zigbee2mqtt:debug 2020-06-18 10:53:13: Received Zigbee message from '0x00124b0018c1c914', type 'attributeReport', cluster 'genBasic', data '{"modelId":"MULTI-MECI--EA01"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-18 10:53:13: No converter available for 'MQMCCGQ001' with cluster 'genBasic' and type 'attributeReport' and data '{"modelId":"MULTI-MECI--EA01"}'
zigbee2mqtt:debug 2020-06-18 10:53:13: Received Zigbee message from '0x00124b0018c1c914', type 'attributeReport', cluster 'genBasic', data '{"swBuildId":"MULTI-MECI-V2.0-PB"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-18 10:53:13: No converter available for 'MQMCCGQ001' with cluster 'genBasic' and type 'attributeReport' and data '{"swBuildId":"MULTI-MECI-V2.0-PB"}'
zigbee2mqtt:debug 2020-06-18 10:53:13: Received Zigbee message from '0x00124b0018c1c914', type 'raw', cluster '65024', data '{"type":"Buffer","data":[28,3,18,2,10,3,0,33,64,56]}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-18 10:53:13: No converter available for 'MQMCCGQ001' with cluster '65024' and type 'raw' and data '{"type":"Buffer","data":[28,3,18,2,10,3,0,33,64,56]}'
zigbee2mqtt:debug 2020-06-18 10:53:13: Received Zigbee message from '0x00124b0018c1c914', type 'attributeReport', cluster 'ssIasZone', data '{"zoneType":21}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-18 10:53:13: No converter available for 'MQMCCGQ001' with cluster 'ssIasZone' and type 'attributeReport' and data '{"zoneType":21}'
zigbee2mqtt:debug 2020-06-18 10:53:13: Received Zigbee message from '0x00124b0018c1c914', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"zonestatus":5,"extendedstatus":0}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-06-18 10:53:13: MQTT publish: topic 'z2m_1000_1/0x00124b0018c1c914', payload '{"contact":false,"warn":true,"linkquality":119}'

Koenkk, 2020-06-18:
Can you provide the herdsman debug logging of the failing pair attempt?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

closed 2020-07-26 by stale[bot]

linoul, 2021-05-04:
> Can you provide the herdsman debug logging of the failing pair attempt?
> 
> To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

zigbee2mqtt version: 1.18.1-1

`2021-05-04T13:31:45.034Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,26,69,132,121,141,0,121,141,20,1,4,1,2,4,0,5,0,0,1,0,3,0,0,5,0,0,1,6,0,200]
2021-05-04T13:31:45.034Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,26,69,132,121,141,0,121,141,20,1,4,1,2,4,0,5,0,0,1,0,3,0,0,5,0,0,1,6,0,200]
2021-05-04T13:31:45.034Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 26 - 2 - 5 - 132 - [121,141,0,121,141,20,1,4,1,2,4,0,5,0,0,1,0,3,0,0,5,0,0,1,6,0] - 200
2021-05-04T13:31:45.035Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - simpleDescRsp - {"srcaddr":36217,"status":0,"nwkaddr":36217,"len":20,"endpoint":1,"profileid":260,"deviceid":1026,"deviceversion":0,"numinclusters":5,"inclusterlist":[0,1,3,1280,0],"numoutclusters":1,"outclusterlist":[6]}
2021-05-04T13:31:45.035Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-05-04T13:31:45.035Z zigbee-herdsman:controller:device:log Interview - got simple descriptor for endpoint '1' device '0x00124b001c491c40'
2021-05-04T13:31:45.037Z zigbee-herdsman:controller:endpoint Read 0x00124b001c491c40/1 genBasic(["appVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2021-05-04T13:31:45.038Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x00124b001c491c40:36217/1 (0,0,1)
2021-05-04T13:31:45.038Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":36217,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":50,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,43,0,1,0]}}
2021-05-04T13:31:45.038Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,121,141,1,1,0,0,50,0,30,5,16,43,0,1,0,205]
2021-05-04T13:31:45.044Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2021-05-04T13:31:45.045Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2021-05-04T13:31:45.045Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2021-05-04T13:31:45.045Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2021-05-04T13:31:45.045Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-05-04T13:31:45.168Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,26,68,129,0,0,0,0,121,141,1,1,0,30,0,147,119,55,0,0,6,24,43,1,1,0,134,121,141,29,188]
2021-05-04T13:31:45.168Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,26,68,129,0,0,0,0,121,141,1,1,0,30,0,147,119,55,0,0,6,24,43,1,1,0,134,121,141,29,188]
2021-05-04T13:31:45.168Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 26 - 2 - 4 - 129 - [0,0,0,0,121,141,1,1,0,30,0,147,119,55,0,0,6,24,43,1,1,0,134,121,141,29] - 188
2021-05-04T13:31:45.169Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":0,"srcaddr":36217,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":30,"securityuse":0,"timestamp":3635091,"transseqnumber":0,"len":6,"data":{"type":"Buffer","data":[24,43,1,1,0,134]}}
2021-05-04T13:31:45.169Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":43,"manufacturerCode":null,"commandIdentifier":1},"Payload":[{"attrId":1,"status":134}],"Command":{"ID":1,"name":"readRsp","parameters":[{"name":"attrId","type":33},{"name":"status","type":32},{"name":"dataType","type":32,"conditions":[{"type":"statusEquals","value":0}]},{"name":"attrData","type":1000,"conditions":[{"type":"statusEquals","value":0}]}]}},"address":36217,"endpoint":1,"linkquality":30,"groupID":0}'
Zigbee2MQTT:debug 2021-05-04 21:31:45: Received Zigbee message from '0x00124b001c491c40', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
2021-05-04T13:31:45.171Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-05-04T13:31:45.285Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,26,68,129,0,0,0,0,121,141,1,1,0,27,0,80,149,55,0,0,6,24,43,1,1,0,134,121,141,29,152]
2021-05-04T13:31:45.286Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,26,68,129,0,0,0,0,121,141,1,1,0,27,0,80,149,55,0,0,6,24,43,1,1,0,134,121,141,29,152]
2021-05-04T13:31:45.286Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 26 - 2 - 4 - 129 - [0,0,0,0,121,141,1,1,0,27,0,80,149,55,0,0,6,24,43,1,1,0,134,121,141,29] - 152
2021-05-04T13:31:45.286Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":0,"srcaddr":36217,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":27,"securityuse":0,"timestamp":3642704,"transseqnumber":0,"len":6,"data":{"type":"Buffer","data":[24,43,1,1,0,134]}}
2021-05-04T13:31:45.287Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":43,"manufacturerCode":null,"commandIdentifier":1},"Payload":[{"attrId":1,"status":134}],"Command":{"ID":1,"name":"readRsp","p [... cut ...]
2021-05-04T13:31:45.650Z zigbee-herdsman:controller:error Interview failed for '0x00124b001c491c40 with error 'Error: Read 0x00124b001c491c40/1 ssIasZone(["iasCieAddr","zoneState"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
Zigbee2MQTT:error 2021-05-04 21:31:45: Failed to interview '0x00124b001c491c40', device has not successfully been paired`