Koenkk/zigbee2mqtt #3771: How to write reporting interval setting to a device?

DatafoxChris, 2020-06-18:
I know i need to set the manufacturer field to 0x1136, but how do i send the reporting-interval setting to the device through MQTT ?

Koenkk, 2020-06-18:
Reporting interval cannot be set through MQTT. Currently "reportings" are configured in the `confiugre()` method, good example: https://github.com/Koenkk/zigbee-herdsman-converters/blob/a009dfc5c5c2079e80cafabec1ef4e488923e533/devices.js#L2893

DatafoxChris, 2020-06-18:
Clear. How to set amount of seconds ? I would like to configure 30 seconds, not 1 minute. 
After changing devices.js i would need to remove and re-add the device to the zigbee network ?

Koenkk, 2020-06-19:
`minimumReportInterval: 30,`.

There are two ways to trigger a reconfigure:
- By bumping the `configureKey` by one (`configureKey: 1` -> `configureKey: 2`)
- With https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure

DatafoxChris, 2020-06-19:
Changed the code in devices.js as follows:

            await endpoint.configureReporting('seMetering', [{
                attribute: {ID: 0x2000, type: 0x29},
                minimumReportInterval: 10,
                maximumReportInterval: repInterval.MINUTES_5,
                reportableChange: 1,
            }], options);

Using MQTT Explorer sent command:

Topic: zigbee2mqtt/bridge/configure
RAW: Meazon Meter 1

Result:
Jun 19 09:26:44 zigbee2mqtt npm[20332]: zigbee2mqtt:info  2020-06-19 09:26:44: Configuring 'Meazon Meter 1'
Jun 19 09:26:45 zigbee2mqtt npm[20332]: zigbee2mqtt:error 2020-06-19 09:26:45: Failed to configure 'Meazon Meter 1', attempt 1 (Error: Write 0x00124b0002cc3bab/10 seMetering({"4101":{"value":32767,"type":25}}, {"timeout":10000,"manufacturerCode":4406,"disableDefaultResponse":true}) failed (Error: Status 'FAILURE')
Jun 19 09:26:45 zigbee2mqtt npm[20332]:     at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:158:23)
Jun 19 09:26:45 zigbee2mqtt npm[20332]:     at Generator.next (<anonymous>)
Jun 19 09:26:45 zigbee2mqtt npm[20332]:     at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:5:58))

Koenkk, 2020-06-20:
The configure fails before the configureReporting is executed (at `await endpoint.write('seMetering', {0x1005: {value: 0x063e, type: 25}}, options);`). For some reason the devices doesn't accept this write.

closed 2020-07-27 by stale[bot]