Koenkk/zigbee2mqtt #3772: Ebyte E18-TBH-01 USB dongle

Boia11, 2020-06-18:
Hi!
I buy an [Ebyte E18-TBH-01 USB dongle](http://www.ebyte.com/en/product-view-news.aspx?id=949) with [E18-MS1PA1-PCB](http://www.ebyte.com/en/product-view-news.aspx?id=121) onboard and CH340G as USB serial.
I solder 5 pin and flash the E18-MS1PA1-PCB with [Z-Stack_3.0.x Coordinator Firmware](coordinator/Z-Stack_3.0.x/bin/CC2530_CC2592_20190523.zip).
Flash without problem using CC Debugger ([connection image](https://user-images.githubusercontent.com/8360230/57625100-e098c600-759b-11e9-81cb-c7034fac2e4b.png)) and SmartRF Flash Programmer.
But when I try to use with zigbee2mqtt not start, where am I wrong?
This is the log:
```
info  2020-06-18 20:57:09: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-06-18.20-57-09' filename: log.txt
info  2020-06-18 20:57:10: Starting zigbee2mqtt version 1.14.0 (commit #9009de2)
info  2020-06-18 20:57:10: Starting zigbee-herdsman...
error 2020-06-18 20:57:10: Error while starting zigbee-herdsman
```
**configuration.yaml**
```
# Home Assistant integration (MQTT discovery)
homeassistant: false

# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://localhost'
  # MQTT server authentication, uncomment if required:
  # user: my_user
  # password: my_password

# Serial settings
serial:
  port: /dev/ttyUSB0
advanced:
  baudrate: 115200
  rtscts: false
```
**dmesg | grep tty**
`[   16.524281] usb 4-2: ch341-uart converter now attached to ttyUSB0`

## Debug Info
Zigbee2mqtt version: 1.14.0
Adapter hardware: CC2530 + CC2592
Adapter firmware version: CC2530_CC2592_20190523 (Z-Stack_3.0.x - Coordinator)

chrisdaloa, 2020-06-18:
I have a similar issue:
same with an E18-MS1-PCB USB DONGLE, the raspberry recognize it but zigbee2mqtt does not start

`zigbee2mqtt:error 2020-06-18 22:24:23: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-06-18 22:24:23: Failed to start zigbee
zigbee2mqtt:error 2020-06-18 22:24:23: Exiting...
zigbee2mqtt:error 2020-06-18 22:24:23: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/zigbee2mqtt-1.14.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:72:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.14.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:6:65)
npm`

sjorge, 2020-06-18:
Since it is based on CH340G, I would try setting:
```
advanced:
  rtscts: false
```

Boia11, 2020-06-18:
> Since it is based on CH340G, I would try setting:
> ```
> advanced:
>   rtscts: false
> ```
> 
Yes, but I set already this option in configuration. :
`# Serial settings
serial:
  port: /dev/ttyUSB0
advanced:
  baudrate: 115200
  rtscts: false`

MPM1107, 2020-06-19:
According to the documentation of your USB stick the RX and TX of the usb-serial converter are connected to P1.5 and P1.4 on the CC2530. The precompiled firmware from Koen's github has them set to P0.2 and P0.3. You will have to compile your own firmware with the pins set properly for your stick, or use some jumper wires to change the connections on your board.

Boia11, 2020-06-19:
> According to the documentation of your USB stick the RX and TX of the usb-serial converter are connected to P1.5 and P1.4 on the CC2530. The precompiled firmware from Koen's github has them set to [...]

Thanks for info...I'll try and update you

chrisdaloa, 2020-06-19:
Hi, I flashed with the koenk cc2530 coordinator (simple, not +2592), then I connected the rx to p03 and tx to p02 and it works!!

Now it should be great to have a firmware deditcated with pin p14 and p15!

chrisdaloa, 2020-06-19:
Thi is the connection

ps: @Boia11  I'm Italian  :)

Boia11, 2020-06-19:
> 
> 
> Hi, I flashed with the koenk cc2530 coordinator (simple, not +2592), then I connected the rx to p03 and tx to p02 and it works!!
> 
> Now it should be great to have a firmware deditcated with pin p14 and p15!

Yes......I'll share custom Z-Stack-firmware for CC2530_CC2592 for this board and other with usb-serial converter connected to P1.5(RX) and P1.4(TX)

chrisdaloa, 2020-06-19:
Yes......I'll share custom Z-Stack-firmware for CC2530_CC2592 for this board and other with usb-serial converter connected to P1.5(RX) and P1.4(TX)

Great! Please post link here thanks a lot!

Boia11, 2020-06-19:
> 
> 
> > Yes......I'll share custom Z-Stack-firmware for CC2530_CC2592 for this board and other with usb-serial converter connected to P1.5(RX) and P1.4(TX)
> 
> Great! Please post link here thanks a lot!

oh yes, but I need time to understand how to do this change hahah

chrisdaloa, 2020-06-20:
@Boia11 I think that the koenkk firmware has the wrong radio connection for this module. The blue led stay off after flashing with firmware. Have you tried?

Boia11, 2020-06-20:
> @Boia11 I think that the koenkk firmware has the wrong radio connection for this module. The blue led stay off after flashing with firmware. Have you tried?

Yes....because the led pin and uart pin is complete different from koenkk firmware.
I try to look the source code to compile and it isn't simple to understand what modify 😅
But I'll try

chrisdaloa, 2020-06-21:
> Yes....because the led pin and uart pin is complete different from koenkk firmware.
> I try to look the source code to compile and it isn't simple to understand what modify 😅
> But I'll try

Have you issues with radio? Have you tested the range? The radio module seems to be off.

MPM1107, 2020-06-21:
There's a link to some firmware in this issue, maybe you could try that out: [https://github.com/Koenkk/zigbee2mqtt/issues/1437](https://github.com/Koenkk/zigbee2mqtt/issues/1437)

Boia11, 2020-06-21:
> 
> 
> There's a link to some firmware in this issue, maybe you could try that out: #1437

Those firmware are based on Z-Stack_Home_1.2.x I want to create a firmware based on Z-Stack_3.0.x

Boia11, 2020-06-21:
Koenkk/Z-Stack-firmware#183

pedroborges1, 2020-07-02:
good, how to flash the Ebyte E18-TBH-01 USB dongle ?

Boia11, 2020-07-02:
> 
> 
> good, how to flash the Ebyte E18-TBH-01 USB dongle ?

You need to solder 5 pin to the board like this picture: 
[image]
For flash I use a CC Debugger of Texas Instruments and SmartRF Flash Programmer.
[image]

Boia11, 2020-07-02:
@chrisdaloa I try to change led configuration but without results :weary:. Have you try to do something?
I open thread on forum of TI E2E Support but but honestly I didn't understand much more :sweat_smile:.
Link: [Edit Pin and UART PIN in project Z-Stack 3.0.2 [CC2530+CC2592]](https://e2e.ti.com/support/wireless-connectivity/zigbee-and-thread/f/158/p/916989/3390541)

pedroborges1, 2020-07-03:
thanks, 
flash you what firmware at the  Ebyte E18-TBH-01 USB dongle ?

Boia11, 2020-07-06:
> 
> 
> thanks,
> flash you what firmware at the Ebyte E18-TBH-01 USB dongle ?

You find all details in the [OP](https://github.com/Koenkk/zigbee2mqtt/issues/3772#issue-641507323) here

pedroborges1, 2020-07-15:
[image]

good, 
found this image of cc debbuger.

on the target voltage sense pin I have to have 3.3 volts too ?

Boia11, 2020-07-15:
> 
> 
> [image]
> 
> good,
> found this image of cc debbuger.
> 
> on the target voltage sense pin I have to have 3.3 volts too ?

yes! you need to connect like this [image](https://github.com/Koenkk/zigbee2mqtt/issues/3772#issuecomment-653040769)

pedroborges1, 2020-07-15:
so it is necessary to chant the 3.3v pin with the target sense voltage pin when it is to flash certain?

Boia11, 2020-07-15:
> 
> 
> so it is necessary to chant the 3.3v pin with the target sense voltage pin when it is to flash certain?

The "target sense voltage pin" its a secure element for say to CC Debugger that use 3,3v for logic interface.
If you don't connect it, you do it at your own risk. Read the TI documentation of CC Debugger for all information about connection and flash.

svh1985, 2020-07-24:
I'm wondering why can't we just use the onboard CH340G to flash the board? Should be possible right? It's widely used to program ESP's.

pedroborges1, 2020-07-26:
after flashing the Ebyte E18-TBH-01 USB dongle you need to do some kind of configuration in the configuration.yaml ?

Boia11, 2020-08-31:
> 
> 
> after flashing the Ebyte E18-TBH-01 USB dongle you need to do some kind of configuration in the configuration.yaml ?

I use zigbee2mqtt add-on with this config:
```
data_path: /share/zigbee2mqtt
devices: devices.yaml
groups: groups.yaml
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://core-mosquitto'
  user: '!secret mqtt_username'
  password: '!secret mqtt_password'
serial:
  port: /dev/ttyUSB0
advanced:
  baudrate: 115200
  rtscts: false
  pan_id: 6754
  channel: 11
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  availability_blacklist: []
ban: []
whitelist: []
queue: {}
socat:
  enabled: false
  master: 'pty,raw,echo=0,link=/dev/ttyZ2M,mode=777'
  slave: 'tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5'
  restartdelay: 1
  initialdelay: 1
  options: '-d -d'
  log: false
```

closed 2020-10-09 by github-actions[bot]

KameDomotics, 2022-01-24:
> > good, how to flash the Ebyte E18-TBH-01 USB dongle ?
> 
> You need to solder 5 pin to the board like this picture: [image] For flash I use a CC Debugger of Texas Instruments and SmartRF Flash Programmer. [image]

Hi! 
Is it possible to flash the E18-MS1-PCB module without using the USB board?  I mean connecting the CC Debugger directly to the E18-MS1-PCB module.

KameDomotics, 2022-01-24:
[image]

KameDomotics, 2022-01-24:
these should be the connecting pins
[image]

Boia11, 2022-01-24:
> Hi! Is it possible to flash the E18-MS1-PCB module without using the USB board? I mean connecting the CC Debugger directly to the E18-MS1-PCB module.

Yes, it's possible if you connect direct to pin or soldered pad

GianmarcoCA, 2022-03-21:
Hi, I have bought the E18-BTMS1PA2 from ebyte, but I can't find a coordinator firmware for it. is that I want to use it as a gateway with esp 32. I only saw it was from vietnam EPCB (https://epcb.vn/blogs/news/esp32-zigbee-gateway-bai-1-ket-noi-esp32-voi-zigbee-module)  , but they use the E18-BTMS1IPX. CAN SOMEONE HELP ME WITH THIS?

Boia11, 2022-03-21:
> Hi, I have bought the E18-BTMS1PA2 from ebyte, but I can't find a coordinator firmware for it. is that I want to use it as a gateway with esp 32. I only saw it was from vietnam EPCB [...]

Mmmh nope, sorry. I don't know.