Koenkk/zigbee2mqtt #3773: HEIMAN HS1SA-E smoke detector - wrong detection and no alarm messages

Stalkk, 2020-06-18:
Hi all,

i just got a new HS1SA smoke detector and have trouble getting it to work properly. Pairing is ok and the battery and linkquality messages too. But if i press the test button or activate the alarm with some smoke no messages or log entries are generated.

The device properties:

- The device label on a battery slot means it is the model HS1SA-E, but it sends a "SmokeSensor-N-3.0’ so the zigbee2mqtt detects it (wrong?) as a HS3SA.
- The version info: Brige: “version”:“1.14.0”,“commit”:“9009de2”
- Coordinator firmware version: ‘{“type”:“zStack12”,“meta”:{“transportrev”:2,“product”:0,“majorrel”:2,“minorrel”:0,“maintrel”:0,“revision”:""}}’

and the log of device pairing:

```
info 2020-06-17 15:20:04: Device ‘0xXXXXXXXXXXX’ joined
info 2020-06-17 15:20:04: Starting interview of ‘0xXXXXXXXXXXX’
info 2020-06-17 15:20:04: MQTT publish: topic ‘zigbee2mqtt/bridge/log’, payload ‘{“type”:“device_connected”,“message”:{“friendly_name”:“0xXXXXXXXXXXX”}}’
info 2020-06-17 15:20:04: MQTT publish: topic ‘zigbee2mqtt/bridge/log’, payload '{“type”:“pairing”,“message”:“interview_started”,“meta”:{“friendly_name”:"0xXXXXXXXXXXX$
debug 2020-06-17 15:20:04: Device ‘0xXXXXXXXXXXX’ announced itself
info 2020-06-17 15:20:04: MQTT publish: topic ‘zigbee2mqtt/bridge/log’, payload '{“type”:“device_announced”,“message”:“announce”,“meta”:{“friendly_name”:"0xXXXXXXXXXXX$
debug 2020-06-17 15:20:17: Received Zigbee message from ‘0xXXXXXXXXXXX’, type ‘readResponse’, cluster ‘genBasic’, data ‘{“modelId”:“SmokeSensor-N-3.0”}’ from endpo$
debug 2020-06-17 15:20:17: No converter available for ‘HS3SA’ with cluster ‘genBasic’ and type ‘readResponse’ and data ‘{“modelId”:“SmokeSensor-N-3.0”}’
debug 2020-06-17 15:20:19: Received Zigbee message from ‘0xXXXXXXXXXXX’, type ‘readResponse’, cluster ‘genBasic’, data ‘{“manufacturerName”:“HEIMAN”}’ from endpoin$
debug 2020-06-17 15:20:19: No converter available for ‘HS3SA’ with cluster ‘genBasic’ and type ‘readResponse’ and data ‘{“manufacturerName”:“HEIMAN”}’
debug 2020-06-17 15:20:22: Received Zigbee message from ‘0xXXXXXXXXXXX’, type ‘readResponse’, cluster ‘genBasic’, data ‘{“powerSource”:3}’ from endpoint 1 with gro$
debug 2020-06-17 15:20:22: No converter available for ‘HS3SA’ with cluster ‘genBasic’ and type ‘readResponse’ and data ‘{“powerSource”:3}’
debug 2020-06-17 15:20:24: Received Zigbee message from ‘0xXXXXXXXXXXX’, type ‘readResponse’, cluster ‘genBasic’, data ‘{“zclVersion”:2}’ from endpoint 1 with grou$
debug 2020-06-17 15:20:24: No converter available for ‘HS3SA’ with cluster ‘genBasic’ and type ‘readResponse’ and data ‘{“zclVersion”:2}’
debug 2020-06-17 15:20:26: Received Zigbee message from ‘0xXXXXXXXXXXX’, type ‘readResponse’, cluster ‘genBasic’, data ‘{“appVersion”:20}’ from endpoint 1 with gro$
debug 2020-06-17 15:20:26: No converter available for ‘HS3SA’ with cluster ‘genBasic’ and type ‘readResponse’ and data ‘{“appVersion”:20}’
debug 2020-06-17 15:20:29: Received Zigbee message from ‘0xXXXXXXXXXXX’, type ‘readResponse’, cluster ‘genBasic’, data ‘{“stackVersion”:2}’ from endpoint 1 with gr$
debug 2020-06-17 15:20:29: No converter available for ‘HS3SA’ with cluster ‘genBasic’ and type ‘readResponse’ and data ‘{“stackVersion”:2}’
debug 2020-06-17 15:20:49: Received Zigbee message from ‘0xXXXXXXXXXXX’, type ‘readResponse’, cluster ‘genBasic’, data ‘{“hwVersion”:16}’ from endpoint 1 with grou$
debug 2020-06-17 15:20:49: No converter available for ‘HS3SA’ with cluster ‘genBasic’ and type ‘readResponse’ and data ‘{“hwVersion”:16}’
debug 2020-06-17 15:20:51: Received Zigbee message from ‘0xXXXXXXXXXXX’, type ‘readResponse’, cluster ‘genBasic’, data ‘{“dateCode”:“2018.9.27”}’ from endpoint 1 w$
debug 2020-06-17 15:20:51: No converter available for ‘HS3SA’ with cluster ‘genBasic’ and type ‘readResponse’ and data ‘{“dateCode”:“2018.9.27”}’
debug 2020-06-17 15:20:53: Received Zigbee message from ‘0xXXXXXXXXXXX’, type ‘readResponse’, cluster ‘genBasic’, data ‘{“swBuildId”:“2000-0001”}’ from endpoint 1 debug 2020-06-17 15:20:53: No converter available for 'HS3SA' with cluster 'genBasic' and type 'readResponse' and data '{"swBuildId":"2000-0001"}' debug 2020-06-17 15:20:55: Received Zigbee message from '0xXXXXXXXXXXX', type 'readResponse', cluster 'ssIasZone', data '{"iasCieAddr":"0x0000000000000000","zoneSt
debug 2020-06-17 15:20:55: No converter available for ‘HS3SA’ with cluster ‘ssIasZone’ and type ‘readResponse’ and data '{“iasCieAddr”:“0x0000000000000000”,“zoneState”:blush:
info 2020-06-17 15:20:55: Successfully interviewed ‘0xXXXXXXXXXXX’, device has successfully been paired
info 2020-06-17 15:20:55: Device ‘0xXXXXXXXXXXX’ is supported, identified as: HEIMAN Smoke detector (HS3SA)
```

Any ideas?

If you need some additional information i would be happy to deliver it.

Thanks a lot.

P.S.  I have tested it with a fresh installation on second raspbery pi with CC2530 as a bridge:
Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
 - same behavior.

Koenkk, 2020-06-20:
Can you post the zigbee herdsman debug log when pairing the device?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

Stalkk, 2020-06-20:
Here is it
[pairing.log](https://github.com/Koenkk/zigbee2mqtt/files/4808493/pairing.log)

I think pairing starts at ca. line 342

Koenkk, 2020-06-23:
Can you check if the issue has been fixed in the latest dev branch.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Stalkk, 2020-06-23:
No difference. i have tested with this version: "version":"1.14.0-dev","commit":"c07c1d5"

Do you need any additional logs?

igorenot, 2020-06-23:
Hi! I have the same problem, I can join device, but no messages or log entries generated if test pressed or smoke detected :(

`zigbee2mqtt:info  2020-06-23 17:07:46: Device '0x00158d0004fc9481' joined
zigbee2mqtt:info  2020-06-23 17:07:46: Starting interview of '0x00158d0004fc9481'
zigbee2mqtt:info  2020-06-23 17:07:46: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x00158d0004fc9481"}}'
zigbee2mqtt:info  2020-06-23 17:07:46: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x00158d0004fc9481"}}'
zigbee2mqtt:debug 2020-06-23 17:07:46: Device '0x00158d0004fc9481' announced itself
zigbee2mqtt:info  2020-06-23 17:07:46: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x00158d0004fc9481"}}'
zigbee2mqtt:debug 2020-06-23 17:07:48: Received Zigbee message from '0x00158d0004fc9481', type 'readResponse', cluster 'genBasic', data '{"modelId":"SmokeSensor-N-3.0"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-23 17:07:48: No converter available for 'HS3SA' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"SmokeSensor-N-3.0"}'
zigbee2mqtt:info  2020-06-23 17:07:48: MQTT publish: topic 'homeassistant/binary_sensor/0x00158d0004fc9481/smoke/config', payload '{"payload_on":true,"payload_off":false,"value_template":"{{ value_json.smoke }}","device_class":"smoke","state_topic":"zigbee2mqtt/0x00158d0004fc9481","json_attributes_topic":"zigbee2mqtt/0x00158d0004fc9481","name":"0x00158d0004fc9481_smoke","unique_id":"0x00158d0004fc9481_smoke_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d0004fc9481"],"name":"0x00158d0004fc9481","sw_version":"Zigbee2mqtt 1.14.0-dev","model":"Smoke detector (HS3SA)","manufacturer":"HEIMAN"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2020-06-23 17:07:48: MQTT publish: topic 'homeassistant/binary_sensor/0x00158d0004fc9481/battery_low/config', payload '{"payload_on":true,"payload_off":false,"value_template":"{{ value_json.battery_low}}","device_class":"battery","state_topic":"zigbee2mqtt/0x00158d0004fc9481","json_attributes_topic":"zigbee2mqtt/0x00158d0004fc9481","name":"0x00158d0004fc9481_battery_low","unique_id":"0x00158d0004fc9481_battery_low_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d0004fc9481"],"name":"0x00158d0004fc9481","sw_version":"Zigbee2mqtt 1.14.0-dev","model":"Smoke detector (HS3SA)","manufacturer":"HEIMAN"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2020-06-23 17:07:48: MQTT publish: topic 'homeassistant/sensor/0x00158d0004fc9481/linkquality/config', payload '{"icon":"mdi:signal","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x00158d0004fc9481","json_attributes_topic":"zigbee2mqtt/0x00158d0004fc9481","name":"0x00158d0004fc9481_linkquality","unique_id":"0x00158d0004fc9481_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d0004fc9481"],"name":"0x00158d0004fc9481","sw_version":"Zigbee2mqtt 1.14.0-dev","model":"Smoke detector (HS3SA)","manufacturer":"HEIMAN"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:debug 2020-06-23 17:07:49: Received Zigbee message from '0x00158d0004fc9481', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"HEIMAN"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-23 17:07:49: No converter available for 'HS3SA' with cluster 'genBasic' and type 'readResponse' and data '{"manufacturerName":"HEIMAN"}'
zigbee2mqtt:debug 2020-06-23 17:07:50: Received Zigbee message from '0x00158d0004fc9481', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-23 17:07:50: No converter available for 'HS3SA' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":3}'
zigbee2mqtt:debug 2020-06-23 17:07:52: Received Zigbee message from '0x00158d0004fc9481', type 'readResponse', cluster 'genBasic', data '{"zclVersion":2}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-23 17:07:52: No converter available for 'HS3SA' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":2}'
zigbee2mqtt:debug 2020-06-23 17:07:55: Received Zigbee message from '0x00158d0004fc9481', type 'readResponse', cluster 'genBasic', data '{"appVersion":20}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-23 17:07:55: No converter available for 'HS3SA' with cluster 'genBasic' and type 'readResponse' and data '{"appVersion":20}'
zigbee2mqtt:debug 2020-06-23 17:07:57: Received Zigbee message from '0x00158d0004fc9481', type 'readResponse', cluster 'genBasic', data '{"stackVersion":2}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-23 17:07:57: No converter available for 'HS3SA' with cluster 'genBasic' and type 'readResponse' and data '{"stackVersion":2}'
zigbee2mqtt:debug 2020-06-23 17:08:01: Received Zigbee message from '0x00158d0004fc9481', type 'readResponse', cluster 'genBasic', data '{"hwVersion":16}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-23 17:08:01: No converter available for 'HS3SA' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":16}'
[... cut ...] ","vendor":"HEIMAN","description":"Smoke detector","supported":true}}'
zigbee2mqtt:debug 2020-06-23 17:08:14: Received Zigbee message from '0x00158d0004fc9481', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":186}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-06-23 17:08:14: MQTT publish: topic 'zigbee2mqtt/0x00158d0004fc9481', payload '{"battery":93,"linkquality":99}'
zigbee2mqtt:info  2020-06-23 17:08:14: Successfully configured '0x00158d0004fc9481'`

nukusinji, 2020-06-23:
> Can you check if the issue has been fixed in the latest dev branch. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

I checked latest dev and than repairing, it works Good~. Thanks

igorenot, 2020-06-23:
> I checked latest dev and than repairing, it works Good~. Thanks

Do you have Bare-metal install?

I'm using latest zigbee2mqtt-edge add-on... How can I fix it?

igorenot, 2020-06-23:
My Zigbee hardware config:
CC2531 coordinator with CC2531_SOURCE_ROUTING_20190619 firmware.
CC2531 router with CC2531_router_2019_02 firmware.

I have 2 HS1SA-E connected. First one was connected long time ago directly to Coordinator and it working as it should! New one - to router (Coordinator is full) and it doesn't work, as described above :(

Home Assistant:

arch | x86_64
chassis | vm
dev | false
docker | true
docker_version | 19.03.8
hassio | true
host_os | HassOS 4.10
installation_type | Home Assistant
os_name | Linux
os_version | 5.4.44
python_version | 3.7.7
supervisor | 227
version | 0.111.4
virtualenv | false

nukusinji, 2020-06-24:
> > I checked latest dev and than repairing, it works Good~. Thanks
> 
> Do you have Bare-metal install? I can't check latest dev if I'm using an add-on, isn't it?
> 
> I'm using latest zigbee2mqtt-edge add-on (7ad98f9c_zigbee2mqtt_edge)... How can I fix it?

I used on Docker. 
if you use on hassio, you can install edge version. but you have to uninstall z2m and reinstall for upgrade

igorenot, 2020-06-24:
> if you use on hassio, you can install edge version. but you have to uninstall z2m and reinstall for upgrade

Thank you so much! It works now!

Koenkk, 2020-06-24:
Great, assume this can be closed now.

closed 2020-06-24 by Koenkk

steve41200, 2020-07-07:
I've repairing the device in zigbee 1.14.1 (no branch dev). It does'nt work. I get test it at each update (i'm little scare to try the dev brannch)

nikopaulanne, 2020-10-03:
I just wondering if this code change can be implemented in the HA core ZHA component too? I have the same HS1SA-E smoke detectors and they are recognised in the ZHA, but the alarm is not sent to HA. Can someone help me to initiate that implementation?