Koenkk/zigbee2mqtt #3776: SPZB0001 won't pair any longer

prankousky, 2020-06-19:
Hi,

I have had this thermostat working for at least a year and a half and didn't change anything that I can point out (all other devices still work, too). This thermostat stopped working, and I tried getting it to work by resetting it, then repairing it.

It **will** show up in my `configuration.yaml`, but the thermostat itself will be at `Lin` (L mirrored, but the initiation process) for quite some time, then eventually change to `Err`. Below is the log.

I am using `{"version":"1.14.0","commit":"5d9ff02","coordinator":{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20191106}},"log_level":"info","permit_join":true}
`

`zigbee2mqtt/bridge/log {"type":"zigbee_publish_error","message":"Publish 'set' 'current_heating_setpoint' to 'Arbeitszimmer/Heizung' failed: 'Error: Write 0x00158d0001922cf7/1 hvacThermostat({\"16387\":{\"value\":1850,\"type\":41}}, {\"timeout\":10000,\"disableResponse\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":4151,\"transactionSequenceNumber\":null}) failed (Error: Timeout - 54163 - 1 - 13 - 513 - 4 after 10000ms)'","meta":{"friendly_name":"Arbeitszimmer/Heizung"}}`

I should probably update the firmware on my `Texas Instruments LAUNCHXL-CC1352P-2`, but it had been working all this time, so this cannot be the only problem. zigbee2mqtt itself is running the latest stable version of an `intel NUC`. I tried

* stopping zigbee2mqtt
* deleting device from `configuration.yaml`
* restarting zigbee2mqtt
* resetting device
* pairing device
* (will be detected and added to `configuration.yaml`, but device itself won't "realize" that it has paired)

as well as

* stopping zigbee2mqtt
* unplugging CC1352P-2
* holding reset button after plugging back in
* restarting zigbee2mqtt

I still have `bridge/log` open via mqtt_sub and just realized: another device throws an error as well (but the second time I tried turning it off, it worked): `zigbee2mqtt/bridge/log {"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'Flur/Kommode' failed: 'Error: Command 0x00158d0003fa5643/1 genLevelCtrl.moveToLevelWithOnOff({\"level\":165,\"transtime\":140}, {\"timeout\":10000,\"disableResponse\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Error: Timeout - 22408 - 1 - 27 - 8 - 11 after 10000ms)'","meta":{"friendly_name":"Flur/Kommode"}}`.

Hardware setup is the same, configuration didn't change, so why would I experience this error? Can somebody please help me figure this out?

Thank you in advance for your help :)

sjorge, 2020-06-19:
Have you checked the network map? Maybe the route changed, I have some issues with them connected to an ikea repeater but they work fine when connected to an ikea or innr bulb.

prankousky, 2020-06-21:
Thank you @sjorge 

I couldn't figure anything out by looking at the network map. According to the map, some devices are not connected at all, but they do work. I just flashed the most recent firmware to the coordinator, and suddenly the thermostat works fine again though :)

[image]

I am not sure how accurate the map is. Those device that look the furthest away work without any issues, but shouldn't they be (visually) routed to something in order to work?

closed 2020-08-01 by stale[bot]