Koenkk/zigbee2mqtt #3777: OSRAM Smart+: No converter available for '...' with cluster 'genBasic' and type '...'

Kosta-Github, 2020-06-19:
with the current version `1.14.0` I am now seeing all kind of errors like this for my `OSRAM Smart+ motion` sensor and `OSRAM Smart+ plug`:

```
zigbee2mqtt:debug 2020-06-19 12:58:45: No converter available for 'AC01353010G' with cluster 'ssIasZone' and type 'readResponse' and data '{"iasCieAddr":"0x00124b000be95bf2","zoneState":1}'
zigbee2mqtt:debug 2020-06-19 12:59:13: Received Zigbee message from 'motion_wohnzimmer_tv', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-19 12:59:13: No converter available for 'AC01353010G' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":3}'
zigbee2mqtt:debug 2020-06-19 12:59:14: Received Zigbee message from 'motion_wohnzimmer_tv', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-19 12:59:14: No converter available for 'AC01353010G' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":1}'
zigbee2mqtt:debug 2020-06-19 12:59:19: Received Zigbee message from 'motion_wohnzimmer_tv', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-19 12:59:19: No converter available for 'AC01353010G' with cluster 'genBasic' and type 'readResponse' and data '{}'
zigbee2mqtt:debug 2020-06-19 12:59:30: Received Zigbee message from 'motion_wohnzimmer_tv', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-19 12:59:30: No converter available for 'AC01353010G' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1}'
zigbee2mqtt:debug 2020-06-19 12:59:38: Received Zigbee message from 'motion_wohnzimmer_tv', type 'readResponse', cluster 'genBasic', data '{"dateCode":"����������������"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-19 12:59:38: No converter available for 'AC01353010G' with cluster 'genBasic' and type 'readResponse' and data '{"dateCode":"����������������"}'
zigbee2mqtt:debug 2020-06-19 12:59:40: Received Zigbee message from 'motion_wohnzimmer_tv', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-19 12:59:40: No converter available for 'AC01353010G' with cluster 'genBasic' and type 'readResponse' and data '{}'
zigbee2mqtt:debug 2020-06-19 12:59:42: Received Zigbee message from 'motion_wohnzimmer_tv', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 2 with groupID 0
zigbee2mqtt:debug 2020-06-19 12:59:42: No converter available for 'AC01353010G' with cluster 'genBasic' and type 'readResponse' and data '{}'
zigbee2mqtt:debug 2020-06-19 12:59:43: Received Zigbee message from 'motion_kitchen_durchgang', type 'readResponse', cluster 'genBasic', data '{"modelId":"Motion Sensor-A"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-19 12:59:43: No converter available for 'AC01353010G' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"Motion Sensor-A"}'
zigbee2mqtt:debug 2020-06-19 12:59:50: Received Zigbee message from 'motion_wohnzimmer_tv', type 'readResponse', cluster 'genBasic', data '{"stackVersion":2}' from endpoint 2 with groupID 0
zigbee2mqtt:debug 2020-06-19 12:59:50: No converter available for 'AC01353010G' with cluster 'genBasic' and type 'readResponse' and data '{"stackVersion":2}'
zigbee2mqtt:debug 2020-06-19 12:59:52: Received Zigbee message from 'motion_kitchen_durchgang', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"CentraLite"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-19 12:59:52: No converter available for 'AC01353010G' with cluster 'genBasic' and type 'readResponse' and data '{"manufacturerName":"CentraLite"}'
zigbee2mqtt:debug 2020-06-19 12:59:59: Received Zigbee message from 'motion_kitchen_durchgang', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"CentraLite"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-19 12:59:59: No converter available for 'AC01353010G' with cluster 'genBasic' and type 'readResponse' and data '{"manufacturerName":"CentraLite"}'
zigbee2mqtt:debug 2020-06-19 13:00:01: Received Zigbee message from 'motion_wohnzimmer_tv', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 2 with groupID 0
zigbee2mqtt:debug 2020-06-19 13:00:01: No converter available for 'AC01353010G' with cluster 'genBasic' and type 'readResponse' and data '{}'
zigbee2mqtt:debug 2020-06-19 13:00:03: Received Zigbee message from 'motion_wohnzimmer_tv', type 'readResponse', cluster 'ssIasZone', data '{"iasCieAddr":"0x00124b000be95bf2","zoneState":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-19 13:00:03: No converter available for 'AC01353010G' with cluster 'ssIasZone' and type 'readResponse' and data '{"iasCieAddr":"0x00124b000be95bf2","zoneState":1}'
zigbee2mqtt:debug 2020-06-19 13:00:07: Received Zigbee message from 'motion_kitchen_durchgang', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-19 13:00:07: No converter available for 'AC01353010G' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":3}'
zigbee2mqtt:debug 2020-06-19 13:00:18: Received Zigbee message from 'motion_kitchen_durchgang', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-19 13:00:18: No converter available for 'AC01353010G' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":1}'
zigbee2mqtt:debug 2020-06-19 13:00:23: Received Zigbee message from 'motion_kitchen_durchgang', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-19 13:00:23: No converter available for 'AC01353010G' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":1}'
zigbee2mqtt:debug 2020-06-19 13:00:31: Received Zigbee message from 'motion_kitchen_durchgang', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-19 13:00:31: No converter available for 'AC01353010G' with cluster 'genBasic' and type 'readResponse' and data '{}'
zigbee2mqtt:debug 2020-06-19 13:00:42: Received Zigbee message from 'motion_kitchen_durchgang', type 'readResponse', cluster 'genBasic', data '{"stackVersion":2}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-19 13:00:42: No converter available for 'AC01353010G' with cluster 'genBasic' and type 'readResponse' and data '{"stackVersion":2}'
zigbee2mqtt:debug 2020-06-19 13:00:54: Received Zigbee message from 'motion_kitchen_durchgang', type 'readResponse', cluster 'genBasic', data '{"dateCode":"����������������"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-19 13:00:54: No converter available for 'AC01353010G' with cluster 'genBasic' and type 'readResponse' and data '{"dateCode":"����������������"}'
zigbee2mqtt:debug 2020-06-19 13:00:55: Received Zigbee message from 'motion_kitchen_durchgang', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-06-19 13:00:55: No converter available for 'AC01353010G' with cluster 'genBasic' and type 'readResponse' and data '{}'
zigbee2mqtt:debug 2020-06-19 13:01:00: Received Zigbee message from 'motion_kitchen_durchgang', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 2 with groupID 0
zigbee2mqtt:debug 2020-06-19 13:01:00: No converter available for 'AC01353010G' with cluster 'genBasic' and type 'readResponse' and data '{}'
zigbee2mqtt:debug 2020-06-19 13:01:08: Received Zigbee message from 'motion_kitchen_durchgang', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 2 with groupID 0
zigbee2mqtt:debug 2020-06-19 13:01:08: No converter available for 'AC01353010G' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1}'
zigbee2mqtt:debug 2020-06-19 13:01:09: Received Zigbee message from 'motion_kitchen_durchgang', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 2 with groupID 0
zigbee2mqtt:debug 2020-06-19 13:01:09: No converter available for 'AC01353010G' with cluster 'genBasic' and type 'readResponse' and data '{}'
[... cut ...]
zigbee2mqtt:debug 2020-06-19 13:01:37: Received Zigbee message from 'plug_floor_2', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 3 with groupID 0
zigbee2mqtt:debug 2020-06-19 13:01:37: No converter available for 'AB3257001NJ' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":1}'
```

config:
```json
{"version":"1.14.0","commit":"9009de2","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":true}
```

Koenkk, 2020-06-20:
These no converter messages are not a problem (and therefore also logged as debug). It just indicated that this message will no be converter to MQTT, which also doesn't make sense for these messages.

Regarding the missing states, try to reconfigure the device. For the motion sensor make sure to wake it up before executing the command; https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure

Kosta-Github, 2020-07-04:
Is there still a way/mechanism to get rid of the `no converter available` messages? by silencing/ignoring it somehow? this would reduce the clutter/noise in the logs a bit.

Koenkk, 2020-07-04:
Yes, done that in the latest dev branch.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-07-05 by Kosta-Github

Kosta-Github, 2020-07-07:
can you do the same for the [Innr Smart Plug 220](https://www.zigbee2mqtt.io/devices/SP_220.html) as well, please?

```
No converter available for 'SP 220' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":3}'
```

Koenkk, 2020-07-07:
@Kosta-Github done.