Koenkk/zigbee2mqtt #3778: how to force reconfiguring an existing device?

Kosta-Github, 2020-06-19:
# Bug Report

Several of my devices stopped working and I re-paired my whole network (4x plugs (all the same), 17x motion sensors (all the same)), but still some of the devices are still not working properly again.

## What happened

I tried to reconfigure them via `zigbee2mqtt/bridge/configure`, but I only get back:

```
Received MQTT message on 'z2m/bridge/configure' with data 'motion_kitchen_window'
Skipping configure of 'motion_kitchen_window', device does not require this.
```

And from time-to-time I am seeing this popping up in the logs:

```
Received Zigbee message from 'motion_kitchen_window', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"manufacturerCode":4174,"imageType":48,"fileVersion":503337744}' from endpoint 1 with groupID 0
Received message from unsupported device with Zigbee model 'undefined'
Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
```

## What did you expect to happen

That the device gets reconfigured.

Is there a way to `force` trigger a reconfiguration action for some specific devices somehow?

## Debug Info

Zigbee2mqtt version:

```
{"version":"1.13.1","commit":"2cb234c","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":true}
```

Koenkk, 2020-06-20:
If `Skipping configure of 'motion_kitchen_window', device does not require this` is logged it means that the device has no `configure` method it in (https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices.js). Not all device need extra configuration after pairing it.

Kosta-Github, 2020-07-04:
both device types I am using are having configure sections

after a painful process of repairing them all >20 devices requiring several attempts it seems to be working again.

closing for now...

closed 2020-07-04 by Kosta-Github