Koenkk/zigbee2mqtt #3779: unhandled exception ZNP exception

Kosta-Github, 2020-06-19:
# Bug Report

## What happened

I have seen this unhandled exception several times in my logs now:

```
zigbee2mqtt:info  2020-06-19 14:59:36: MQTT publish: topic 'z2m/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"motion_wohnzimmer_coach"}}'
 zigbee2mqtt:debug 2020-06-19 14:59:52: Saving state to file /app/data/state.json
 (node:16) UnhandledPromiseRejectionWarning: Error: SREQ '--> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":189,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,66,2,11,254,0]}}' failed with status '2' (expected '0')
     at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:249:27)
     at Generator.next (<anonymous>)
     at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:5:58)
 (node:16) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
 (node:16) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

## Debug Info

Zigbee2mqtt version:

```
{"version":"1.13.1","commit":"2cb234c","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":true}
```

Koenkk, 2020-06-20:
Can you post the zigbee herdsman debug logging when this happens?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

Kosta-Github, 2020-07-04:
I haven't observed this within the past 14 days. closing for now

closed 2020-07-04 by Kosta-Github