Koenkk/zigbee2mqtt #3781: Ikea Tradfri LED1650R5 issues

eyeson, 2020-06-19:
I've had Zigbee2MQTT running for a while, currently on version 1.14.0. 
I run Zigbee2MQTT in docker on a Ubuntu VM on a FreeNAS host, and pass via USB-over-IP a Texas Instruments LAUNCHXL-CC26X2R1 which is plugged into a raspberry pi. 
Been almost flawless for a long time
Today I suffered multiple power cuts, in short succession and then a power cut that took a few hours to recover from.
I use Home Assistant as my 'smart' control, after the power came back, toggling any of the ikea lights in home assistant had no effect on the state of the lights (I have maybe 12 separate LED1650R5 bulbs and another 8 LED1837R5). In the Zigbee2MQTT logs I can see it publishing the "on" and "off" states. I ran MQTT Explorer on my laptop and I can also see the states being passed there.  So messages are correctly going to my MQTT broker (mosquitto) and being passed to Zigbee2MQTT. 
I then wondered if the power cut had effected the CC26XR1, however I have a number of Aqara sensors, and they're all correctly reporting and updating temperature/pressure humidity etc, so the Zigbee network is still correctly functioning. 
What I did notice, that is new, is that all of the lights have a 'color_temp' value (of 357) attached to them now, and is in each message. These bulbs don't do that, as they're on off and dimmable only. 

I am not sure where this is coming from? It certainly didn't use to happen, and I am wondering if this is what is causing them to not response as expected.
Example of the log for a toggle on:
`2020-06-19 14:48:06: MQTT publish: topic 'zigbee2mqtt/kitchen Lights 8', payload '{"state":"ON","brightness":254,"update_available":false,"color_temp":357}'`

And a toggle off:

`2020-06-19 14:50:38: MQTT publish: topic 'zigbee2mqtt/kitchen Lights 8', payload '{"state":"OFF","brightness":0,"update_available":false,"color_temp":357}'`

Koenkk, 2020-06-20:
Can you activate the debug logging and post the log when controlling the bulbs fails?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

eyeson, 2020-06-24:
> Can you activate the debug logging and post the log when controlling the bulbs fails?
> 
> To enable debug logging set in `configuration.yaml`:
> 
> ```yaml
> advanced:
>   log_level: debug
> ```

Sorry Koenkk, unfortunately with the lights completely not responding after trying to troubleshoot why something had changed and borked the network. I decided to change the PAN ID, take the opportunity to update the firmware on the adaptor, and re pair all devices. Unsurprisingly that fixed the issue, but got rid of any logs (I deleted everything except the config).

Its all working again, and I am no longer getting "color_temp" 357 added to all my lights... having not changed anything other than the Zigbee2MQTT network, something must have got upset there but I can now not help identify what.
I will close this down now, as I guess with out any more information it becomes a useless report. Cheers

closed 2020-06-24 by eyeson

Nauttor, 2020-11-19:
Hi @Koenkk , 

I think I'm experimenting something similar to what eyeson see when using LED1650R5.

I have the following scenario:

RemotePlus:
E1743
Group "Sofa" Bulbs LED1650R5:
Sofa1,Sofa2, Sofa3
"Sofa" group is bind to the RemotePlus

RemoteMini:
E1743
Bulbs LED1537R6/LED1739R5:
Table1,Table2, Table3
"Table" group is bind to the RemoteMini

The issue:
If I switch on/off the remoteMini, my "Table" group always report the status in Mqtt as expected. 
`Client mosq-ximszruCENPXUUPgmR received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/bridge/logging', ... (88 bytes))
{"level":"info","message":"MQTT publish: topic 'zigbee2mqtt/table/state', payload 'ON'"}
Client mosq-ximszruCENPXUUPgmR received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/table/state', ... (2 bytes))
ON`

If I switch on/off the remotePlus, my "sofa" group never report the status of the group just the status of the remoteMini. 

At first I though maybe the remotes are involve in the error, but I unbind the "plus" and the "mini" from their groups and the bind them in crossover to the other group and the bulbs behave equally, so the problem is not within the remotes, is more on the LED1650R5 side. Every time I bind the remotes to other groups of LED1537R6/LED1739R5 it works.

Nauttor, 2020-11-19:
I tried to recreate the group and the re-bind the remote and it works only for the first time(bulbs + remote sent status), the second time only the remote report the status again.  

BTW my system is running:
Product: TI CC2531 USB CDC
Zigbee2mqtt 1.16.1