Koenkk/zigbee2mqtt #3782: Hue bulb turning on with 0% brightness after 1.14

kcarriello, 2020-06-19:
## What happened

After upgrade to 1.14 my hue bulbs started to have brightness 0% when turned on. This only happens when service light turn on is used on home assistant, when using hue dimmer the lights bright as expected. 

## What did you expect to happen

The previous behavior (until 1.13.1), when I turned off a light and turned it on again, using Home Assistant service, the light returned to the previous state or full brightness, now it always turned on with 0% brightness, which makes me every time open brightness configuration to increase the value.

## How to reproduce it (minimal and precise)

- Add a hue bulb to Home Assistant;
- Use light turn off service to turn off the bulb;
- Use light turn on service to turn on the bulb;

## Debug Info
Version of Zigbee2Mqtt: 1.14
Coordinator version: 20190619
CC2531
Firmware version is zStack12.

Koenkk, 2020-06-20:
@mr2c12 and @kcarriello this may already be fixed in #3736 .

Can you check if the latest dev branch fixed this issue? (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

kcarriello, 2020-06-21:
I installed the dev version, it fixed the issue.

Thank you!

closed 2020-06-21 by Koenkk

mr2c12, 2020-06-22:
> @mr2c12 and @kcarriello this may already be fixed in #3736 .
> 
> Can you check if the latest dev branch fixed this issue? (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

@Koenkk in my case it has been fixed, too.

kcarriello, 2020-06-22:
There's an odd behavior on this. It seems that if I take a long time (don't know precisely how long) without turning on the lamp it shows the 0% brightness behavior, turning off and on immediately doesn't trigger the 0% brightness but, from one day to another, it shows up again. Today I tried to turn on 2 group of lights, all of them turned on with 0% brightness, i'm running the dev version.

mr2c12, 2020-06-24:
> @mr2c12 and @kcarriello this may already be fixed in #3736 .
> 
> Can you check if the latest dev branch fixed this issue? (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

@Koenkk , yesterday the problem occurred again in my setup. It happens in a group of Philips HUE bulbs, while it used to happen in a group of Osram bulbs. Dev branch.