Koenkk/zigbee2mqtt #3787: Zigbee2MQTT and Raspberry Pi 4 2GB

scargill, 2020-06-20:
I am on the latest Zigbee2 MQTT version - and on my Raspberry Pi 3 talking to Mosquitto MQTT and the Sonoff pre-programmed dongle, it worked perfectly. A week ago we put Home Assistant on a Raspberry Pi 4 2GB and tried Z2M on that - no joy. It turns out that was a red herring so today I took an SSD with Mosquitto and Node-Red on it - all up to date - added Z2M... on the Raspberry Pi 3 it works perfectly. Taking the exact same dongle and SSD with Z2 M onto my RPi4 - identical setup  no joy, it cannot communicate with any Zigbee devices. Mosquitto of course comes up but it looks like the code is not talking to the dongle. Taking the whole lot back to the Raspberry Pi 3 - perfect. There are no power supply issues on the RPi4.

Any ideas

Pete

# Bug Report
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version:
Adapter hardware: CC2531, CC2530, CC26X2R1 or CC1352P-2
Adapter firmware version:

Koenkk, 2020-06-20:
Is the adapter connected through a usb extension cable? The pi 4 is known to generate a lot of interference.

scargill, 2020-06-20:
Hi there - I'll  put on a USB extension but the test sockets are just a few feet away in this case and I'm not getting poor performance on the 4, I'm getting NO performance :-).

notenoughtech, 2020-07-14:
@Koenkk I think I have a lead on this. 
Once I connected SSD nothing would work properly.  I would see the CC2531 ok, but pairing would fail (no messages at all) and no device would post properly. 
At first I thought it was interference, but when I connected the mSATA SSD things were ok. Also when I connected SSD with a powered hub - I was able to pair devices again.  There is a small voltage drop when the SSD is connected - I don't know how relevant is this, perhaps it triggers instability? I was running CC2531 for ages on RPI4 without issues until I started to play with SSD

Koenkk, 2020-07-14:
@notenoughtech makes sense, I've added this to the [FAQ](perhaps it triggers instability), thanks!

notenoughtech, 2020-07-14:
happy to help. 
Took me 2 nights to pin down why I have such issues especially that all this happened about the time I flashed the latest firmware and played with new sonoff devices. Thanks for great work btw

Koenkk, 2020-07-14:
@notenoughtech :smile: also thanks to you for the videos!

closed 2020-08-21 by github-actions[bot]