Koenkk/zigbee2mqtt #3788: CC1352 -  some stupid questions...

olili, 2020-06-21:
since I had many paring problems in the past with my CC2530 due to memory restrictions I decided to use a more powerful device for the coordinator. Finally I decided to usage of this CC1352 [variant](https://de.aliexpress.com/item/4000321518715.html?spm=a2g0o.productlist.0.0.73513cf7vjDvY8&algo_pvid=7b21826d-4679-4065-82de-b920d0271421&algo_expid=7b21826d-4679-4065-82de-b920d0271421-10&btsid=0ab6f83915927307050055263e7b1d&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_)
 
Now I have the following Beginner questions:  

1)  these devices have a  native bootloader support, right? I can access the bootloader via a ordinary 2-wire UART and can flash with Ti's Uniflash. Correct? 
On which pins I can find the TxD and RxD on CC1352 side in  this case?

 2) communication with Z2M will be done by UART as well. Which pins for RxD and TxD of CC1352 I  have to use of Koenkk'S coordinator firwmare? 

Many thx in advance. 

A CC1352 greenhorn.

enesbcs, 2020-06-21:
The item you've linked is a CC1352P-1 with max 5dBm not a P-2.
https://github.com/Koenkk/zigbee2mqtt/issues/1429#issuecomment-631323561

CC2538 might be better choice...
https://github.com/Koenkk/zigbee2mqtt/issues/1568

co-Palko, 2020-06-22:
> since I had many paring problems in the past with my CC2530 due to memory restrictions I decided to use a more powerful device for the coordinator. Finally I decided to usage of this CC1352 [...]

Native bootlooader is supported, on standard UART pins or other, if you are remapping it.

"On which pins I can find the TxD and RxD on CC1352 side in this case?" - read TI manual for LAUNCHXL-CC1352P-1 or LAUNCHXL-CC1352P-2.

Koenkk FW uses pins as shown for LAUNCHXL-CC1352P board.

1igeza, 2020-06-26:
Can it be used as coordinator? I don't care about range because I have routers in my house. Doas someone use this module?

co-Palko, 2020-06-26:
> Can it be used as coordinator? I don't care about range because I have routers in my house. Doas someone use this module?

Yes, I use e-byte module with 1352p like coordinator with z2m.

kandal99, 2020-06-26:
I have build my coordinator with z2m and the e-byte module (only for testing).

The default UART pin is DIO13 (TX) and DIO12 (RX). I have use a cheap USB/UART board with CP2102 and connected RX,TX,3.3 and GND (see pic).
A push button is connected to DIO15 to boot into serial loader (is required after a valid firmware is flashed).

I have flash Koenkks firmware (CC1352P_2_20200417.hex) what supports the boot loader (use hex editor):

address 57FD8 , value C5 , bootloader backdoor enabled
address 57FD9 , value 0F , DIO15 is used
address 57FDA , value FE , active low
address 57FDB , value C5 , bootloader enabled

How to flash the hex file see https://github.com/JelmerT/cc2538-bsl

[image]

Is not pretty but it works fine.

1igeza, 2020-06-26:
Great tutorial, Is TMSC, TCKC, TDO, TDI and RESET connected to anything or is it just only soldiered to pins on the board? When I receive CC1352P I will try to use it wireless connected to esp8266.

kandal99, 2020-06-26:
TMSC, TCKC, TDO, TDI, RESET is only connected to the pins on the board for debugger (e.g. j-link). 

If you disable the Bootloader, then you can only flash with debugger.

1igeza, 2020-06-26:
I bought this one from the first comment. As I understand  it is no programmed, so bootloader should be unlocked or should I also buy jtag for first programming?

kandal99, 2020-06-26:
When no firmware (or no valid firmware) was flashed, the Chip boot automatically in Bootloader Mode and you can flash your firmware without j-link. 
You can check any firmware file with a hex editor for the values at the addresses befor you flash it.

Look at https://electrolama.com/projects/zig-a-zig-ah/ (its a cool project with the CC2652 chip with serial loader feature and debug interface)

olili, 2020-06-29:
@kandal99, many thx. This is exactly what I was looking for!!!

1igeza, 2020-07-15:
@kandal99
> I have flash Koenkks firmware (CC1352P_2_20200417.hex) what supports the boot loader (use hex editor):
> 
> address 57FD8 , value C5 , bootloader backdoor enabled
> address 57FD9 , value 0F , DIO15 is used
> address 57FDA , value FE , active low
> address 57FDB , value C5 , bootloader enabled

Could you share modified bootloader?   I don't want to flash bad file and I'm not sure if I modified this file correctly because modified decoded text looks different then the rest. 
[image]

kandal99, 2020-07-15:
@ 1igeza

You dont need to modify Koenkks firmware file ( CC1352P_2_20200417.hex ). It has the right bootloader options. 

You can use www.hexed.it to check the hex file for the values.

1igeza, 2020-07-15:
@kandal99

No luck with programing. No communication with device "ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'" 
[image]
Could it be broken? No power usage after connecting to PC by power meter.

kandal99, 2020-07-16:
@1igeza

It seems that the chip dont boot into the bootloader (perhaps it is a valid firmware on it). I hope the default values for backdoor pin are valid. 

1. So you can connect DIO_15 to GND and power it up and release. Then test to flash a firmware file.
or
2. Connect DIO_15 to VCC and power it up and release. Then test again to flash a firmware file.

When this dont help you can test it with a J-Link (EDU Version or a Clone).

1igeza, 2020-08-03:
@kandal99
Thank you for your help, I have J-link v8 and I'm not sure about wiring. 
(1)Vtref - VCC
(4)GND - GND
(5)TDI -  TDI
(7)TMS - TMSC
(9)TCK - TCKC
(13)TDO - TDO
(15)RESET - RESET
Is that correct? Do I need anything else? 
If I want connect external antenna to pins ANT and GND, do I need to change something or is it already connected to board.

kandal99, 2020-08-04:
Sorry for my late reply and congratulation !

The onboard external antenna is for Sub-1GHz and the printed antenna is for 2.4 GHz (zigbee). The CP1352P is a dual-band modul.

1igeza, 2020-08-08:
I have second unit and today I checked memory content.  It look's that bootloader is locked. I thought that there is something wrong with my old laptop, but after flashing firmware with unlocked bootloader (by J-link), I can connect to it by UART. 
It's from orginal firmware:
[image]

th0m4sek, 2020-08-24:
> @kandal99
> Thank you for your help, I have J-link v8 and I'm not sure about wiring.
> (1)Vtref - VCC
> (4)GND - GND
> (5)TDI - TDI
> (7)TMS - TMSC
> (9)TCK - TCKC
> (13)TDO - TDO
> (15)RESET - RESET
> Is that correct? Do I need anything else?
> If I want connect external antenna to pins ANT and GND, do I need to change something or is it already connected to board.
> 
> edit.
> I flashed it successfully using UniFlash. Zigbee2mqtt is working OK.
> Coordinator firmware version: '{"type":"zStack3x0","meta":{"transportrev":2,"product":1,"majorrel":2,"minorrel":7,"maintrel":1,"revision":20200721}}'
> 
> Once again thank you for your help.

How did you manage to force bootloader and flash with UniFlash ?

1igeza, 2020-08-24:
I flash it using UniFlash and J-link. In my unit the bootloader was locked. 
Pin connected to J-link:
(1)Vtref - VCC
(4)GND - GND
(5)TDI - TDI
(7)TMS - TMSC
(9)TCK - TCKC
(13)TDO - TDO
(15)RESET - RESET

th0m4sek, 2020-08-25:
I founded https://gist.github.com/hwhw/fc43892785aa84913d03495c97b0f25a but i constantly loose communication and i can't find any flash registers in cc1352 manual.
So i think i will need to buy a jlink...

closed 2020-10-03 by github-actions[bot]