Koenkk/zigbee2mqtt #3790: Smarthing Arrival Sensor PGC410 support?

Sk3wby, 2020-06-21:
Hi,

Apologies, I am new to zigbee2mqtt and am just transferring all my automation stuff across to HA with this addon.  I have successfully paired my Aqara & Smarthings sensors & switches but i am having trouble with my Smartthings Arrival sensor.

I have tried to follow the guide for adding new devices but get stuck at SSHing into HA, think i'm doing something wrong with the authorised_keys file so cannot access devices.js.

I get these log entries for the device:

{"ieeeAddr":"0xd052a800xxxxxxxx","type":"EndDevice","networkAddress":29735,"model":"PGC410","vendor":"-","description":"-","friendly_name":"Keys","manufacturerID":4362,"manufacturerName":"SmartThings","powerSource":"Unknown","modelID":"PGC410","hardwareVersion":3,"dateCode":"130324","lastSeen":1592748934795}]}'

It seems almost identical in function to this model:

https://github.com/Koenkk/zigbee2mqtt/issues/459

same battery & beep function just a different model number, PGC410.

Would it be possible to add support for this device?

Any help would be appreciated.

thanks,
ryan

Koenkk, 2020-06-24:
Although it may  be similar, it is no guarantee that it works so this first has to be tested.

Try following https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html and add the following to `devices.js`.

```js
{
    zigbeeModel: ['PGC410'],
    model: 'PGC410',
    vendor: 'SmartThings',
    description: 'Arrival sensor',
    supports: 'presence',
    fromZigbee: [
        fz.STS_PRS_251_presence, fz.battery_3V,
        fz.STS_PRS_251_beeping,
    ],
    toZigbee: [tz.STS_PRS_251_beep],
    meta: {configureKey: 2},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'genBinaryInput']);
        await configureReporting.batteryVoltage(endpoint);
        await configureReporting.presentValue(endpoint);
    },
},
```

Sk3wby, 2020-06-26:
Thankyou for the reply.  I have added into devices.js and now i get:

> zigbee2mqtt:debug 2020-06-26 10:55:13: No converter available for 'PGC410' with cluster 'genBasic' and type 'readResponse' and data '{}'
zigbee2mqtt:error 2020-06-26 10:55:23: Failed to interview '0xd052a800e81bxxxx', device has not successfully been paired
zigbee2mqtt:info  2020-06-26 10:55:23: Configuring '0xd052a800e81bxxxx'
zigbee2mqtt:info  2020-06-26 10:55:23: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0xd052a800e81bxxxx"}}'
zigbee2mqtt:error 2020-06-26 10:55:34: Failed to configure '0xd052a800e81bxxxx', attempt 3 (Error: Bind 0xd052a800e81bxxxx/1 genPowerCfg from '0x00124b001938a13e/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
    at Endpoint.<anonymous> (/zigbee2mqtt-1.14.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:255:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.14.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))

I have reopened devices.js with vi after this log and new entry is still there so i'm hoping i've done this right?

thanks,
ryan

Koenkk, 2020-06-27:
Looks good, only the device fails to pair due to some timeouts. Can you make sure to keep the device awake by pressing buttons on it during pairing?

closed 2020-08-03 by stale[bot]

liamstears, 2020-08-06:
Hi, I have the same device and running into the same issue and hope you can resolve it?

I'm in the UK and I believe this model is the only SmartThings arrival sensor available in the UK so I'm sure I won't be the only person grateful if we get this working.

During testing I found that my device reports itself as "PGC410EU" instead of just "PGC410" so I made the necessary adjustments in devices.js

I read the previous and you mention keeping it awake, I have done this but also ran debug level logging and here are the results:

```zigbee2mqtt:info  2020-08-06 22:31:57: Device '0xd052a81e8ca00005' joined
zigbee2mqtt:info  2020-08-06 22:31:57: Starting interview of '0xd052a81e8ca00005'
zigbee2mqtt:info  2020-08-06 22:31:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0xd052a81e8ca00005"}}'
zigbee2mqtt:info  2020-08-06 22:31:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0xd052a81e8ca00005"}}'
zigbee2mqtt:debug 2020-08-06 22:31:58: Device '0xd052a81e8ca00005' announced itself
zigbee2mqtt:info  2020-08-06 22:31:58: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0xd052a81e8ca00005"}}'
zigbee2mqtt:debug 2020-08-06 22:32:18: Received Zigbee message from '0xd052a81e8ca00005', type 'readResponse', cluster 'genBasic', data '{"modelId":"PGC410EU"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-06 22:32:18: No converter available for 'PGC410EU' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"PGC410EU"}'
zigbee2mqtt:debug 2020-08-06 22:32:28: Received Zigbee message from '0xd052a81e8ca00005', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"SmartThings"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-06 22:32:28: No converter available for 'PGC410EU' with cluster 'genBasic' and type 'readResponse' and data '{"manufacturerName":"SmartThings"}'
zigbee2mqtt:debug 2020-08-06 22:32:38: Received Zigbee message from '0xd052a81e8ca00005', type 'readResponse', cluster 'genBasic', data '{"powerSource":0}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-06 22:32:38: No converter available for 'PGC410EU' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":0}'
zigbee2mqtt:debug 2020-08-06 22:32:48: Received Zigbee message from '0xd052a81e8ca00005', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-06 22:32:48: No converter available for 'PGC410EU' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":1}'
zigbee2mqtt:debug 2020-08-06 22:32:58: Received Zigbee message from '0xd052a81e8ca00005', type 'readResponse', cluster 'genBasic', data '{"appVersion":4}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-06 22:32:58: No converter available for 'PGC410EU' with cluster 'genBasic' and type 'readResponse' and data '{"appVersion":4}'
zigbee2mqtt:debug 2020-08-06 22:33:08: Received Zigbee message from '0xd052a81e8ca00005', type 'readResponse', cluster 'genBasic', data '{"stackVersion":71}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-06 22:33:08: No converter available for 'PGC410EU' with cluster 'genBasic' and type 'readResponse' and data '{"stackVersion":71}'
zigbee2mqtt:debug 2020-08-06 22:33:18: Received Zigbee message from '0xd052a81e8ca00005', type 'readResponse', cluster 'genBasic', data '{"hwVersion":3}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-06 22:33:18: No converter available for 'PGC410EU' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":3}'
zigbee2mqtt:debug 2020-08-06 22:33:28: Received Zigbee message from '0xd052a81e8ca00005', type 'readResponse', cluster 'genBasic', data '{"dateCode":"130925"}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-06 22:33:28: No converter available for 'PGC410EU' with cluster 'genBasic' and type 'readResponse' and data '{"dateCode":"130925"}'
zigbee2mqtt:debug 2020-08-06 22:33:38: Received Zigbee message from '0xd052a81e8ca00005', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-08-06 22:33:38: No converter available for 'PGC410EU' with cluster 'genBasic' and type 'readResponse' and data '{}'
zigbee2mqtt:error 2020-08-06 22:33:48: Failed to interview '0xd052a81e8ca00005', device has not successfully been paired
zigbee2mqtt:info  2020-08-06 22:33:48: Configuring '0xd052a81e8ca00005'
zigbee2mqtt:info  2020-08-06 22:33:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_failed","meta":{"friendly_name":"0xd052a81e8ca00005"}}'```

Hopefull you can help get this resolved?

Sk3wby, 2020-08-09:
Sorry i didn't follow up sooner but i too tried again by repeatedly pushing the only button on the device during pairing. i tried over 20 different pairing attempts pushing the button once every 10 seconds right down to pushing it several times a second to keep it awake.  It gets picked up on first press after reset but always times out with 'not successfully paired'.  Shame, as it's the only thing i miss from my old SmartThings and Hubitat setups.  The battery life on this sensor was not great but it was a reliable presence sensor.  I'm happy to post the sensor PCB and pay for return if it helps to get it working?

liamstears, 2020-08-10:
> Sorry i didn't follow up sooner but i too tried again by repeatedly pushing the only button on the device during pairing. i tried over 20 different pairing attempts pushing the button once every 10 [...]

No problem, follow the new thread though: https://github.com/Koenkk/zigbee2mqtt/issues/4055

Hopefully me and koenkk can get this working