Koenkk/zigbee2mqtt #3791: Zigbee2mqtt stop pairing new devices after Failing to configure a device

dinhchinh82, 2020-06-21:
# Bug Report
## What happened
My Zigbee2mqtt adapter stop to pair new devices. There is no any "announcement message" or "Starting interview" message in the log so it is unable to pair new devices.

This issue only occurs after it failed to pair a device.
The error message in the log:

> zigbee2mqtt:error 2020-06-22 00:35:53: Failed to configure '0xec1bbdfffe91e950', attempt 1 (Error: Bind 0xec1bbdfffe91e950/1 genBasic from '0x00124b0008c98648/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
    at Endpoint.<anonymous> (/zigbee2mqtt-1.14.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:255:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.14.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))

## What did you expect to happen
Be able to pair new devices even after failing to pair a new device.

## How to reproduce it (minimal and precise)
1. Pair some routers & end devices. Mostly they are zigbee switch & sensors
2. Failed to configure a device (even that device is supported)
3. Pair other device => no any "start to interview..." message and unable to pair any new devices even they are supported.

If reset the zigbee2mqtt (change key or channel) then start to pair the failed device above first, it can be paired successfully but the error occurred again after pairing around 15-16 devices. 

## Debug Info
Zigbee2mqtt version: 1.14.0
Adapter hardware: CC2538
Adapter firmware version: zStack30x rev.20200327
Total connected device: 16 routers + 16 end devices

> zigbee2mqtt:info  2020-06-22 00:35:41: Starting zigbee2mqtt version 1.14.0 (commit #unknown)
zigbee2mqtt:info  2020-06-22 00:35:41: Starting zigbee-herdsman...
zigbee2mqtt:info  2020-06-22 00:35:43: zigbee-herdsman started
zigbee2mqtt:info  2020-06-22 00:35:43: Coordinator firmware version: '{"type":"zStack30x","meta":{"transportrev":2,"product":2,"majorrel":2,"minorrel":7,"maintrel":2,"revision":20200327}}'
zigbee2mqtt:info  2020-06-22 00:35:43: Currently 33 devices are joined:

clockwork6666, 2020-07-23:
had the same problem after failed unbind.

replugging the stick got it into pairing again.

thanks for the hint :+1:

closed 2020-08-29 by dinhchinh82

dinhchinh82, 2020-08-29:
it seems the issue comes from the network devices, not the z2m itself