Koenkk/zigbee2mqtt #3792: Adding Unsupported Device

Pashovski, 2020-06-22:
Z2M is amazing! I've got almost all of my Hue stuff going and am this close to throwing out my old Hue bridge!
I have my unsupported device (hue bulb), and with the docs I have been able to get all the way to step 5 (Hass.io addon) inside of the inside of the zigbee2mqtt container, and just want to double check to make sure I am doing the right thing. This info that I got from the log, when I did my Touchlink factory reset (another amazing feature) :

zigbee2mqtt:info  2020-06-20 20:01:24: Successfully interviewed '0x0017880106c01547', device has successfully been paired
zigbee2mqtt:warn  2020-06-20 20:01:24: Device '0x0017880106c01547' with Zigbee model 'LWE001' is NOT supported

Am I correct that I need to open vi /app/node_modules/zigbee-herdsman-converters/devices.js , and that the entry for my device would look like this?
{
zigbeeModel: ['LWE001'],
model: 'LWE001-1-E12DLv1',
vendor: 'Philips',
description: 'Dimmable light',
supports: 'Dimming',
extend: hue.light_onoff_brightnes
//fromZigbee: [], // We will add this later
//toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
},

Home Assistant on a Raspberry Pi 4. Again. Just amazing. Can't believe I am able to my stuff out of the cloud =)

Koenkk, 2020-06-24:
Yes that should work. Can you also provide a link to the product page?

Let me know if it works then I can add it.

Pashovski, 2020-06-25:
Getting stuck on the final step. Would you mind taking a look at my short vid - I took a page out of your book and kept it under 45 seconds https://youtu.be/n7r366XqrNE

Pashovski, 2020-06-26:
Holy **** it worked. ``zigbee2mqtt/0x0017880106c01547/set`` payload ``off`` worked. I am seeing it in the Z2M Logs. Amazing.

Is there another step to make it one of my entities? I do not see it in developer tools: states.

Here is the product page https://www2.meethue.com/en-us/p/hue-white-2-pack-e12/046677548285

Koenkk, 2020-06-27:
I've added it, will be supported in zigbee2mqtt 1.14.1 which will be released in a few days.

closed 2020-06-27 by Koenkk

Pashovski, 2020-06-27:
Thank You! Hey one point of clarification: I noticed you wrote the model as the EAN/UPC - product number (046677548285), not the Material number (12NC) (929002039801) . Is there a reason for that? When I look on the device for a model number I see the Material number. That Is what I put in my files. Should I change it to the UPC?
https://imgur.com/WU9aN4a
https://imgur.com/3LG51Ov