Koenkk/zigbee2mqtt #3793: CC2531 insufficient? (will CC26X2R1 solve this?)

nielsi, 2020-06-22:
# Bug Report
## What happened

**Issue in short:**
CC2531 adapter results in random and unwanted behavior. Before I invest in a CC26X2R1 adapter, I wondered if I can expect these issues to be resolved with this adapter. A big thanks to everyone spending time on this project. I appreciate that it might not offer the solution for every use case. I'm now trying to determine if it's the right option for me.

**Background in full:**

I come from a situation where I had a Python script sending COAP commands to the Ikea Tradfri bridge. This worked as expected. Sending 22 individual COAP commands to the Ikea bridge resulted in the 22 lights in the living room (mostly GU10 spots) changing to their expected brightness (conceivably) simultaneously. 
I switched to zigbee2mqtt and flashed a CC2531 because I like the open source nature of the project, the potential of easily mixing different hardware from different suppliers and the elegant solution of using MQTT as a bridge. 
Probably unsurprisingly to those with experience with the CC2531, the new results were not similar. Only a part of that could be remedied by more efficient use of groups. It started with lamps responding to their MQTT->Zigbee commands but with a noticeable delay (one after the other popping on). Without change in hardware of software after about two days the behavior became more unpredictable like lights not at all responding to command or after a lag of sometimes minutes. To the current point where (again without a change in hardware of software) /dev/ttyACM0 disapears after zigbee2mqtt is started and zigbee-herdmans exits with a "AREQ - ZDO - stateChangeInd after 60000ms". Suggestions like using the reset button on the CC2531, reinstalling software, etc. result in no different behavior to the point that I'm suspecting the CC2531 is broken.

Because I have since learned that controlling 30+ devices with the CC2531 will result in unexpected behavior and even in the best case scenario I had lights turning on with a noticeable delay, I am now considering buying a CC26X2R1. However, I am somewhat skeptical this might make a difference. My thinking is that, because the Zigbee network was working well with the Tradfri bridge, range should't have to be an issue. All light sources are 220v connected hubs and the adapter is within 2 meters of the first (multiple) lights. Unless the issue could be adapter-memory related and a CC26X2R1 would fix that.

## Debug Info
software and firmware all latest versions from last week (june 15th 2020)

elbueno222, 2020-06-22:
What firmware did you flash to the CC2531? The recommended firmware for more than 30 devices is the source routing. I have 20 devices but I changed the firmware to source routing for a different reason (I was having issues with couple of devices) and from the moment I updated it I haven´t had any other problem regarding the coordinator. Before buying a new coordinator I would give it a try, you can find the firmware here:

https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_Home_1.2/bin/source_routing/CC2531_SOURCE_ROUTING_20190619.zip

nielsi, 2020-06-23:
Thanks for thinking along. Yes, that firmware gives the relative best result described above. Connecting the adapter via a USB extension cable or moving the entire device into different locations makes no difference. I decided to buy the CC26X2R1 anyway. The potential of a working zigbee2mqtt network is too interesting. I currently have about 50 devices to connect and that will probably increase if I also start converting the other floors. If this issue is still open by the time the CC26X2R1 arrives, I'll report back if it made the difference.

kovaga, 2020-06-23:
It did make a difference for me, I had cc2530+cc2591 coordinator and the network started misbehaving when I was on 28 devices. Tried source routing firmware, but cant say it helped much. Having switched to CC26X2R1 it made such a significant difference in overall experience, from the time it took for new devices to join to the faster operation. I would definetely recommend the upgrade.

IVI053, 2020-06-23:
I had the same expirience as @kovaga 
Totally worth the invest. Should solche your issues

mr2c12, 2020-06-24:
I suggest to invest in CC26X2R1. I used to have the CC2530+cc2591 that has a better signal strenght respect to cc2531, but the CC26X2R1 is so much more stable. No more problems in 4 months now.

nielsi, 2020-06-24:
That all sounds very promising. It should arrive next week. I will report back.

nielsi, 2020-06-30:
I'm happy to report that this does indeed make a huge difference. I'm getting much more stable results. The only (rare) quirk I find is that the order at which commands arrive at devices isn't always the same as they were sent out (I'm taking miliseconds sent after each other). That might be something I can solve in my controlling software.

Closing note: I know I bought my CC2531 because it's pushed pretty hard on www.zigbee2mqtt.io and seemed the obvious choice to make ("What do I need?" > "Flashing the CC2531", etc.). Maybe an extra line on the "What do I need?" page , that for 20+ devices something like the CC26X2R1 is recommended could help cases like mine. But that site might be in a different repo.

Koenkk, 2020-06-30:
@nielsi good point! Added this to https://www.zigbee2mqtt.io/getting_started/what_do_i_need.html.

Assuming this can be closed now.

closed 2020-06-30 by Koenkk