Koenkk/zigbee2mqtt #3794: Reporting Interval for Sonoff SNZB-02 Temperature & Humidity Sensor

nanobra1n, 2020-06-22:
# Bug Report
## What happened
I got a [Sonoff SNZB-02](https://www.itead.cc/sonoff-snzb-02-zigbee-temperature-and-humidity-sensor.html) Temperature & Humidity Sensor. It paired fine and was identified as an eWeLink RHK08. The problem I am having (unless this is by design) is that it reports new values every 10 s. I can mitigate this slightly by setting the temperature & humidity precision to 1 decimal, but even then it takes at most 20 seconds for a new value to be reported. 
## What did you expect to happen
Behavior similar to my Xiaomi sensors, which report values much less frequent. 
## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version: 
Adapter hardware: CC1352P-2
Adapter firmware version: 'type': 'zStack3x0', 'meta': {'transportrev': 2, 'product': 1, 'majorrel': 2, 'minorrel': 7, 'maintrel': 1, 'revision': 20200417}}

Koenkk, 2020-06-24:
In `devices.js` can you try changing 

```js
temperature: async (endpoint) => {
    const payload = [{
        attribute: 'measuredValue',
        minimumReportInterval: 0,
        maximumReportInterval: repInterval.HOUR,
        reportableChange: 25,
    }];
    await endpoint.configureReporting('msTemperatureMeasurement', payload);
},
```

to

```js
temperature: async (endpoint) => {
    const payload = [{
        attribute: 'measuredValue',
        minimumReportInterval: 5,
        maximumReportInterval: repInterval.HOUR,
        reportableChange: 100,
    }];
    await endpoint.configureReporting('msTemperatureMeasurement', payload);
},
```

After that restart zigbee2mqtt and reconfigure the device via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure (make sure to wake up the device before executing this).

If that doesn't help, try increasing the `reportableChange` further to e.g. `500`.

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

nanobra1n, 2020-06-26:
The reconfiguration of the device keeps failing with:
```
error 2020-06-26 17:10:58: Failed to configure 'Sonoff LR', attempt 1 (Error: Bind 0x00124b001b783e4a/1 msTemperatureMeasurement from '0x00124b001ca5cb1c/1' failed (Error: AREQ - ZDO - bindRsp after 10000ms)
    at Endpoint.<anonymous> (/zigbee2mqtt-1.14.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:255:23)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.14.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:6:65))
```
so whatever change I made apparently didn't take. I see no different behavior

Koenkk, 2020-06-28:
@nanobra1n did you wakeup the device before doing this? (by clicking a button on it)

nanobra1n, 2020-06-29:
I thought I did, but I tried again and apparently once per second just wasn't enough, I had to press faster than that to get the device configured, but not too fast. The sweet spot was about twice a second.  Unfortunately even going up to `reportableChange: 1000` didn't help much. However looking at the reported data, it appeared temperature was the wrong tree to bark up to. While the reported temperature in consecutive measurements is pretty stable, the reported humidity is fairly noisy (+- 0.8%). 

So in addition to 
```
  temperature: async (endpoint) => {
    const payload = [{
        attribute: 'measuredValue',
        minimumReportInterval: 5,
        maximumReportInterval: repInterval.HOUR,
        reportableChange: 100,
    }];
    await endpoint.configureReporting('msTemperatureMeasurement', payload);
},
```
I also changed
```
    humidity: async (endpoint, min = 10, max = repInterval.HOUR, change = 10) => {
        const payload = [{
            attribute: 'measuredValue',
            minimumReportInterval: min,
            maximumReportInterval: max,
            reportableChange: change,
        }];
        await endpoint.configureReporting('msRelativeHumidity', payload);
    },
```
to 
```
    humidity: async (endpoint, min = 10, max = repInterval.HOUR, change = 100) => {
        const payload = [{
            attribute: 'measuredValue',
            minimumReportInterval: min,
            maximumReportInterval: max,
            reportableChange: change,
        }];
        await endpoint.configureReporting('msRelativeHumidity', payload);
    },
```
and that calmed down things a lot (now ~ 1 minute between reported values)

Koenkk, 2020-06-29:
1 minute is still a lot I think, can you try increasing it further?

nanobra1n, 2020-06-29:
When I leave the sensor alone and don't breathe on it as much I get several minutes between reports now.

Koenkk, 2020-06-30:
Good, can you make a PR?

Koenkk, 2020-07-03:
Merged, assuming this can be closed.

closed 2020-07-03 by Koenkk

sergimola, 2020-12-04:
I actually need for SNZB-02 to report much more frequently than it is doing it now. 
Is there a way to configure this so it reports data more frequently?
Thanks.

Snakesms, 2020-12-14:
> I actually need for SNZB-02 to report much more frequently than it is doing it now.
> Is there a way to configure this so it reports data more frequently?
> Thanks.

I just managed to do that easily, thanks to this PR:
There are 2 values that can/should be tuned, as example:

await configureReporting.temperature(endpoint, {min: 5, max: repInterval.MINUTES_15, change: 20});
where "MINUTES_15" scraping interval
and
change: 20 is 0.2 degrees difference
I suppose whatever changes faster

sergimola, 2021-01-05:
> I just managed to do that easily, thanks to this PR:
> There are 2 values that can/should be tuned, as example:
>
> `await configureReporting.temperature(endpoint, {min: 5, max: repInterval.MINUTES_15, [...]

Thank you! That was helpful.

Zoolombo, 2021-03-05:
Hello!
I have a problem. My sonoff SNZB-02 is't measuring humidity. 
I try re-join, but no result...
Battary, temperature, linq is Ok.

btb42, 2021-03-05:
Hi, i have recently the same problem.  literaly 2 hours ago. 

I was traying to re-join, remove device from zigbee2mqtt network. pair it close to different plugs, coordinators, routers.. 
( my device did not sent status data, not even temperature ) , and if it successly join network, and send data - only temperature will be reported to my OH3.

fnaly i just unlink item from humidity channel, and link it back.  because in MQTT message I notice that there are both information.  - so problem was on side that reads data.

paul-at-pebbletech, 2021-12-24:
I had the same problem with SNZB-02. After much tinkering I managed to get it to report properly.
Undertook the following steps.
Remove the device from Z2M with force option.
remove battery from device.
Waited for 30 minutes (Dont know if this is important or not. Was walking the dog)
Restarted HA, MQTT & Z2M containers
Shutdown RASPI (Note restart did not work)
Powered RASPI back on.
Started containers.
Waited 10 minutes
Inserted battery into device
Re-added device.
Bingo the device is now reporting
Hope there is a clue in there somewhere to help with a bug fix