Koenkk/zigbee2mqtt #3798: CC2531 works on Home Assistant SD card image, not on docker.

dave-sausages, 2020-06-23:
After 3 days of failed problem solving I think I'm finally going to have to ask for help.

Short version: zigbee2mqtt addon works on Home assistant SD image, addon doesn't work on Home Assistant running in docker.

Long version: I had a CC2531 stick working fine on the Home assistant image you burn to an SD card and insert in a Pi, all was well. It's a Pi4 4GB btw.

This was the config file:
`data_path: /share/zigbee2mqtt
devices: devices.yaml
groups: groups.yaml
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://core-mosquitto:1883'
  user: REDACTED
  password: REDACTED
serial:
  port: >-
    /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B000BE8B47A-if00		
advanced:
  pan_id: 6754
  channel: 11
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  availability_blacklist: []
ban: []
whitelist: []
queue: {}
socat:
  enabled: false
  master: 'pty,raw,echo=0,link=/dev/ttyZ2M,mode=777'
  slave: 'tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5'
  restartdelay: 1
  initialdelay: 1
  options: '-d -d'
  log: false
`
However I recently changed to running Raspberry pi OS on a USB3 SSD drive, and running Home assistant in docker. And then running Zigbee2MQTT as the home assistant addon just the same as before. But no matter what I try, there are no comms through the CC2531 stick.

Here are the checks I've tried so far:

This is the log output: 
`zigbee2mqtt:info  2020-06-22 22:41:50: Starting zigbee2mqtt version 1.14.0 (commit #unknown)
zigbee2mqtt:info  2020-06-22 22:41:50: Starting zigbee-herdsman...
zigbee2mqtt:info  2020-06-22 22:41:54: zigbee-herdsman started
zigbee2mqtt:info  2020-06-22 22:41:54: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
zigbee2mqtt:info  2020-06-22 22:41:54: Currently 0 devices are joined:
zigbee2mqtt:info  2020-06-22 22:41:54: Zigbee: disabling joining new devices.
zigbee2mqtt:info  2020-06-22 22:41:54: Connecting to MQTT server at mqtt://192.168.2.109:1883
zigbee2mqtt:info  2020-06-22 22:41:54: Connected to MQTT server
zigbee2mqtt:info  2020-06-22 22:41:54: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
zigbee2mqtt:info  2020-06-22 22:41:54: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.14.0","commit":"unknown","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":false}'
zigbee2mqtt:info  2020-06-22 22:42:20: Zigbee: disabling joining new devices.
zigbee2mqtt:info  2020-06-22 22:42:20: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.14.0","commit":"unknown","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":false}'
zigbee2mqtt:info  2020-06-22 22:42:25: Zigbee: allowing new devices to join.
zigbee2mqtt:info  2020-06-22 22:42:25: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.14.0","commit":"unknown","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":true}'` Note: the joining enable and disable is me selecting that on the lovelace dashboard in HA.  I'm pressing the pair button for 3 seconds on my zigbee sensor to try and make it pair.

To confirm my zigbee sensor and stick are still fine, I made another HA SD image and fired up the same pi4 with the same CC2531 stick in it, selected join devices in lovelace, and it paired just fine and was seen in the logs as such.

I had a linux wizard friend TeamViewer into my computer and setup Zigbee2MQTT as it's own docker container, and the exact same behaviour was witnessed. We played around with changing the serial device to /dev/ttyACM0 on both versions and it made no difference. This is the log of the Z2MQTT docker container, and shows basically the same thing:
`zigbee2mqtt:info  2020-06-22 22:29:08: Logging to console and directory: '/app/data/log/2020-06-22.22-29-07' filename: log.txt
zigbee2mqtt:debug 2020-06-22 22:29:08: Removing old log directory '/app/data/log/2020-06-22.22-00-58'
zigbee2mqtt:debug 2020-06-22 22:29:08: Loaded state from file /app/data/state.json
zigbee2mqtt:info  2020-06-22 22:29:08: Starting zigbee2mqtt version 1.14.0 (commit #9009de2)
zigbee2mqtt:info  2020-06-22 22:29:08: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-06-22 22:29:08: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]                                                                   ,"networkKey":"HIDDEN"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","seria                                                                   lPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyACM0"},"adapter":{"concurrent":null}}'
zigbee2mqtt:info  2020-06-22 22:29:11: zigbee-herdsman started
zigbee2mqtt:info  2020-06-22 22:29:11: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revi                                                                   sion":20190608}}'
zigbee2mqtt:debug 2020-06-22 22:29:11: Zigbee network parameters: {"panID":6754,"extendedPanID":"0xdddddddddddddddd","channel":11}
zigbee2mqtt:info  2020-06-22 22:29:11: Currently 0 devices are joined:
zigbee2mqtt:warn  2020-06-22 22:29:11: `permit_join` set to  `true` in configuration.yaml.
zigbee2mqtt:warn  2020-06-22 22:29:11: Allowing new devices to join.
zigbee2mqtt:warn  2020-06-22 22:29:11: Set `permit_join` to `false` once you joined all devices.
zigbee2mqtt:info  2020-06-22 22:29:11: Zigbee: allowing new devices to join.
zigbee2mqtt:info  2020-06-22 22:29:11: Connecting to MQTT server at mqtt://192.168.2.109:1883
zigbee2mqtt:info  2020-06-22 22:29:12: Connected to MQTT server
zigbee2mqtt:info  2020-06-22 22:29:12: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
zigbee2mqtt:info  2020-06-22 22:29:12: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.14.0","commit":"9009de2","coordinator":{"type":"zStack12","                                                                   meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":true}'
zigbee2mqtt:error 2020-06-22 22:31:58: Adapter disconnected, stopping
zigbee2mqtt:debug 2020-06-22 22:31:58: Saving state to file /app/data/state.json
zigbee2mqtt:info  2020-06-22 22:31:58: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
zigbee2mqtt:info  2020-06-22 22:31:58: Disconnecting from MQTT server
zigbee2mqtt:error 2020-06-22 22:31:58: Failed to stop zigbee
`
You can see when I remove the stick, Z2MQTT notices it's missing.

In the docker container, when I type `test -w /dev/ttyACM0 && echo success || echo failure` 
it says success.

I've tried so many other things I'm sure I've forgotten them, but the same problem is present: If I run a Home assistant SD image, Zigbee2QMTT addon works flawlessly, but when I run Home assistant in Docker with Zigbee2MQTT addon, or Zigbee2MQTT solo in docker, it doesn't work, no matter what I try.

Can anyone help? Please? It's driving me nuts.

MPM1107, 2020-06-23:
When you switched from one setup from the other, what data did you transfer? Did you just use the same config options or did you transfer over anything else?

dave-sausages, 2020-06-23:
> When you switched from one setup to the other, what data did you transfer? Did you just use the same config options or did you transfer over anything else?

Initially all the config options and data were the same as I took a hassio snapshot and then restored it on the new docker version of home assistant. In further testing I deleted the add-on completely including the zigbee2mqtt folder that gets left behind and started from scratch. And then after that failed, used the previous configs as another test. The same was done for the standalone docker zigbee2mqtt instance.

MPM1107, 2020-06-23:
Did you change the PAN id and extended PAN id after starting from scratch in the config? I have the feeling that your devices are trying to connect to a network they already know, but z2m doesn't know about the devices anymore as the data got lost. So changing those two IDs and then repairing might help, not sure though.

dave-sausages, 2020-06-23:
I didn't change either of those from default, but I also never configured them in the original hassio version either, they have always been default. I should note that during all this testing, I have made another hassio image and fired it up on the same pi with the same CC2531. zigbee2Mqtt loads fine and when I press the join/reset button on my zigbee device, it joins just fine.

MPM1107, 2020-06-23:
Oh, that's a bit odd then. Herdsman prints out the coordinator version properly, so the communication between the stick and the software is working as expected, that can be ruled out. Apart from what I said with the network IDs I'm also a bit lost on ideas, not sure why the docker version would behave differently.

dave-sausages, 2020-06-23:
Yeah I noticed herdsman displays the firmware I flashed it with (I've also got a spare stick and reflashed both as part of testing) so it appears to be talking the stick okay. And when I remove the stick, herdsman doesn't load, which is another bit of proof.

dave-sausages, 2020-06-23:
Are there any tests regarding the "network IDs" you can suggest?

MPM1107, 2020-06-23:
Since it can't hurt anything, just try changing both PAN IDs (they can be any random numbers) and try pairing a device again (with joining of new devices set to true of course). If that doesn't work either it's just another thing to rule out I guess.

dave-sausages, 2020-06-25:
So I changed the pan ID in the config from 6754 to 6753. Now the red and green LED on the CC2531 are flashing (2hz red and 0.5hz Green approx).

I've set the Pan ID back to 6754 and it's still doing the same flashing.

dave-sausages, 2020-06-28:
Can someone confirm this is a bug with zigbee2mqtt and I should give up trying to get it to work on anything other than the SD card image of Home Assistant until it is fixed?

Koenkk, 2020-06-29:
Maybe I missed it, but it's unclear to me what your exact problem is. As @MPM1107 already noticed, Zigbee2mqtt starts fine and communication works. 

Seeing that you are running on a Raspberry Pi 4, make sure to connect the CC2531 through a USB extension cable as the Pi 4 generates a lot of interference (killing all wireless Zigbee communication)

dave-sausages, 2020-06-29:
> Maybe I missed it, but it's unclear to me what your exact problem is. As @MPM1107 already noticed, Zigbee2mqtt starts fine and communication works.
>
> Seeing that you are running on a Raspberry Pi 4, [...]

Problem is:

Zigbee2mqtt works fine on RPi4 with Home assistant running on SD card image with my CC2531 stick plugged straight into the usb port.

Zigbee2mqtt does NOT work with same RPi4 and same stick in same port with Home Assistant running on SSD. I have also tried the CC2531 on an extension lead, no difference. And by "not work" I mean there is zero communication through the zigbee stick. The MQTT part works, but the zigbee part doesn't which is quite crucial.

dave-sausages, 2020-06-29:
I should also note that since following @MPM1107 suggestion to change the pan id, my CC2531 now starts flashing red and green whenever zigbee2mqtt gets to the "Starting zigbee-herdsman..." step in the log. I've reflashed it, and it still does it. I've also removed the zigbee2mqtt addon and reinstalled it, and the same thing happens.

Koenkk, 2020-06-29:
So Zigbee2mqtt starts correctly?

Did you try to repair your device? Before doing this first make sure you complete zigbee network is down (all routers)

dave-sausages, 2020-06-29:
Oh Jeez, I finally solved it!

It turns out that my USB extension was faulty. So with the CC2531 in the SD card version, there was no USB traffic (so it worked straight in the port).

And my USB extension cable was faulty, and giving the same issues. I decided to use the 5m extension cable from my Oculus Rift, and it fired straight up!

That's some bad luck right there. I've cut the bloddy thing in half to make sure I never use it!

Thanks for all the effort though.

Koenkk, 2020-06-29:
Good that you find the issue. Enjoy zigbee2mqtt :)

closed 2020-06-29 by Koenkk