Koenkk/zigbee2mqtt #3801: CC2531 keeps disconnecting (Intel Nuc, Ubuntu)

bamaas, 2020-06-23:
# Bug Report
## What happened
I ordered a pre flashed CC2531 stick and it worked quite well. occasionally I had to unplug and replug the device because the network was frozen(?). This happens when I sent a lot of messages. So I unplugged the device and plugged it back in but from now on the device keeps dropping connection. It is visible for a few seconds and then disconnected.

What needs to be said: I have pressed both buttons on the stick, and long pressed.

## How to reproduce it
Plug in the device.

## Debug Info
Zigbee2mqtt version: latest
Adapter hardware: CC2531
Adapter firmware version: unkown
Already removed modemmanager from Ubuntu
Tried with and without an extension cable, no difference.
The Ubuntu log is showing errors like these: _usb 2-4: device descriptor read/64, error -110_
Also tried on a Mac, couldn't get zigbee2mqtt working, but I saw the same behaviour of the stick: the device is visible on my system and then disconnects. 

**dmesg log (random partial)**
`[ 1621.064221] usb 2-4: new full-speed USB device number 19 using xhci_hcd
[ 1621.216987] usb 2-4: New USB device found, idVendor=0451, idProduct=16a8, bcdDevice= 0.09
[ 1621.216992] usb 2-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1621.216995] usb 2-4: Product: TI CC2531 USB CDC
[ 1621.216998] usb 2-4: Manufacturer: Texas Instruments
[ 1621.217000] usb 2-4: SerialNumber: __0X00124B0019389D77
[ 1621.218609] cdc_acm 2-4:1.0: ttyACM0: USB ACM device
[ 1635.391454] br-3b2735e843d4: port 3(veth4f41597) entered blocking state
[ 1635.391458] br-3b2735e843d4: port 3(veth4f41597) entered disabled state
[ 1635.391755] device veth4f41597 entered promiscuous mode
[ 1635.915348] eth0: renamed from veth0ed5bb2
[ 1635.932586] IPv6: ADDRCONF(NETDEV_CHANGE): veth4f41597: link becomes ready
[ 1635.932669] br-3b2735e843d4: port 3(veth4f41597) entered blocking state
[ 1635.932673] br-3b2735e843d4: port 3(veth4f41597) entered forwarding state
[ 1639.661794] usb 2-4: USB disconnect, device number 19
[ 1639.661977] cdc_acm 2-4:1.0: failed to set dtr/rts
[ 1639.972005] usb 2-4: new full-speed USB device number 20 using xhci_hcd
[ 1655.395768] usb 2-4: device descriptor read/64, error -110
[ 1671.011488] usb 2-4: device descriptor read/64, error -110
[ 1671.247460] usb 2-4: new full-speed USB device number 21 using xhci_hcd
[ 1686.627189] usb 2-4: device descriptor read/64, error -110
[ 1697.330991] usb 2-4: device descriptor read/64, error -110
[ 1697.439051] usb usb2-port4: attempt power cycle
[ 1698.090946] usb 2-4: new full-speed USB device number 22 using xhci_hcd
[ 1698.115744] usb 2-4: New USB device found, idVendor=0451, idProduct=16a8, bcdDevice= 0.09
[ 1698.115750] usb 2-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1698.115755] usb 2-4: Product: TI CC2531 USB CDC
[ 1698.115757] usb 2-4: Manufacturer: Texas Instruments
[ 1698.115760] usb 2-4: SerialNumber: __0X00124B0019389D77
[ 1698.117757] cdc_acm 2-4:1.0: ttyACM0: USB ACM device
[ 1755.948710] usb 2-4: USB disconnect, device number 22
[ 1756.257770] usb 2-4: new full-speed USB device number 23 using xhci_hcd`

**zigbee2mqtt log:**
`zigbee2mqtt:info  2020-06-23 23:19:08: Logging to console and directory: '/app/data/log/2020-06-23.23-19-08' filename: log.txt

zigbee2mqtt:info  2020-06-23 23:19:08: Starting zigbee2mqtt version 1.14.0 (commit #9009de2)

zigbee2mqtt:info  2020-06-23 23:19:08: Starting zigbee-herdsman...

zigbee2mqtt:info  2020-06-23 23:19:11: zigbee-herdsman started

zigbee2mqtt:info  2020-06-23 23:19:11: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'

zigbee2mqtt:info  2020-06-23 23:19:11: Currently 13 devices are joined:

zigbee2mqtt:info  2020-06-23 23:19:11: living_light_desk (0x00178801084a63ec): 929001953101 - Philips Hue White and Color Ambiance GU10 (Router)

zigbee2mqtt:info  2020-06-23 23:19:11: living_light_tv (0x00178801084a554f): 929001953101 - Philips Hue White and Color Ambiance GU10 (Router)

zigbee2mqtt:info  2020-06-23 23:19:11: lavalamp_socket (0x00158d00039fb096): ZNCZ02LM - Xiaomi Mi power plug ZigBee (Router)

zigbee2mqtt:info  2020-06-23 23:19:11: bedroom_light_window (0x00178801084cb0d1): 929001953101 - Philips Hue White and Color Ambiance GU10 (Router)

zigbee2mqtt:info  2020-06-23 23:19:11: bedroom_temperature (0x00158d000421de08): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor (EndDevice)

zigbee2mqtt:info  2020-06-23 23:19:11: living_temperature (0x00158d00046545e1): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor (EndDevice)

zigbee2mqtt:info  2020-06-23 23:19:11: kitchen_light_sink (0x00178801084cbe11): 929001953101 - Philips Hue White and Color Ambiance GU10 (Router)

zigbee2mqtt:info  2020-06-23 23:19:11: toilet_light (0x00178801084042c9): 929001953101 - Philips Hue White and Color Ambiance GU10 (Router)

zigbee2mqtt:info  2020-06-23 23:19:11: landing_light_stairwell (0x0017880108401d71): 929001953101 - Philips Hue White and Color Ambiance GU10 (Router)

zigbee2mqtt:info  2020-06-23 23:19:11: kitchen_light_table (0x001788010853b604): 929001953101 - Philips Hue White and Color Ambiance GU10 (Router)

zigbee2mqtt:info  2020-06-23 23:19:11: bedroom_light_bed (0x001788010840c891): 929001953101 - Philips Hue White and Color Ambiance GU10 (Router)

zigbee2mqtt:info  2020-06-23 23:19:11: kitchen_light_window (0x00178801068f4c00): 929001953101 - Philips Hue White and Color Ambiance GU10 (Router)

zigbee2mqtt:info  2020-06-23 23:19:11: entrance_light (0x0017880108ac6d93): 9290022166 - Philips Hue white and color ambiance E26/E27 (Router)

zigbee2mqtt:info  2020-06-23 23:19:11: Zigbee: disabling joining new devices.

zigbee2mqtt:error 2020-06-23 23:19:11: Adapter disconnected, stopping

zigbee2mqtt:error 2020-06-23 23:19:11: Not connected to MQTT server!

zigbee2mqtt:error 2020-06-23 23:19:11: Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: 'offline

zigbee2mqtt:info  2020-06-23 23:19:11: Disconnecting from MQTT server

(node:17) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'end' of undefined

    at MQTT.disconnect (/app/lib/mqtt.js:87:21)

    at async Controller.stop (/app/lib/controller.js:174:9)

    at async Controller.onZigbeeAdapterDisconnected (/app/lib/controller.js:187:9)

(node:17) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)

(node:17) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

(node:17) UnhandledPromiseRejectionWarning: Error: Cannot request when znp has not been initialized yet

    at Znp.request (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:240:19)

    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:142:32)

    at Generator.next (<anonymous>)

    at /app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:8:71

    at new Promise (<anonymous>)

    at __awaiter (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:4:12)

    at Object.func (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:140:44)

    at Queue.<anonymous> (/app/node_modules/zigbee-herdsman/dist/utils/queue.js:29:46)

    at Generator.next (<anonymous>)

    at /app/node_modules/zigbee-herdsman/dist/utils/queue.js:8:71

(node:17) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 3)`

Koenkk, 2020-06-24:
Looks that something may be wrong with the stick. Can you try reflashing it with the source routing firmware? (https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing)

bamaas, 2020-06-25:
Oke I will try that later this week when I've got my cc debugger.

purche42, 2020-06-26:
I have had the same problem. I have re-flashed the stick and it worked again.
I'm not sure what caused the problem. I have added a new device (from 32th -> 33th device) and I have udapted to Zigbee2MQTT version 1.14.0 one day before. (BTW great software).
Maybe the limit was hit and stick was confused even I use source routing firmware.

bamaas, 2020-07-04:
Oke, reflashing solved the problem. Everything seems stable so far....

lucaspiller, 2020-07-08:
My experience has been that I need to reflash the coordinator every month or so as the same thing happens. I have two CC2531 devices now, so I flash them both and can quickly switch them over if one dies, but obviously it's not a great solution. I assume it's simply something in the NVRAM on the device that is messed up, it would be great if the reset button could be used to clear that back to the state after being flashed.

purche42, 2020-07-08:
Yes .. it would be also nice to initiate a reset from Z2M . 
I get the advise from my supplier to use iobroker.
I did not try this but it seems to be possible to hard reset without flashing.    

Until now it works fine and I also have a 2nd stick for backup ;)

closed 2020-08-16 by github-actions[bot]