Koenkk/zigbee2mqtt #3805: RTCGQ11LM doesn't report units of measure for illuminance sensor

aalbul, 2020-06-24:
# Bug Report
Hello, even though device works well for me, it doesn't report units of measure to home assistant for illuminance sensor.
Xiaomi temperature/humidity sensors reports C and % without problems, so i assume that technically it must be possible. Missing functionality?

## What did you expect to happen
Illuminance sensors must have "lx" unit of measure

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version: 1.14.0 via Docker
Adapter hardware: CC2531
Adapter firmware version: CC2531_DEFAULT_20190608

MPM1107, 2020-06-24:
This does indeed seem to be one of the few illuminance sensors that only reports the value without a unit, most of the others are set up to report both versions. I have to admit that I don't know the reason for that, probably just a simple human mistake?

If you want to get this fixed quickly on your end you will have to edit this line of code: [https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/extension/homeassistant.js#L753](https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/extension/homeassistant.js#L753)
Change that line so that the list also includes `cfg.sensor_illuminance_lux` in addition to everything else.
Otherwise you'll have to wait for this to be fixed and merged into the main codebase which can take a while.

aalbul, 2020-06-24:
Thank you for your reply. I'm not in rush, so i will definitely wait. Just wanted to report an issue to make this project even better :)

MPM1107, 2020-06-24:
Ah, the rabbit hole goes a bit deeper: [https://github.com/Koenkk/zigbee-herdsman-converters/issues/1017](https://github.com/Koenkk/zigbee-herdsman-converters/issues/1017)

@Koenkk What's the best way to solve this? Remove normal illuminance from the homeassistant config and just report the lux version? I've seen that done for one other sensor. Would that break anything?

Koenkk, 2020-06-24:
Initially we only had `illuminance` which was the raw `illuminance` value from the device which was not always (!) in Lux. This was very confusing as sensor A could return much different values than sensor B. However as converting all values to Lux would be a big breaking change (leading to many broken automations) we decided to keep the `illuminance` as is and add a new value `illuminance_lux` which is guaranteed to be the illuminance value in lux. 

It seems that this illuminance_lux sensor was not yet added to the RTCGQ11LM so I just did this. Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

MPM1107, 2020-06-24:
Well it's not that it hasn't been added, it was intentionally removed as part of the issue I linked above. That's why I'm wondering whether it's okay to just add it back now. Here's the commit: [https://github.com/Koenkk/zigbee2mqtt/commit/11ec1915803391d3264f59eab4009bba999f01f5](https://github.com/Koenkk/zigbee2mqtt/commit/11ec1915803391d3264f59eab4009bba999f01f5)

Koenkk, 2020-06-24:
Yes you are right, I missed this.

So I think we should keep the `illuminance` sensor as is and just add the `lx` unit for the `RTCGQ11LM` (done this now in the latest dev branch

closed 2020-07-26 by Koenkk